lwfinger/rtl8188eu #87: tp-link wr1043n v2

expandables2, 2015-01-06:
Hi i install the rtl8188eu and it is not showing up in openwrt network wifi. How to do get it to appear and be able to scan for networks? I am a noob!

expandables2, 2015-01-06:
Please help

expandables2, 2015-01-06:
Hi sir i went to system, software and then updatelist. I then went to the packages and install the r8188eu-firmware.  I plug in my alfa awus036nhv and went to network and then wifi and the Wireless Overview does not show the alfa awus036nhv. Only the build in router wifi.

lwfinger, 2015-01-06:
Thst makes no sense to me. Do it again, and paste a copy of what you did in the terminal here. Do it in a terminal window, not using a GUI.

expandables2, 2015-01-06:
I install in web browser 192.168.1.1 I don't know how to install in terminal can you show me the instructions please thank you

expandables2, 2015-01-06:
I am using openwrt

lwfinger, 2015-01-06:
That helps. The driver in both the master and kernel_code branches will not work with openWRT because they do not support cfg80211. The one in the v4.1.8_9499 branch does support cfg80211 and may work with openWRT; however, that one has had very little testing. For example, if your device has bid-endian architecture, I am 85% certain that it will not work.

expandables2, 2015-01-06:
How do i install the v4.1.8_9499 branch on openwrt? Can I get some instructions please thanks

lwfinger, 2015-01-06:
Sorry. I do not work with openWRT very often. When I make a change like this, I build the driver with the commands needed to cross-compile the driver on my laptop, use scp to load the resulting ko file onto the device in /tmp, then ssh into the device to load and test the device. Once the driver is OK, then I place a copy in /lib/modules/... I have no idea what GUI you are using, nor how you would compile and load a driver using it.

expandables2, 2015-01-06:
Ok can i get the .ko file? i think openwrt has opkg to install packages

lwfinger, 2015-01-06:
It is made by using the "make" command. Because it is dependent on the exact kernel you are using, no one else can build it.

expandables2, 2015-01-06:
OK thanks

expandables2, 2015-01-06:
I am using   Firmware Version OpenWrt Barrier Breaker 14.07 / LuCI Trunk (0.12+svn-r10530)
Kernel Version	3.10.49

expandables2, 2015-01-06:
Not working driver r8188eu-firmware 2014-03-16-f8c22c692bdee57a20b092e647464ff6176df3ed-1
RealTek r8188eu firmware