lwfinger/rtl8188eu #88: BUG - hang debian if wifi frommaster works

nortikin, 2015-01-10:
downloaded, made install and difference between master and stable versions - stable version is slow, but it works, master is fast, but hangs Debian completely
```
~$ lsusb 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 002: ID 0bda:0179 Realtek Semiconductor Corp. 
Bus 001 Device 003: ID 13fe:3e00 Kingston Technology Company Inc. 
```
stable version now installed
```
~$ lsmod | grep wifi
rtlwifi                71765  0 
mac80211              171389  1 rtlwifi
cfg80211              113445  3 8188eu,mac80211,rtlwifi
usbcore               104555  6 ehci_hcd,ohci_hcd,usb_storage,8188eu,rtlwifi
```
```
~$ sudo iwconfig 
lo        no wireless extensions.

wlan0     IEEE 802.11bgn  ESSID:"host1"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.412 GHz  Access Point: C4:6E:1F:85:FC:07   
          Bit Rate:72.2 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=100/100  Signal level=97/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

master hangs and nothing can do with it

lwfinger, 2015-01-10:
I am a little confused. Why is rtlwifi loaded? The RTL8188EU does not use it.

What do you mean by "stable" and "master" versions?

artantaaa, 2015-01-11:
I'm using Debian Wheezy, and my system also froze immediately after installing this driver and connecting to a network. After a restart, everything was fine until I tried to open a browser. Then everything would freeze again, as in the mouse cursor won't move and the clock stops ticking. A few months ago, and as far back as a year, it was working fine. 

I installed it like this:
    mkdir ~/RTL8188EU
    cd ~/RTL8188EU
    git clone git://github.com/lwfinger/rtl8188eu
    cd ~/RTL8188EU/rtl8188eu
    make
    sudo make install
    sudo cp -v ~/RTL8188EU/rtl8188eu/rtl8188eufw.bin /lib/firmware/rtlwifi/
    sudo depmod -a
    sudo update-initramfs -u
    sudo modprobe 8188eu

Anything wrong there?

I'm not at that computer now, so I can't give you a lsusb, but I'm using one of these:
[image]

nortikin, 2015-01-11:
https://github.com/lwfinger/rtl8188eu/tree/v4.1.8_9499
this is stable version
master branch is danger to install as always ))
previously installed other drivers, but last is this one stable and it works.