lwfinger/rtl8188eu #90: Was working, but now running into problems

IAmTheRealGoof, 2015-01-20:
First, thanks for all the work on this driver. I have used it on and off w/ mostly good results on a virtualized ubuntu 10.04 (think it was 2013 11 04) with my TL-WN725N (ID 0bda:8179 , RTL8188EUS 802.11n)

As of late this has started giving me a bit of a headache, as it broke (think an update), and since I've had mixed, but mostly not so good results going with a more recent releases of Centos/mint/xbuntu. Either problems compiling, no signal, driver failing to load, etc.

I can go into detail but I would like to try what you use, or that you know that works. (since at least part of it is my fault juggling driver versions, etc) I prefer lightweight/low install requirement installs but am happy to try anything that is known to work.

But if willing to help chase problems, which distro do you want to chase, and what information would you like?

lwfinger, 2015-01-21:
The distro does not matter. Use any one you want. If the problems are with compiling on older kernels on Ubuntu, then look to Ubuntu for the problems. They have benn making some unwise changes that make supporting external drivers very difficult. In fact, I have already wasted too much time and refuse to support Ubuntu.

Do some research and learn how to use git to bisect a problem. If you can point to a particular commit that causes problems, then that would be a help. BTW, both this repo and the kernel version of the drivers work fine for me.

IAmTheRealGoof, 2015-01-21:
I gathered that ubuntu based and older kernels don't compile correctly.(reading these forums after spending a bit of time trying to get it to work on xbuntu/ubuntu 14.04(kernel 3.13.0) and centos 5.11(kernel 2.6.18). I'm not asking for you help on those, but would like to start over with a combination that you know that works.

I've also tried centos 7, but couldn't even get it to be seen as a wireless nic (and that was after blacklisting it's provided drivers, etc Oh and lets not forget the mess of dealing w/ the new way of dealing with network and then systemd).

Between different dristros and driver versions, I've created a bit of a mess. So please, can you give me some recommendations of what you know works (or pretty certain works) 

PS there is also a chance my mac got blocked w/ all the failed attempts to connect but am pursuing that separately. And I wont have access to git until I can connect via wifi :(

Thanks again in advance!

IAmTheRealGoof, 2015-01-21:
Centos 7
3.10.0-123.el7.x86_64 #1 SMP Mon Jun 30 12:09:22 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux

Using driver downloaded today via "Download Zip" on master branch. Compiles and installs w/o errors. I still have blacklist r8188eu. Rebooted, ifconfig shows only ethernet interface. lsusb shows device , lsmod shows its loaded but used by 0.

Relevant info from modinfo 8188eu:
filename:       /lib/modules/3.10.0-123.el7.x86_64/kernel/drivers/net/wireless/8188eu.ko
version:        v4.1.4_6773.20130222
vermagic:       3.10.0-123.el7.x86_64 SMP mod_unload modversions 

Cherry picking some info from /var/log/messages
Jan 21 11:54:31 localhost kernel: Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
Jan 21 11:54:32 localhost kernel: usbcore: registered new interface driver r8188eu
Jan 21 11:54:32 localhost systemd-udevd[514]: renamed network interface wlan0 to wlp0s11u1
...
Jan 21 11:54:34 localhost NetworkManager[823]: <info> monitoring kernel firmware directory '/lib/firmware'.
Jan 21 11:54:34 localhost NetworkManager[823]: <info> Loaded device plugin: /usr/lib64/NetworkManager/libnm-device-plugin-adsl.so
Jan 21 11:54:34 localhost NetworkManager[823]: <info> Loaded device plugin: /usr/lib64/NetworkManager/libnm-device-plugin-bluetooth.so
Jan 21 11:54:34 localhost NetworkManager[823]: <info> Loaded device plugin: /usr/lib64/NetworkManager/libnm-device-plugin-wwan.so
...
Jan 21 11:54:34 localhost NetworkManager[823]: <info> WiFi enabled by radio killswitch; enabled by state file
Jan 21 11:54:34 localhost NetworkManager[823]: <info> WWAN enabled by radio killswitch; enabled by state file
Jan 21 11:54:34 localhost NetworkManager[823]: <info> WiMAX enabled by radio killswitch; enabled by state file
...
Jan 21 11:54:34 localhost NetworkManager[823]: <warn> (wlp0s11u1): failed to initialize WiFi driver

/var/log/dmesg
[    2.618522] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
[    2.892916] usbcore: registered new interface driver r8188eu
[    2.939127] systemd-udevd[514]: renamed network interface wlan0 to wlp0s11u1

from here I'm kinda stuck. I was able to force the interface at one point, but it made it an ethernet interface but could not apply wifi security stuff to it.

IAmTheRealGoof, 2015-01-21:
I turned logging to debug and saw that it referenced itself as r8188eu, so grasping, I changed os_dep/usb_intf.c and include/drv_types.h driver name to not include the "r", but alas, that wasn't it. Bellow was one of the debug logs 

Jan 21 14:15:06 localhost kernel: usb 1-1: new high-speed USB device number 8 using ehci-pci
Jan 21 14:15:06 localhost kernel: usb 1-1: New USB device found, idVendor=0bda, idProduct=8179
Jan 21 14:15:06 localhost kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 21 14:15:06 localhost kernel: usb 1-1: Product: 802.11n NIC
Jan 21 14:15:06 localhost kernel: usb 1-1: Manufacturer: Realtek
Jan 21 14:15:06 localhost kernel: usb 1-1: SerialNumber: 00E04C0001
Jan 21 14:15:06 localhost kernel: Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
Jan 21 14:15:06 localhost kernel: EEPROM ID = 0x8129
Jan 21 14:15:06 localhost NetworkManager[4058]: <debug> [1421874906.645421] [platform/nm-linux-platform.c:1477] event_notification(): netlink event (type 16) for link: wlan0 (9, family 0)
Jan 21 14:15:06 localhost NetworkManager[4058]: <debug> [1421874906.645476] [platform/nm-linux-platform.c:257] get_kernel_object(): get_kernel_object for link: wlan0 (9, family 0)
Jan 21 14:15:06 localhost mtp-probe: checking bus 1, device 8: "/sys/devices/pci0000:00/0000:00:0b.0/usb1/1-1"
Jan 21 14:15:06 localhost mtp-probe: bus: 1, device: 8 was not an MTP device
Jan 21 14:15:06 localhost NetworkManager[4058]: <debug> [1421874906.687893] [platform/nm-linux-platform.c:1477] event_notification(): netlink event (type 16) for link: wlp0s11u1 (9, family 0)
Jan 21 14:15:06 localhost NetworkManager[4058]: <debug> [1421874906.687962] [platform/nm-linux-platform.c:257] get_kernel_object(): get_kernel_object for link: wlp0s11u1 (9, family 0)
Jan 21 14:15:06 localhost systemd-udevd[4105]: renamed network interface wlan0 to wlp0s11u1
Jan 21 14:15:06 localhost NetworkManager[4058]: <debug> [1421874906.695064] [platform/nm-linux-platform.c:3296] handle_udev_event(): UDEV event: action 'add' subsys 'net' device 'wlp0s11u1' (9); seqnum=1628
Jan 21 14:15:06 localhost NetworkManager[4058]: <debug> [1421874906.695226] [platform/nm-platform.c:2189] log_link(): signal: link added  : 9: wlp0s11u1 <DOWN> mtu 1500 wifi driver 'r8188eu' udi '/sys/devices/pci0000:00/0000:00:0b.0/usb1/1-1/1-1:1.0/net/wlp0s11u1'
Jan 21 14:15:06 localhost NetworkManager[4058]: <debug> [1421874906.695316] [devices/nm-device-wifi.c:303] constructor(): (wlp0s11u1): kernel ifindex 9
Jan 21 14:15:06 localhost NetworkManager[4058]: <warn> (wlp0s11u1): failed to initialize WiFi driver
Jan 21 14:15:06 localhost NetworkManager[4058]: <debug> [1421874906.736702] [platform/nm-linux-platform.c:3296] handle_udev_event(): UDEV event: action 'move' subsys 'net' device 'wlp0s11u1' (9); seqnum=1637
Jan 21 14:15:06 localhost NetworkManager[4058]: <debug> [1421874906.737439] [platform/nm-platform.c:2189] log_link(): signal: link changed: 9: wlp0s11u1 <DOWN> mtu 1500 wifi driver 'r8188eu' udi '/sys/devices/pci0000:00/0000:00:0b.0/usb1/1-1/1-1:1.0/net/wlp0s11u1'
Jan 21 14:16:38 localhost NetworkManager[4058]: <debug> [1421874998.185754] [nm-modem-manager.c:196] poke_modem_cb(): Requesting to (re)launch modem-manager...

lwfinger, 2015-01-22:
The zip files from GitHub are a lot less useful that doing a "git clone" .

It does not matter that the driver refers to r8188eu. What is important is the name of the .ko file produced.

When I plug in my device, this is what gets logged in dmesg:

[196570.962540] usb 3-6: new high-speed USB device number 10 using xhci_hcd
[196571.127477] usb 3-6: New USB device found, idVendor=0bda, idProduct=8179
[196571.127486] usb 3-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[196571.127491] usb 3-6: Product: 802.11n NIC
[196571.127495] usb 3-6: Manufacturer: Realtek
[196571.127499] usb 3-6: SerialNumber: 00E04C0001
[196571.129602] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_A_CUT_1T1R_RomVer(0)
[196571.153608] EEPROM ID = 0x8129
[196572.211673] r8188eu 3-6:1.0 wlp0s20u6: renamed from wlan0
[196572.262069] R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1
[196572.591391] MAC Address = a0:f3:c1:15:41:cf
[196572.596614] R8188EU: ERROR indicate disassoc
[196573.923314] R8188EU: ERROR indicate disassoc
[196574.123501] R8188EU: ERROR assoc success

At this point, NetworkManager had made a connection. Please post the same output for your system.

IAmTheRealGoof, 2015-01-22:
This again using centos 7, 3.10, driver version 4.1.4_6773.20130222 and using dmesg -w as I plug the device in.

[   59.135061] usb 1-1: new high-speed USB device number 2 using ehci-pci
[   59.318797] usb 1-1: New USB device found, idVendor=0bda, idProduct=8179
[   59.318831] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   59.318860] usb 1-1: Product: 802.11n NIC
[   59.318887] usb 1-1: Manufacturer: Realtek
[   59.318910] usb 1-1: SerialNumber: 00E04C0001
[   59.330281] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
[   59.703329] EEPROM ID = 0x8129
[   59.752021] systemd-udevd[3073]: renamed network interface wlan0 to wlp0s11u1

10+ min pass and no changes w/ dmesg, however this was logged in messages (keeping to 2 lines) 
Jan 21 18:52:41 localhost systemd-udevd[3073]: renamed network interface wlan0 to wlp0s11u1
Jan 21 18:52:41 localhost NetworkManager[818]: <warn> (wlp0s11u1): failed to initialize WiFi driver

I can use a different version but I get mostly the same results.(or failures to compile IIRC)

Thanks again!

lwfinger, 2015-01-22:
What does the command 'md5sum /lib/firmware/rtlwifi/rtl8188eufw.bin' show? I get
aaef52a47852e599cbff63a3e7f96a94  /lib/firmware/rtlwifi/rtl8188eufw.bin

If you get the same, how do you tell NetworkManager to make a connection? I usually start it from the applet.

IAmTheRealGoof, 2015-01-22:
same MD5, network manager does it automagically, I haven't had the chance do do anything with an interface as I never was presented one.(in centos 7) and I have started it in non daemon debug mode but it wasn't showing me anything more

I have to say, centos 7(and gnome 3) is very different than what I'm used to on many levels, but I did I've also try adding the wifi as an interface via cli, or via some of the gui's, or adding a wifi connection. As far as I can tell everything I need for wifi is installed..

I was able to run yum update on another network but that didn't fix anything.

So, do you have TL-WN725N (ID 0bda:8179 , RTL8188EUS 802.11n)? And distro/release/kernel?

and doing the get clone https://github.com/lwfinger/rtl8188eu.git pulls the most recent changes for this branch? And clicking to download the zip gives me what? What's the difference between the 3 branches?

Thanks again for all your help, I really do appreciate it!

lwfinger, 2015-01-22:
I am not a Gnome nor Centos user, and I cannot help you with that.

Yes, I have a TL-WN725N and I use openSUSE 13.2. Kernel 3.17.4 is the current default, but I am usually running the development kernel from either mainline, or wireless-drivers git repo. At the moment, that means 3.19-rc4 or -rc5.

Downloading the zip file gets you the current state of the GitHub repo, just as 'git clone' does. The difference is what happens when a small change is submitted. With the zip file, you need to download everything again. With the git repo, you do 'git pull' and download only a small amount of information. For a small repo such as this, it does not make much difference, but for the kernel, it is the difference between 80 MB, and 200 KB.

The master branch is the working branch. Branch v4.1.8_9499 contains a newer driver that some claim gives better performance; however, I do not support it, just make it available. Branch kernel-code is self explanatory.

IAmTheRealGoof, 2015-01-23:
Putting this on the back burner for now, but this isn't just a driver problem. It's compounded with NetworkManger not handling WPA2 psk correctly and virtualbox USB bugs.

For those using this to help them with similar problems: 
Lots of examples of people uninstall network manager and installing wicd to authenticate correctly. (one example: http://forums.linuxmint.com/viewtopic.php?f=53&t=94495&start=40 ) And some people having unstable/unpredictable WiFi usage in VM's with a long standing bug:
https://www.virtualbox.org/ticket/9511

lwfinger, 2015-01-24:
I have no idea what you are talking about. I just installed this driver on an openSUSE 13.2 KDE desktop in a VirtualBox VM. After adding the necessary extensions to allow passing a USB connection from the host (also openSUSE 13.2) to the guest, I was able to connect to both WPA and WPA2 PSK. That was using NetworkManager, which works flawlessly for me. Yes, it may not handle some specific network configurations, but it works for simple wired and wireless setups.

That bug in VB is very old. Are you sure it has not been fixed?

BTW, I consider wicd to be the closest thing to malware that I have ever found in Linux. When I have installed it to test for some user, I end up with endless system crashes. In one instance, I even had to reinstall to geet my system to work.

IAmTheRealGoof, 2015-01-27:
I have many different copies/builds/version of the driver, I am pretty sure one of them worked many months until a reboot for my ubuntu 10.04.4. The only things updated/changed from last reboot was things like unzip, libcurl3, coreutils, but where working for ~2 weeks since their update. 

With removing networkmanager and installing wicd, there are dozens of examples of people using that as a fix to passphrase not being accepted, but there are some where remove/reinstall work(and some going the other way). This seamed pretty common for ubuntu/mint forums.

Re: VB bug. It's not listed as fixed, I found that by finding more recent posts pointing to that. (and there are newer posts to that bug)

I don't feel that I have done a good enough job keeping track of things to single out any problem. It's even possible something changed on my host to cause something on the guests. The first two times this was pretty much download compile install. IIRC it was an older version of your provided source in a tar file that gave me good stable signal where the current one I'd work for a short time and disconnect where I'd have to restart the guest.

With Centos 7 (and I think a newer ubuntu) came with a driver that almost worked but would never successfully join re-asking for the passphrase, none of the drivers I have worked better than that(most worse).
 
That Ubuntu 10.4 with v4.1.8_9499.20131104, srcversion 35EC910F3D0635B1D006357, I can do a scan and see 100/100 signal, but will largely will repeat. 
RTL871X: _rtw_pwr_wakeup call ips_leave....
RTL871X: ==>ips_leave cnts:18
RTL871X: ===> rtw_ips_pwr_up..............
RTL871X: ===> ips_netdrv_open.........
RTL871X: Fail to polling Offset[0x6]
RTL871X: 3InitPowerOn_rtl8188eu: run power on flow fail
RTL871X: rtl8188eu_hal_init in 52ms
RTL871X: rtw_hal_init: hal__init fail
RTL871X: -ips_netdrv_open - drv_open failure, bup=1
RTL871X: <=== rtw_ips_pwr_up.............. in 52ms
RTL871X: nolinked power save leave
RTL871X: ==> ips_leave.....LED(0x00000000)...
RTL871X: ======> ips_leave fail.............

While I can trace the error Fail to polling Offset to what looks to be power management part of the code, I'm not sure what it's doing, or what's wrong. I just assumed it's not the driver that worked for me and started trying different versions, some compiled, some warn, some wouldn't compile, some I would never see any AP's, and some failed to load. 

So then I tried a few newer distos and then came to this board.  But I'll give opensuse 13.2 a try for this, I use to use SUSE all the time, and my home server is still running 12.3. 

I admit that I'm a newb when it comes to networkmanager or the wifi as until recently I've always config network by vi'ing ifcfg-eth0(or interfaces) and disabled things like networkmanager if present.

thanks for you patients!