lwfinger/rtl8188eu #91: Compilation OK, but not shown in lsusb

gaara92, 2015-01-21:
Hi,
I'm trying to install this driver on a Orange-Pi, this is the onboard wifi card (rtl8188etv)
So, I have compiled the 3.4.105 kernel to have the header, and make this compilation. (make all & sudo make install).
No errors to compile it, but 'lsusb' don't show the wifi card after reboot.

You can see my question here to see the details: http://unix.stackexchange.com/questions/180294/install-rtl8188etv-rtl8188eu-driver-on-orange-pi

If you have an idea... Thanks

lwfinger, 2015-01-21:
No driver is needed for a device to show up in the lsusb output.

I think that your problem is that you have a PCIe device - note the "e" in the name. If so, the correct driver is rtl8188ee from a kernel 3.10 or later. You can get an external version from http://github.com/lwfinger/rtlwifi_new.git.

Does the Pi have the lspci command?

gaara92, 2015-01-21:
Ah ok, I thought so. 
The lspci don't shows anythings, it says "lspci: Cannot find any working access method."
I have also tried hwinfo command to see the card but nothing... I'm affraid that it is a material problem...
On the chip, (very little but with a magnifying glass) I've read: rtl8188etv. Just to precise.
But I will try your link and the other driver. 
Anyway, thanks for your work !