lwfinger/rtl8188eu #94: Problem with hostapd (TP-LINK TL-WN725N v.2, Linux kernel 3.10.64)

jjelonek, 2015-01-25:
I've built the driver and hostapd without any problems, the AP is visible but I can't connect to it. 

**sudo service hostapd start**
 * Starting advanced IEEE 802.11 management hostapd                                                                                          ioctl[RTL_IOCTL_HOSTAPD]: Invalid argument
ioctl[RTL_IOCTL_HOSTAPD]: Invalid argument
ioctl[RTL_IOCTL_HOSTAPD]: Invalid argument
ioctl[RTL_IOCTL_HOSTAPD]: Invalid argument
ioctl[RTL_IOCTL_HOSTAPD]: Invalid argument

**sudo hostapd -d /etc/hostapd/hostapd.conf**
random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd/hostapd.conf
drv->ifindex=5
l2_sock_recv==l2_sock_xmit=0x0x54638
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
Completing interface initialization
Mode: IEEE 802.11g  Channel: 6  Frequency: 2437 MHz
RATE[0] rate=10 flags=0x1
RATE[1] rate=20 flags=0x1
RATE[2] rate=55 flags=0x1
RATE[3] rate=110 flags=0x1
RATE[4] rate=60 flags=0x0
RATE[5] rate=90 flags=0x0
RATE[6] rate=120 flags=0x0
RATE[7] rate=180 flags=0x0
RATE[8] rate=240 flags=0x0
RATE[9] rate=360 flags=0x0
RATE[10] rate=480 flags=0x0
RATE[11] rate=540 flags=0x0
Flushing old station entries
Deauthenticate all stations
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=2
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Invalid argument
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=0)
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Invalid argument
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=1)
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Invalid argument
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=2)
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Invalid argument
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=3)
Using interface wlan0 with hwaddr 14:cc:20:15:99:70 and ssid 'vtracker'
Deriving WPA PSK based on passphrase
SSID - hexdump_ascii(len=8):
     76 74 72 61 63 6b 65 72                           vtracker        
PSK (ASCII passphrase) - hexdump_ascii(len=8): [REMOVED]
PSK (from passphrase) - hexdump(len=32): [REMOVED]
rtl871x_set_wps_assoc_resp_ie
rtl871x_set_wps_beacon_ie
rtl871x_set_wps_probe_resp_ie
random: Got 20/20 bytes from /dev/random
GMK - hexdump(len=32): [REMOVED]
Key Counter - hexdump(len=32): [REMOVED]
WPA: group state machine entering state GTK_INIT (VLAN-ID 0)
GTK - hexdump(len=16): [REMOVED]
WPA: group state machine entering state SETKEYSDONE (VLAN-ID 0)
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Invalid argument
Failed to set encryption.
rtl871x_set_beacon_ops
wlan0: Setup of interface done.
+rtl871x_get_sta_wpaie, 98:0c:82:55:7b:45 is sta's address
wlan0: STA 98:0c:82:55:7b:45 IEEE 802.11: associated
STA included RSN IE in (Re)AssocReq
  New STA
wlan0: STA 98:0c:82:55:7b:45 WPA: event 1 notification
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Invalid argument
Failed to set encryption.
IEEE 802.1X: Ignore STA - 802.1X not enabled or forced for WPS
wlan0: STA 98:0c:82:55:7b:45 WPA: start authentication
WPA: 98:0c:82:55:7b:45 WPA_PTK entering state INITIALIZE
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Invalid argument
Failed to set encryption.
wlan0: STA 98:0c:82:55:7b:45 IEEE 802.1X: unauthorizing port
WPA: 98:0c:82:55:7b:45 WPA_PTK_GROUP entering state IDLE
WPA: 98:0c:82:55:7b:45 WPA_PTK entering state AUTHENTICATION
WPA: 98:0c:82:55:7b:45 WPA_PTK entering state AUTHENTICATION2
WPA: Re-initialize GMK/Counter on first station
GMK - hexdump(len=32): [REMOVED]
Key Counter - hexdump(len=32): [REMOVED]
GTK - hexdump(len=16): [REMOVED]
rtl871x_set_key_ops

lwfinger, 2015-01-26:
What branch are you using? If master, that driver cannot handle encryption. It only works in AP mode with an unencrypted network. No, I do not know why. If I did, I would fix it.

Use the 4.1.8_9499 branch. It works in encrypted mode, at least when I create an AP using NetworkManager. For this version, you should use the standard hostapd with the nl80211 driver interface for hostapd.

ajeet17181, 2015-04-07:
I encountered the same issue and found that the sizeof struct ieee_param in kernel driver is 60 where as in hostapd it is 64. if I comment out following lines in rtw_set_encryption in os_dep/ioctl_linux.c (line number 4880) file everything works fine. I am able to connect with wpa2 and with WEP too.

if (param_len !=  sizeof(struct ieee_param) + param->u.crypt.key_len) {
                ret =  -EINVAL;
                goto exit;
        }

I have used master branch of the code.

lwfinger, 2015-04-07:
If you change the size of struct ieee_param to 64, does it work without without deleting the above code? I am worried about the elimination of the check above as ignoring errors like that runs the risk of system crashes.

ajeet17181, 2015-04-07:
I haven't tried that but in hostapd I reduced param len by 4 in ioctl call
and that also worked.
Regards,
Ajeet V

lwfinger, 2015-04-07:
I found a couple of problems in the definition of rtw_ieee80211_ht_cap. Do a git pull and try with an unmodified hostapd.

Thanks for finding this problem.

ajeet17181, 2015-04-07:
it worked but rtw_ap failed to compile with following error

  CC [M]
/data/openwrt_new/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_generic/compat-wireless-2015-03-09/drivers/net/wireless/rtl8188eu/core/rtw_ap.o
In file included from
/data/openwrt_new/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_generic/compat-wireless-2015-03-09/drivers/net/wireless/rtl8188eu/include/rtw_ht.h:24:0,
                 from
/data/openwrt_new/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_generic/compat-wireless-2015-03-09/drivers/net/wireless/rtl8188eu/include/drv_types.h:36,
                 from
/data/openwrt_new/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_generic/compat-wireless-2015-03-09/drivers/net/wireless/rtl8188eu/core/rtw_ap.c:23:
/data/openwrt_new/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_generic/compat-wireless-2015-03-09/drivers/net/wireless/rtl8188eu/include/wifi.h:636:29:
error: field 'mcs' has incomplete type
   struct ieee80211_mcs_info mcs;
                             ^

I added
#include <linux/ieee80211.h>
in wifi.h and got compiled fine and tested with wpa2 it worked. I have
reverted hostapd change as well.

Thanks,
Ajeet V

lwfinger, 2015-04-07:
Thanks. The fix is applied.

For some reason, v4.1.8 built with no problem, and of course that is what I checked.

closed 2015-04-07 by lwfinger

mastykin, 2015-09-16:
Hello, I'm building for arm, kernel 3.10.
On running hostapd i see:
ioctl[RTL_IOCTL_HOSTAPD]: Invalid argument
Failed to set encryption.

Thanks,
Dmitry

lwfinger, 2015-09-16:
What version of hostapd? Which branch of the repo?

mastykin, 2015-09-18:
hostapd-0.8 and driver are from master branch. I have another adapter for which I build Realtek's 8192eu.ko - it works without this problem.

lwfinger, 2015-09-18:
Of course. That is a different device and driver. You could use Realtek's version of the driver for the RTL8188EU. Or, more simply, you should try the v4.1.8_9499 branch from this repo and the newest version of hostapd. Both will implement the cfg80211 calls.

mastykin, 2015-09-18:
unfortunately:
insmod: ERROR: could not insert module 8188eu.ko: Unknown symbol in module

lwfinger, 2015-09-18:
Did you check the output of dmesg to see what symbols are missing? Obviously, the cfg80211 version *needs* cfg80211. You need to 'sudo modprobe cfg80211' before you run the insmod. Alternatively, you can install the driver and 'modprobe 8188eu'. That handles the dependencies.

mastykin, 2015-09-23:
Thank you for help. Seems that cfg80211 was not build for this kernel. Pity that version from master branch failed.