lwfinger/rtl8188eu #95: Problem download simultaneous(TP-LINK TL-WN725N v2.0, Linux kernel 3.12.35+)

bellicheone, 2015-01-26:
Problem download simultaneous(TP-LINK TL-WN725N v2.0, Linux kernel 3.12.35+)
I use raspberry with raspbian(Debian Wheezy).
I compiled RTL8188EU driver and hostapd for create AP.
I install web server (Nginx)
I install hostapd and dnsmasq.

I have 7 ANDROID tablets for my tests
I launch donwload same file (size is 20Mo) on all tablets in same time
the download on tablets is very slow.

I test with different setting of ht_capab in hostapd.conf

1 - without ht_capab
    result : 7 download succes with 1min 50s
2 - ht_capab=[HT40-][SHORT-GI-40][DSSS_CCK-40]
    result : 6 download succes And 1 download  failt with 2min 10s
3 - ht_capab=[HT40-][SHORT-GI-40]
    result : 6 download succes And 1 download  failt with 3min 56s 
4 - ht_capab=[HT40-][DSSS_CCK-40]
    result : 7 download succes with 3min 50s 

my hostpad.conf:
interface=wlan0
#ctrl_interface=/var/run/hostapd
ssid=XXXXXXXXXXXXXX
channel=8
driver=rtl871xdrv
beacon_int=100
hw_mode=g
ieee80211n=1
wmm_enabled=1
wme_enabled=1
max_num_sta=20
#ht_capab=[HT40-][DSSS_CCK-40]

I must connect 16 tablets and download simultaneous same file. how to optimize hostapd for this ? 

sorry for my english !

lwfinger, 2015-01-26:
You may need a different host. The RPi is remarkable for the cost, but the CPU is not really fast. I would try the setup using a fast PC as the AP host.

How fast is the transfer to a single tablet? With multiple receivers that rate would need to be divided by the number of receiving tablets.

If the downloaded data can tolerate errors, the fastest way to handle many downloads simultaneously would be to multicast it to all your tablets using a single stream. If no errors can be tolerated, then each station would need to do some kind of error checking and request resending. Depending on the error rate, this could actually be slower than unicasting.

You might also try the driver in branch 4.1.8_9499. That one would require using the nl80211 driver in hostapd. I just tested it using an AP created with NetworkManager and it had an average transfer rate of about 7.5 Mbps using an 802.11g device as the station. Unfortunately, I do not have any possibility of connecting to an 802.11n  device.

bellicheone, 2015-01-28:
thank you for this information.

So I compiled the driver of the 4.1.8 branch and here are the results:
With hostapd v1.0 without ht_capab:
7 download success with 1min 22s.

I then compiled hostapd v2.3 (see http://quadfinity.blogspot.fr/2014/09/compile-latest-hostapd-v2.3-on-Raspberry-Pi-or-ODROID.html)
I only change this:
sed -i 's / ^ # CONFIG_DRIVER_NL80211 = y / CONFIG_DRIVER_NL80211 = y / g' .config

Finally, with hostapd v2.3 and ht_capab = [HT40 -] [SHORT-GI-20] [SHORT-GI-40] I got 7 download success with 1min 11s.

This result is satisfactory compared to tests with other Hostpot.

Again thank you for your help.

I have one last question. Can we add nl80211 support the driver 8188CUS / 8192CU

bellicheone, 2015-01-28:
For information, I tested dongle Edimax EW-7711UAn (https://wikidevi.com/wiki/Edimax_EW-7711UAn_v1) and result is more good
the driver rt2800usb is used and I got 7 download success with 43s.
i use hostapd v2.3

unfortunately i must use TP-LINK TL-WN725N.

lwfinger, 2015-01-28:
The Linux driver for the rt2800usb is supported by the manufacturer. The USB group at Realtek is not very responsive to issues such as this.

The nl80211 interface is present in the kernel driver rtl8192cu. If you do not want to use it, then you will have to add those commands to any out-of-kernel driver. I only support the in-kernel version.

bellicheone, 2015-01-29:
I continue my test and I have a new problem.
I use the 8188eu driver 4.1.8 branch with hostapd v2.3.

I connect 16 shelves and 1 computer to my AP. All my equipment are connected to the AP, but the 17 facilities, only 7 are functional. For others, it is impossible to communicate with the rpi.

My hostapd.conf I added the parameter max_num_sta = 255. but nothing changes.

is what the 8188eu driver 4.1.8 branch has a limitation on the maximum number of station? what should he change?

I tested with driver and the hostapd 8188eu 0.8 from the master branch, I can connect the equipment 17 but the wifi connection is unstable  and the longest response time.

lwfinger, 2015-01-29:
I did a quick check of the v4... branch to see if there are any obvious parameters that limit the number of attached stations. I did not see any, but I may have missed them. Eventially, I plan to look carefully at the differences between the two versions, but it may be some time before I do that. In fact, I am now trying to get v4... to run on big-endian platforms.

You have the sources. Perhaps you can locate a paramter to change. If you find it, please let me know.

lwfinger, 2015-01-29:
After I wrote the previous response, I remembered that the limitation would be in the nujmber of slots in the CAM. That would be set by the firmware. In the driver, parameter NUM_STA would need to match the firmware. It is set to 32, thus that is not the problem.