lwfinger/rtl8188eu #96: Compiles ok but causes kernel issue when loading on Raspberry Pi's laters kernel 3.18.5+

elpetak, 2015-02-02:
It compiled without any errors or warnings, but when I did "modprobe" I got this:

[  215.094040] Unable to handle kernel paging request at virtual address 7e20b434
[  215.105916] pgd = d3c24000
[  215.110631] [7e20b434] *pgd=00000000
[  215.116961] Internal error: Oops: 5 [#1] PREEMPT ARM
[  215.123445] Modules linked in: 8188eu(O+) snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd uio_pdrv_genirq uio
[  215.137371] CPU: 0 PID: 2242 Comm: modprobe Tainted: G           O   3.18.5+ #746
[  215.146396] task: d6254380 ti: d63e4000 task.ti: d63e4000
[  215.153312] PC is at load_module+0x1948/0x1f1c
[  215.159252] LR is at load_module+0x1934/0x1f1c
[  215.165168] pc : [<c00892b0>]    lr : [<c008929c>]    psr: 30000013
[  215.165168] sp : d63e5e88  ip : bf117304  fp : d63e5f44
[  215.179586] r10: bf11719c  r9 : 00000000  r8 : bf1171a8
[  215.186276] r7 : c05613a8  r6 : d63f02e0  r5 : 7e20b420  r4 : d63e5f48
[  215.194283] r3 : 00000000  r2 : 00000000  r1 : d73cffa0  r0 : c0824120
[  215.202315] Flags: nzCV  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  215.210964] Control: 00c5387d  Table: 13c24008  DAC: 00000015
[  215.218184] Process modprobe (pid: 2242, stack limit = 0xd63e41b0)
[  215.225886] Stack: (0xd63e5e88 to 0xd63e6000)
[  215.231776] 5e80:                   bf1171a8 00007fff c00862f4 c02fe1a8 00000013 00000000
[  215.243106] 5ea0: da04f000 d63e5f7c d63e5f50 d63e5eb8 00000000 bf1171a8 d63e4008 bf1171e4
[  215.254633] 5ec0: bf117304 00000000 b6d20000 d63e4000 00002db0 00000000 00000000 00000000
[  215.266223] 5ee0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  215.277982] 5f00: 00000000 00000000 00000000 00000000 00000000 00000000 00000080 000e98f0
[  215.289886] 5f20: b6c45000 b7d13d88 00000080 c000eb44 d63e4000 00000000 d63e5fa4 d63e5f48
[  215.301832] 5f40: c0089970 c0087974 da04f000 000e98f0 da0f04c4 da0f0356 da12e1fc 000a1e14
[  215.313875] 5f60: 000b25a4 00000000 00000000 00000000 00000023 00000024 00000018 0000001c
[  215.326139] 5f80: 0000000c 00000000 00000000 0000001e 00040000 b7d13db0 00000000 d63e5fa8
[  215.338544] 5fa0: c000e8c0 c0089890 0000001e 00040000 b6c45000 000e98f0 b7d13d88 b6c45000
[  215.351107] 5fc0: 0000001e 00040000 b7d13db0 00000080 b7d13d00 000e98f0 b7d13d88 00000000
[  215.363762] 5fe0: 00000000 bebdd40c b6f4dfb4 b6eb8ab4 60000010 b6c45000 00000000 00000000
[  215.376543] [<c00892b0>] (load_module) from [<c0089970>] (SyS_init_module+0xec/0x100)
[  215.386775] [<c0089970>] (SyS_init_module) from [<c000e8c0>] (ret_fast_syscall+0x0/0x48)
[  215.399505] Code: e51bc084 e15c0005 e2455008 0a000009 (e5953014) 
[  215.412408] ---[ end trace 63b65594807d3c31 ]---

lwfinger, 2015-02-02:
Without traceback information on that stack dump, there is little I can do. There are likely tools to interpret a kernel stack dump on ARM, but I could not find them with a quick Google search.

Does the same crash occur when you build the in-kernel driver r8188eu? Has the repo worked with earlier versions of 3.18.X, or any other kernel version?

elpetak, 2015-02-02:
in-kernel r8188eu (staging) compiles and loads ok, but it has serious stability issues.
With the rtw_power_mgnt=0 and rtw_enusbss=0 options it improves a little bit but it's pretty much unusable.
I was able to compile v4.1.8_9499 on 3.18 after fixing the select_queue warnings using your patch from commit 3b3d6fa1694516f54601b300772ec90cafb0cf3f 
but that one also crashes.
Both v4.1.8 and v4.3.0 compile on 3.12 and work ok.
v4.3.0 compiles without warnings for 3.18 but it also crashes.
I haven't tried your code on 3.12 for a while but I can try later if that helps.

lwfinger, 2015-02-02:
If 4.1.8 works OK on kernel 3.12 but crashes on 3.18, does that not argue for a kernel problem?

If you can provide the symbolic values for the traceback, I will try to help.

elpetak, 2015-02-02:
Not sure how to get the symbolic values for the traceback, but yeah, it all points to something that changed between 3.12 and 3.18.
For the time being I'll stick to 3.12 with the 4.1.8 version which has been rock solid for a long time.
After a couple of tests with 4.3.0 it showed that it wasn't that stable after all. It failed to bring up the device after a reboot and it also logged a lot which is not a nice thing for my RPI's SD card.
Feel free to close this issue, or use it as a reminder that eventually the code may need some tweaking for newer kernels.
I give up with 3.18 for the RPI. Besides, it also broke my overclocking settings... maybe there is something really wrong with it after all.
Thanks!

lwfinger, 2015-02-02:
I have run this driver with kernels up to 3.19-rc4 without any problems on x86-64. I do not usually do anything with the .X stable kernels, as those patches are in the next version. I'm pretty sure that your problem is in the ARM version. The 4.1.8 version fails on PPC due to endian issuues, as does the kernel version. The driver in the master branch is OK on a big-endian system such as PPC.

diakbar, 2015-02-04:
same here, i tried make and make install, running well. but when modprobe 8188eu I got kernel error.

[   68.541689] Unable to handle kernel paging request at virtual address 7e225420
[   68.555748] pgd = da158000
[   68.565804] [7e225420] *pgd=00000000
[   68.575488] Internal error: Oops: 5 [#1] PREEMPT ARM
[   68.586232] Modules linked in: 8188eu(O+) snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd joydev evdev uio_pdrv_genirq uio
[   68.604280] CPU: 0 PID: 2229 Comm: modprobe Tainted: G           O   3.18.5+ #744
[   68.617418] task: da28b600 ti: da214000 task.ti: da214000
[   68.628591] PC is at load_module+0x1948/0x1f1c
[   68.638800] LR is at load_module+0x1934/0x1f1c
[   68.648683] pc : [<c00892b0>]    lr : [<c008929c>]    psr: 30000013
[   68.648683] sp : da215e88  ip : bf1242f0  fp : da215f44
[   68.671063] r10: bf124188  r9 : 00000000  r8 : bf124194
[   68.681674] r7 : c05613a8  r6 : da0f98e0  r5 : 7e22540c  r4 : da215f48
[   68.693805] r3 : 00000000  r2 : 00000000  r1 : d80e2d20  r0 : c0824120
[   68.705965] Flags: nzCV  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   68.718576] Control: 00c5387d  Table: 1a158008  DAC: 00000015
[   68.729943] Process modprobe (pid: 2229, stack limit = 0xda2141b0)
[   68.741786] Stack: (0xda215e88 to 0xda216000)
[   68.751700] 5e80:                   bf124194 00007fff c00862f4 c02fe1a8 00000013 00000000
[   68.765561] 5ea0: deacd000 da215f7c da215f50 da215eb8 00000000 bf124194 da214008 bf1241d0
[   68.779480] 5ec0: bf1242f0 00000000 b6ca0000 da214000 00002db0 00000000 00000000 00000000
[   68.793420] 5ee0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   68.807348] 5f00: 00000000 00000000 00000000 00000000 00000000 00000000 00000080 000e98e0
[   68.821096] 5f20: b6bbd000 b6ece948 00000080 c000eb44 da214000 00000000 da215fa4 da215f48
[   68.834839] 5f40: c0089970 c0087974 deacd000 000e98e0 deb6e4b4 deb6e346 debac1ec 000a1dfc
[   68.848734] 5f60: 000b258c 00000000 00000000 00000000 00000023 00000024 00000018 0000001c
[   68.862690] 5f80: 0000000c 00000000 00000000 00000000 00040000 b82a1ba8 00000000 da215fa8
[   68.876652] 5fa0: c000e8c0 c0089890 00000000 00040000 b6bbd000 000e98e0 b6ece948 b6bbd000
[   68.890634] 5fc0: 00000000 00040000 b82a1ba8 00000080 b82a1c98 000e98e0 b6ece948 00000000
[   68.904488] 5fe0: 00000000 bed648fc b6ec5fb4 b6e30ab4 60000010 b6bbd000 00000000 00000000
[   68.918406] [<c00892b0>] (load_module) from [<c0089970>] (SyS_init_module+0xec/0x100)
[   68.932166] [<c0089970>] (SyS_init_module) from [<c000e8c0>] (ret_fast_syscall+0x0/0x48)
[   68.946266] Code: e51bc084 e15c0005 e2455008 0a000009 (e5953014) 
[   68.965510] ---[ end trace 77cadf959a9f1f5b ]---

lwfinger, 2015-02-04:
Without symbols, these tracebacks are useless!!

Domentarion, 2015-02-05:
I got the same problem, is there somebody who has a solution for this?

lwfinger, 2015-02-05:
Until someone converts those stack dumps to symbols, there will be no fix.

Domentarion, 2015-02-05:
how can these stack dumps be coverted to symbols?

lwfinger, 2015-02-05:
I do not know how to do it automatically, but on x86 the stack dumps have the symbolic traceback. The critical thing is to have the file System.map, which will show the locations, but that is a lot of work. It is possible that the kernel will generate the necessary info if CONFIG_DEBUG_INFO=y in .config.

Domentarion, 2015-02-06:
sorry I cant help with that, but I found a workaround for this problem. The Raspian kernel 3.18.xx has 8188eu drivers already, but the only thing you have to do is to download : 
https://github.com/lwfinger/rtl8188eu/raw/c83976d1dfb4793893158461430261562b3a5bf0/rtl8188eufw.bin
And place this file in /lib/firmware/rtlwifi/

zaazbb, 2015-02-06:
i used this bin file, under rpi-3.18.5+, when connet to wifi, met a "Association request to the driver failed" error.

elpetak, 2015-02-06:
I tried the staging driver that comes with the 3.18.xx RPI kernel and it seems to work at first but after a while it starts losing packets and then stops responding completely.

elpetak, 2015-02-06:
Maybe this can help convert the trace to symbols: http://www.hollenback.net/call2sym/call2sym.1.html
call2sym - convert Linux Kernel oops call trace listings to System.map symbolic names.

lwfinger, 2015-02-08:
I had some time to look at this problem. Actually, I got tired of doing what I should. I downloaded the latest RPi kernel (3.18.6+) and the tools so that I could build a new kernel. In the configuration, I disabled r8188eu. I then build this repo against that kernel, and added 8188eu.ko to the new kernel's module tree. The new kernel, its modules, its firmware, and an installation script are at http://www.lwfinger.com/RPI_software. Download all files to the Pi. The installation script assumes that you are logged in as user "pi". After all 4 files are in pi's home, do the following:

chmod a+x install_new_kernel.sh
sudo ./install_new_kernel.sh
sudo reboot

After the system comes up issue the commands

sudo depmod -a
sudo reboot

After the system comes up, driver 8188eu is available for configuration. This version of the driver also disables the extraneous logging.

elpetak, 2015-02-09:
I'm curious, did you have to change any code?

lwfinger, 2015-02-09:
Not a line other than to restrict the logging.

elpetak, 2015-02-09:
Not sure what I'm doing wrong then.
I decided to try again with 3.18.6 (las time I was on 3.18.5) and did this (which used to work for all the 3.12.x kernels and I have it scripted):
 - update with rpi-update
 - download kernel sources from raspberry pi repo
 - extract config from /proc/config.gz
 - make oldconfig && make modules_prepare
 - get Module.symvers from raspberry pi firmware repo matching the running kernel
 - add build and source symlinks to /lib/modules/...
 - compile the module
I also had the in-kernel module blacklisted in modprobe.d
But I still get the oops when I compile it this way.

lwfinger, 2015-02-09:
Your kernel and module are likely not compiled with the same compiler. You can check the versions by using the modinfo command on 8188eu and some other module. Usually that kind of mismatch will fail to load, but perhaps "force" is specified. In my case, the kernel and the 8188eu module were built using the same compiler.

Have you tried downloading my kernel and modules?

elpetak, 2015-02-09:
Ah, compiler may be the problem then. I'll check again later today.
About your kernel and modules, I'm sure they will work if they worked for you, but I still want to be able to compile the module myself so I don't need to bother other people next time a new kernel is released.
Thanks.

lwfinger, 2015-02-09:
OK. You need either to cross compile the kernel and module on some other platform, or do the 12 hour kernel build on the Pi.

elpetak, 2015-02-14:
Well, it was the compiler apparently.
Kernel is compiled with gcc-4.8 but I had gcc-4.6.
I updated my gcc to 4.8, compiled your code again and it worked!
I just compiled the module, not the entire kernel.
This is what I usually do (with a script) on the RPI after updating the firmware:

1) download the raspberry pi kernel source (tarball from https://github.com/raspberrypi/linux)
2) untar in /usr/src/linux-`uname -r`
3) cd /usr/src/linux-`uname -r`
    sudo bash -c "zcat /proc/config.gz > /usr/src/linux-`uname -r`/.config"
    sudo make oldconfig
    sudo make modules_prepare
    sudo make scripts
    sudo wget https://github.com/Hexxeh/rpi-firmware/raw/`cat /boot/.firmware_revision`/Module.symvers
    sudo rm /lib/modules/`uname -r`/source ; sudo ln -s /usr/src/linux-${VERSION}+ /lib/modules/`uname -r`/source
    sudo rm /lib/modules/`uname -r`/build ; sudo ln -s /usr/src/linux-${VERSION}+ /lib/modules/`uname -r`/build
4) compile the module normally

Thanks for your help and sorry for reporting an issue that wasn't such.

closed 2015-02-14 by elpetak