lwfinger/rtl8188eu #97: v4.3.0 ?

elpetak, 2015-02-02:
I think there is a much newer version out there: http://dl.lazyzhu.com/file/RTL8188EU/
RTL8188EUS_linux_v4.3.0.7_12758.20141114.zip

lwfinger, 2015-02-02:
Thanks for the link. At the moment, I have no time to look at this code.

fopina, 2016-02-24:
@elpetak thanks for the link, just compiled this driver and it works perfectly unlike version 4.1.4 and 4.1.8.

Quality/Signal shows up with `iwlist` (which was always 0 before) and wicd

elpetak, 2016-02-25:
@fopina , did you manage to get the led to blink woth traffic ?
Mine stays on all the time with this version

fopina, 2016-02-25:
@elpetak I didn't notice that actually, I was amazed with having signal quality perfectly working (which also solved the issue of wicd disconnecting after some time) and not having to use the patched version of hostapd (this version works with standard one - driver `hostap`!)...

Now that you mention, mine never lights up, it's off all the time (so the other way around).
But I haven't tried version .7, I downloaded .8..

But I haven't managed to compile this in my raspberry, only in my desktop.
For raspberry I [found](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=62371) a compiled version of this driver for my kernel and everything works there, including leds..

elpetak, 2016-03-25:
@fopina I've switched to that driver from the forums now, but I still don't get a blinking led. It just stays on. Do you have any special options set for the module?

fopina, 2016-03-25:
Only the one he packages in his tarball, to disable power management