lwfinger/rtl8188eu #98: Can not make in fedora21

skambc, 2015-02-05:
I make the driver in fedora,but not right,The follow is the error message.
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/3.17.4-301.fc21.x86_64/build M=/home/jinhd/usb/rtl8188eu  modules
make[1]: *** /lib/modules/3.17.4-301.fc21.x86_64/build: No such file or directory.  Stop.
Makefile:147: recipe for target 'modules' failed
make: *** [modules] Error 2

lwfinger, 2015-02-05:
To build an out-of-kernel module, you must have the kernel headers installed. You are missing those headers.

skambc, 2015-02-05:
I have installled the kernel headers.Wher I query the kernel-header use rpm -qa|grep kernel-headers
,it shows "kernel-headers-3.18.3-201.fc21.x86_64"
And I installed it again,but it not work yet.

sudo yum install kernel-headers
[sudo] password for liu: 
Loaded plugins: langpacks
fedora/21/x86_64/metalink                                   | 7.0 kB  00:00     
updates/21/x86_64/metalink                                  | 6.9 kB  00:00     
updates                                                     | 4.9 kB  00:00     
Package kernel-headers-3.18.3-201.fc21.x86_64 already installed and latest version
Nothing to do

skambc, 2015-02-05:
Is the kernel-headers not right?

lwfinger, 2015-02-05:
There must be a link at /lib/modules/3.17.4-301.fc21.x86_64/build that points to the location of the "include" directory that contains the headers. The error message says that that link is not correctly setup. As I do not use Fedora, I have no idea what you need to do. Check on the Fedora Forums.

unclechu, 2015-02-27:
@skambc you need to install `kernel-devel` package too.

Sekhar-jami, 2015-04-24:
@unclechu  I am facing the same issue, and I have installed both kernel-devel & kernel- header, but still the problem persists.

[nitin@localhost rtl8188eu]$ sudo make all
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/3.17.4-301.fc21.x86_64/build M=/home/nitin/rtl8188eu  modules 
make[1]: *** /lib/modules/3.17.4-301.fc21.x86_64/build: No such file or directory.  Stop.
Makefile:147: recipe for target 'modules' failed
make: *** [modules] Error 2

Is something else the problem?

unclechu, 2015-04-24:
@Sekhar-jami may be u installed kernel-devel package of other kernel version, please, show your `uname -a` and `ls -la /lib/modules/`.

And see this:
```
$ yum provides /lib/modules/3.17.4-301.fc21.x86_64/build
Загружены модули: langpacks
kernel-core-3.17.4-301.fc21.x86_64 : The Linux kernel
Источник: fedora
Совпадения с:
Имя файла   : /lib/modules/3.17.4-301.fc21.x86_64/build
```
<b>kernel-core</b> package provides this directory.

tyagiakhilesh, 2016-12-28:
Problem that happened with me, the soft link /lib/modules/<>/build was broken. When I did install headers, it appears to have installed the header for kernel version: 4.8.15-300.fc25.x86_64, whereas my uname -a shows me, that OS is using this kernel: 4.8.6-300.fc25.x86_64.

For the time being, I removed the build link, and recreated it manually. So make all succeeded.

I am on Fedora 25.

lwfinger, 2016-12-28:
That is either a Fedora problem, or there were missing packages needed to build out-of-kernel modules. In either case, the link at /lib/modules/`uname -r`/build should point to the headers for the kernel actually being run. There should be one entry for 4.8.6 and a different one for 4.8.15.