lwfinger/rtl8188eu #99: Cross compilation for ARM target (kernel 3.6.3)

ssinfod, 2015-02-09:
Hello, 
my goal is to build the rtl8188eu module for an embedded ARM target.
The target is the phidget SBC3 board (CPU: Freescale i.mx28).
http://www.phidgets.com/products.php?product_id=1073

It is running a custom kernel (3.6.3) from a debian distro.

I have some knowledge about linux but I don't know much about driver.

First, I thought to compile the 8188eu driver directly on the host but it does not have the kernel-headers because the kernel is custom. (ie: patch RT_PREEMPT and some other customization)

The people from phidget told me it would be better to cross compile it.
Therefore, I installed the gcc arm compiler (arm-linux-gnueabi).

I also downloaded the kernel 3.6.3 from kernel.org and apply the phidget patch to it.
Then, I tried to compile the 8188eu driver and I have the following output:
 Building modules, stage 2.
 MODPOST 0 modules

I don't understand why I have this message. I tried to change the configuration of the 3.6.3 kernel but I don't see the 8188eu driver under "Device Driver"  (network) nor "Staging Driver".

Here are my quesitons:
Q1) Do I need a more recent kernel to cross compile the 8188eu driver ?
Q2) Can I compile the driver with another kernel (more recent) and use it with my older 3.6.3 kernel ?
Q3) Can I use directly the binary file 8188eu.ko from another person. (does It have to be compiled for the exact same kernel?)
Q4) Does the kernel needs to have the driver selected as a module (M) to use it ? 
Does the kernel needs to have the driver selected as a module to build it ? (It is not clear to me why I must change the kernel configuration to build the module)

I found some good reference on the following linux-sunxi website. I was able to follow these instructions and cross-compile the driver for the sunxi (allwinner) target. However, it is not working for my target. 

Any help with my issue would be very appreciated.
Thanks,
ssinfod

lwfinger, 2015-02-09:
A1. No. Kernel 3.6.3 will be OK.
A2. No. All drivers must be compiled using the save kernel.
A3. Possibly. It will work as long as the other person has the same kernel and compiler versions.
A4. Kernel 3.6.3 does not have a driver for the RTL8188EU. It was added in 3.12. Thus you will need the out-of-kernel driver built as a module.

The script that I use to build this driver for a Raspberry Pi (also ARM) is

export ARCH=arm
export CROSS_COMPILE=/home/finger/tools/arm-bcm2708/arm-bcm2708-linux-gnueabi/bin/arm-bcm2708-linux-gnueabi-
make KSRC=/home/finger/raspberry/

The cross compile symbol should point to the location of the cross-compiler, and KSRC should point to the location of the kernel source.

ssinfod, 2015-02-10:
Thanks, the cross-compilation worked !
It seems I was only missing the KSRC variable. 

It would be great if the driver exists in 32 bits or 64 bits flavor for x86, ARM and other architecture.
I don't understand all the internals but I find it strange that linux drivers are incompatible between kernel versions. What is the reason behind this ?

I don't mind to do this kind of operations but that is less accessible for general public.

Anyway, thanks a lot for your help.
ssinfod

lwfinger, 2015-02-10:
Windows keeps the same API (applications interface) for an entire release (XP, Vista, etc.). Linux is much more dynamic and the API can change from version to version. You only need to wait about 3 months for a new version, not 2-3 years! Because the API is changing, you must build a driver using the headers for the kernel version you want to use. The configuration file contains the platform details (32- or 64-bit, architecture, etc., and the headers describe the API for that version. As the entire source is *always* available, that is no problem. The general public probably does not run special drivers on any platform.