lwfinger/rtl8188eu #100: Which one should I use? Master ot 4.1.8 branch?

elpetak, 2015-02-14:
It was a nice surprise to see a 4.1.8 branch. I was using the original 4.1.8 sources for my RPI.
I've also notices a few commits on master that say something about porting the 4.1.8 updates to master.
But then I saw the 4.1.8 branch has more recent activity.
So which one should a normal user compile for day to day usage? master or 4.1.8?
Thanks.

lwfinger, 2015-02-14:
Use the one that works. :)

Master works with a big-endian platform, whereas neither the kernel nor 4.1.8 branches work on PowerPC, which is my BE test machine. Of course, that does not affect the PI. Version 4.1.8 has the CFG80211 ioctl commands, thus it will work with new versions of hostapd. Some users report that it also has better sensitivity - I do not know if that is true.

elpetak, 2015-02-15:
Excelent advise! Thanks!
I did try to compile the 4.1.8 branch from here but I got an invalid pointer error somewhere in the cfg80211 part. My other copy of 4.1.8 which apparently is realtek's original code compiled just fine after I changed the rtw_select_queue code for new kernels.
Sorry I don't have more details right now. I'll try to compile it again tomorrow and will share the details.

elpetak, 2015-02-15:
Here is the warning I get compiling 4.1.8 branch against raspberry pi's 3.18.7 kernel sources:

  CC [M]  /home/pi/8188eu/lwfinger/v4.1.8/os_dep/ioctl_cfg80211.o
/home/pi/8188eu/lwfinger/v4.1.8/os_dep/ioctl_cfg80211.c:5338:2: warning: initialization from incompatible pointer type [enabled by default]
  .del_station = cfg80211_rtw_del_station,
  ^
/home/pi/8188eu/lwfinger/v4.1.8/os_dep/ioctl_cfg80211.c:5338:2: warning: (near initialization for ‘rtw_cfg80211_ops.del_station’) [enabled by default]

lwfinger, 2015-02-15:
That warning is due to a missing "const" attribute for one of the arguments. On x86, that difference is insignificant as far as operations are concerned. Perhaps that is not true for the Pi/arm. It is fixed now.

elpetak, 2015-03-23:
After a while without messing with it I decided to try again your 4.1.8 branch and I have to say I like it.
It compiles without warnings and it loads perfectly fine and I avoid having to manually patch my copy of the original 4.1.8 version to work with newer kernels because you do it already. Thanks!

One question though: why did you remove all the LED definitions from autoconf.h ?
I tried to add the LED part back and got some weird error when loading the module (maybe I should reboot between modules).
Should it work if I add these or are there different options?
#define CONFIG_LED
#define CONFIG_SW_LED
#define CONFIG_LED_HANDLED_BY_CMD_THREAD

elpetak, 2015-03-23:
Found the commit: 7f426e6764abfb2ea6ec5aefecab6caf618c82fc
why was LED removed?

lwfinger, 2015-03-23:
I have been removing the dead code for configuration variables that are not selected. That was one of those.

I have no idea what it will take to make the LEDs work. For me, that is low priority, but I will accept any patches. As I recall, LEDs never worked with this version.

lwfinger, 2015-03-23:
I just found how to make the LED work. BTW, it has nothing to do with CONFIG_LED_HANDLED_BY_CMD_THREAD . In fact, very little is handled by that kind of thread in Linux.

elpetak, 2015-03-25:
Great! I'll test it tonight when I get home. Thanks!

elpetak, 2015-03-26:
Confirmed led works.
I can say my original question is now answered: I'll stick to 4.1.8 either from here or the realtek one.

closed 2015-03-26 by elpetak