lwfinger/rtl8188eu #101: How to detect driver version I am using?

techtonik, 2015-02-16:
Got a whole lot of question about driver management. The reason why I need it is the same problem as in #76. How to detect if I am using driver that comes with kernel or driver that is custom compiled from this repository (I don't remember if I previously installed driver from sources on this system)?

If I compile driver now, won't it override future kernel version? How can I detect when new kernel comes with newer version of the driver? Thanks.

lwfinger, 2015-02-16:
This repo was cleverly designed to provide a *different* name than the kernel. If 'lsmod | grep 8188eu' returns "r8188eu", you have the kernel version. If it returns "8188eu", you have this one.

When you install a new kernel from your distro, you wipeout any external driver. That also applies to reinstalling the same kernel.

techtonik, 2015-02-16:
Cool, I wish it was a link to FAQ somewhere. =) `lsmod | grep 8188` shows `r8188eu`.

techtonik, 2015-02-16:
The only question left - if I compile driver from this repository, will it conflict with current kernel driver?

lwfinger, 2015-02-16:
Possibly. If you install the driver from this repo, you should blacklist the kernel version.

closed 2015-02-16 by lwfinger

techtonik, 2015-02-16:
It looks like the kernel driver is missing on Fedora. I don't get why - the kernel version is higher than on Ubuntu, where kernel driver is present.

lwfinger, 2015-02-16:
Not all distros configure staging drivers. It has nothing to do with kernel version.