lwfinger/rtl8188eu #102: 8188eu: disagrees about version of symbol module_layout

wyatt-wong, 2015-02-19:
I am using Raspberry Pi 2 and the linux kernel version is 3.18.7-v7+. After compiled the rtl8188eu and installed it, I rebooted my Raspberry Pi 2 and I typed the dmesg and got the following error:

8188eu: disagrees about version of symbol module_layout

How do I resolve it ?

lwfinger, 2015-02-19:
Compile the module with *exactly* the same kernel headers, kernel configuration, and compiler version. That is the only way that an externally built module will load. In your case, it is likely the compiler version that is messing you up. The command 'cat /proc/version' will show the particulars for the running kernel.

wyatt-wong, 2015-02-20:
How do I ensure the same kernel headers, kernel configuration and compiler version was used to compile rtl8188eu ? I found there are no linux headers for my kernel version with the following command:

sudo apt-get install linux-headers-$(uname -r)

I use sudo make headers_install to install the kernel headers and also grab the .config file from /proc/config.gz and rebuild the rtl8188eu for a few times but I still got the above issue. It is very frustrating.

lwfinger, 2015-02-20:
What does 'cat /proc/version' say?

What does 'gcc --version' say?

wyatt-wong, 2015-02-20:
cat /proc/version
Linux version 3.18.7-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #755 SMP PREEMPT Thu Feb 12 17:20:48 GMT 2015

gcc (Debian 4.6.3-14+rpi1) 4.6.3
Copyright (C) 2011 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

lwfinger, 2015-02-20:
The kernel was compiled with gcc version 4.8.3 20140303 .

You are compiling the module with gcc 4.6.3. That is not the same version!

wyatt-wong, 2015-02-20:
I know but I don't know how to fix it. I found that I got gcc 4.7.2 and I had updated the symbolic link of gcc to point to gcc-4.7

lwfinger, 2015-02-20:
The kernel was cross compiled on some other platform. You have three choices:

(1) Build the kernel on the Pi. That takes ~12 hours.
(2) Set up the same tool chain on a PC and build the module there.
(3) Build gcc 4.8.3 on the Pi. That will take a very long time.

Option 2 takes a bit of setup, but there are plenty of examples on the web. I would not recommend the others, particularly as you are not experienced.

wyatt-wong, 2015-02-20:
I have already rebuilt the linux kernel but I have no gcc-4.8.3 installed and so I don't know why the rebuilt linux kernel still show gcc-4.8.3

I rolled back to the original Raspbian linux kernel image and found that it was using gcc-4.8.3 too.

I would like to rebuild gcc-4.8.3 but I can only download gcc-4.8 from github and have no idea on how to get the gcc-4.8.3 source. Moreover, I can't find any sources from the web on building gcc in Raspberry Pi.

wyatt-wong, 2015-02-22:
I have updated to gcc-4.8.3 and re-compiled the rtl8188eu, however after installed the 8188eu.ko module, I got the "Invalid module format" when I type the "insmod" command:

pi@raspberrypi ~/Downloads/rtl8188eu $ gcc --version
gcc (Raspbian 4.8.3-13) 4.8.3
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

pi@raspberrypi /lib/modules/3.18.7-v7+/kernel/drivers/net/wireless $ ls -l 8188eu.ko
-rw-r--r-- 1 root root 1062933 Feb 22 19:07 8188eu.ko

pi@raspberrypi /lib/modules/3.18.7-v7+/kernel/drivers/net/wireless $ sudo insmod 8188eu.ko 
Error: could not insert module 8188eu.ko: Invalid module format

dmitryzakharov1, 2015-02-23:
wyatt-wong, you are not alone with problem

dmitryzakharov1, 2015-02-23:
Right now I'm rebuilding 4.18.7-v7+ kernel with upgraded gcc 4.8.3 on rpi 2. After that i build driver from source. I hope this will help us. If it help i upload kernel and driver to rpi forum.

extralucide, 2015-02-23:
Hi,
I'm a new user of raspberry pi 2 and I am also stuck with the same issue than wyatt-wong
 (gcc version 4.8.3 20140303 versus gcc 4.6.3).
I really look forward to your build, dmitryzakharov1 !

dmitryzakharov1, 2015-02-23:
extralucide, still rebuilding.

dmitryzakharov1, 2015-02-23:
grab a beer brothers. I'v got wlan0

dmitryzakharov1, 2015-02-23:
https://www.dropbox.com/s/fcnx8u7opzddjmo/rtl8188eu.tar.bz2?dl=0
/boot with kernel
and ko modules

wyatt-wong, 2015-02-23:
Thanks, Dmitry. I updated rtl8188eufw.bin to /lib/firmware/rtlwifi/ and 8188eu.ko to /lib/modules/3.18.7-v7+/kernel/drivers/net/wireless. After executed sudo insmod 8188eu.ko, I was able to see "wlan0" in ifconfig.

I will configure the wlan setting later then.

wyatt-wong, 2015-02-23:
I was able to use wifi in my RPi2 using the 8188eu.ko with my linux kernel 3.18.7-v7+. I had also configured RPi2 to auto startup the wlan0 after reboot. However, when I issue the "sudo ifup wlan0" command. I got the following 3 errors on isoctl:

Do you know why there is "Operation not permitted" and "Invalid argument" errors ?

==============================
pi@raspberrypi ~ $ sudo ifup wlan0
ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
Internet Systems Consortium DHCP Client 4.2.2
Copyright 2004-2011 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/14:cf:92:d3:0b:69
Sending on   LPF/wlan0/14:cf:92:d3:0b:69
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 5
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 8
DHCPREQUEST on wlan0 to 255.255.255.255 port 67
DHCPOFFER from 192.168.1.1
DHCPACK from 192.168.1.1
bound to 192.168.1.119 -- renewal in 36120 seconds.
==============================

lwfinger, 2015-02-23:
Which branch of the driver did you build? I realize that it is my fault for having more than one branch, but it would be helpful if people would automatically include this information.

Is ifup a script on your distro? If so, what are the commands being used in that script. The driver may not be correctly implementing all the WEXT compatibility calls.

wyatt-wong, 2015-02-24:
Refer to the reply thread above, the 8188eu.ko was not compiled by myself but it was compiled and provided by the user Dmitry Zakharov. So I have no idea what branch of the drive that he used to build the rtl8188eu binary.

The ifup command is bundled in my Raspberry Pi 2 Linux distribution. I need to check the contents of the ifup script and get back to you later.

wyatt-wong, 2015-02-24:
Do I need to use your kernel.img to boot my my RPi2 as well, Dmitry ?

dmitryzakharov1, 2015-02-24:
wyatt-wong, yes you need. Right now it is not needed. 
http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=62371&sid=085c38cc434972948204d7c741c32b8a&start=550
MrEngman build file with drivers for 3.18.7-v7+ kernel

Rajssss, 2017-07-05:
hi... im also having the same problem..here what i got .....

root@...:~/Desktop/rtl8188eu-master# cat /proc/version
Linux version 4.9.0-kali3-686-pae (devel@kali.org) (gcc version 6.3.0 20170321 (Debian 6.3.0-11) ) #1 SMP Debian 4.9.18-1kali1 (2017-04-04)
root@...:~/Desktop/rtl8188eu-master# gcc --version
gcc (Debian 6.3.0-12) 6.3.0 20170406
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

i think i have same versions but still insmod gives error.. please help me i am a beginner in kali linux...

wyatt-wong, 2017-07-11:
Rajssss

I bought Raspberry Pi official WiFi dongle and it works perfectly for Raspberry Pi 2, so I did not use rtl8188eu module anymore.