lwfinger/rtl8188eu #104: 8188eu on raspberry pi (kernel 3.12.13+)

FishTest, 2015-03-01:
When i executed:
pi@raspberrypi ~ $ sudo ifconfig wlan0 up
SIOCSIFFLAGS: Operation not permitted

pi@raspberrypi ~ $ dmesg
[ 1782.730609] Request firmware failed with error 0xfffffffe

pi@raspberrypi ~ $ lshw -C network
WARNING: you should run this program as super-user.
  *-network:0 DISABLED
       description: Wireless interface
       physical id: 1
       bus info: usb@1:1.1
       logical name: wlan0
       serial: 08:57:00:a2:17:13
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=r8188eu ip=192.168.1.10 multicast=yes wireless=unassociated

pi@raspberrypi ~ $ lsmod
Module                  Size  Used by
8188eu                736583  0
joydev                  9085  0
rp_usbdisplay          13452  3
asix                   18908  0
libphy                 26688  1 asix
leds_gpio               2114  0
led_class               4199  1 leds_gpio
evdev                  10434  4

pi@raspberrypi ~ $ lsusb
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. 4-Port HUB
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0bda:8179 Realtek Semiconductor Corp.
Bus 001 Device 004: ID 0b95:772b ASIX Electronics Corp.

How to solve this problem?

FishTest, 2015-03-01:
and how to set 8188eu work in ap mode when modprobe?

lwfinger, 2015-03-01:
To solve the first problem, install the firmware. If you had used 'sudo make install', that step would have been done for you.

To use it as an AP, the master branch needs to use version 0.8 of hostapd. If you switch to branch v4.1.8_9499, then you can use a new version of hostapd.

FishTest, 2015-03-08:
Thanks! i've rebuild rpi linux kernel on rpi and rebuild your 8188eu driver,and it works.

closed 2015-03-08 by FishTest