lwfinger/rtl8188eu #105: Ubuntu 14.04 not using correct drivers?

P3tras, 2015-03-09:
Hello,

First of all, I wanted to thank LWfinger for great job maintaining the drivers. Everything worked great so far (since last kernel update).

While using Ubuntu I encountered 2 drivers that were used by the system
* Stock staging drivers in the kernel. It shows low connection strength, often has very slow connection and usually in 15 minutes losts connection. Green LED indicator works on wifi dongle.
* Lwfinger drivers. Shows full connection strength, has good speed and never drops connection. However, green LED indicator on dongle remains off.

After each kernel update I need to recompile lwfinger drivers and I replace driver in kernel folder. After computer restart I enjoy wonders of wifi. However, on 3.13.0-45 kernel update I did not have to replace wifi drivers file and it worked right after compiling (make install). But after 3.13.0-46 kernel update wifi did not work at all. In "DMESG" window I noticed error: 8188eu: disagrees about version of symbol module_layout (seems like update utility did not update driver).

So I recompiled newest version of Lwfinger drivers (master branch) and replaced system file as I always do. Wifi started working, but it seems that system uses staging drivers (dropping conection, LED indicator works).

I checked which drivers are used with "sudo lshw -C network" and it shows R8188eu drivers. Modinfo shows where drivers are located and drivers attributes. Modinfo report did change when I swapped stock driver to lwfinger one, but performance remained exactly the same. I suspect that system still uses stock staging drivers. Maybe someone has an idea how to track if it is the case or force system to use lwfinger drivers.

Thanks in advance for any help,

P.S. Using Ubuntu 14.04 32 bit version with all newest updates

lwfinger, 2015-03-09:
You will always need to rebuild the driver from this repo *every* time you change the kernel. That is a must.

I carefully named this driver as "8188eu", whereas the kernel version is "r8188eu". If you want to prevent loading the kernel version, then blacklist r8188eu. I leave finding how to do that as an exercise. If these drivers are installed, and the kernel version is not blacklisted, then it will be a race to see which one is controlling the device.

P3tras, 2015-03-09:
Yes, I always download newest drivers and recompile them. I know that your drivers are named 8188eu. I used to renamed .ko file to r8188eu and replace stock driver file with it. Did not think about blacklisting, thanks! I hope it will work.

P3tras, 2015-03-12:
Sorry, but it still does not work. It seems that system was using 8188eu.ko drivers (not stock staging ones). I blaclisted r8188eu drivers and there was no difference. I tried blackilsting 8188eu and wifi completely dissapeared (no matter if r8188eu drivers were blacklisted or not). Noticed that system was using 8188eu.ko driver installed by DKMS, thus I deleted it and reinstalled your driver via terminal. Now modinfo gives correct path to drivers:
petras@petras-HP-d530-SFF-DK907A:~/Atsiuntimai/rtl8188eu-4.1.8_9499$ modinfo 8188eu
filename:       /lib/modules/3.13.0-46-generic/kernel/drivers/net/wireless/8188eu.ko
version:        v4.1.4_6773.20130222

But performance is still the same. Is it possible that new kernel update is causing these issues, as it worked great before? I still suspect something with drivers as LED indicator never blinked while using your drivers, but now it works normally.

P.S. I tried to compile 4.1.8 drivers, but it always fails:
petras@petras-HP-d530-SFF-DK907A:~/Atsiuntimai/rtl8188eu-4.1.8_9499$ make all
make ARCH=i386 CROSS_COMPILE= -C /lib/modules/3.13.0-46-generic/build M=/home/petras/Atsiuntimai/rtl8188eu-4.1.8_9499  modules
make[1]: Entering directory `/usr/src/linux-headers-3.13.0-46-generic'
  CC [M]  /home/petras/Atsiuntimai/rtl8188eu-4.1.8_9499/os_dep/ioctl_cfg80211.o
/home/petras/Atsiuntimai/rtl8188eu-4.1.8_9499/os_dep/ioctl_cfg80211.c:4491:2: warning: ‘struct cfg80211_mgmt_tx_params’ declared inside parameter list [enabled by default]
  u64 *cookie)
  ^
/home/petras/Atsiuntimai/rtl8188eu-4.1.8_9499/os_dep/ioctl_cfg80211.c:4491:2: warning: its scope is only this definition or declaration, which is probably not what you want [enabled by default]
/home/petras/Atsiuntimai/rtl8188eu-4.1.8_9499/os_dep/ioctl_cfg80211.c: In function ‘cfg80211_rtw_mgmt_tx’:
/home/petras/Atsiuntimai/rtl8188eu-4.1.8_9499/os_dep/ioctl_cfg80211.c:4530:41: error: dereferencing pointer to incomplete type
  struct ieee80211_channel *chan = params->chan;
                                         ^
/home/petras/Atsiuntimai/rtl8188eu-4.1.8_9499/os_dep/ioctl_cfg80211.c:4531:24: error: dereferencing pointer to incomplete type
  const u8 *buf = params->buf;
                        ^
/home/petras/Atsiuntimai/rtl8188eu-4.1.8_9499/os_dep/ioctl_cfg80211.c:4532:21: error: dereferencing pointer to incomplete type
  size_t len = params->len;
                     ^
/home/petras/Atsiuntimai/rtl8188eu-4.1.8_9499/os_dep/ioctl_cfg80211.c: At top level:
/home/petras/Atsiuntimai/rtl8188eu-4.1.8_9499/os_dep/ioctl_cfg80211.c:5054:2: warning: initialization from incompatible pointer type [enabled by default]
  .mgmt_tx = cfg80211_rtw_mgmt_tx,
  ^
/home/petras/Atsiuntimai/rtl8188eu-4.1.8_9499/os_dep/ioctl_cfg80211.c:5054:2: warning: (near initialization for ‘rtw_cfg80211_ops.mgmt_tx’) [enabled by default]
make[2]: *** [/home/petras/Atsiuntimai/rtl8188eu-4.1.8_9499/os_dep/ioctl_cfg80211.o] Error 1
make[1]: *** [_module_/home/petras/Atsiuntimai/rtl8188eu-4.1.8_9499] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-3.13.0-46-generic'
make: *** [modules] Error 2

P3tras, 2015-03-12:
By the way, booted system with older 3.13.0-45 kernel and everything works great. LED indicator on wifi dongle is off, shows full connection strength, connection does not drop out. May stick with older kernel :)

P3tras, 2015-03-24:
Figured out that master branch version does not work well for me (connection drops after some time), but 4.1.8_9499 branch works great. But last kernel that it compiled with was 3.13.0-45. It does not compile with 3.13.0.46 and now with 3.13.0.48. Probably ubuntu made some changes to the kernel (however, master branch compiles fine). Tried using newest and some old versions from 4.1.8_9499 branch, but they all give the same error:

/home/petras/Atsiuntimai/rtl8188eu-5b8f5bc69d6a874c32257249ab0c97ce3a075665/os_dep/ioctl_cfg80211.c:4792:2: warning: ‘struct cfg80211_mgmt_tx_params’ declared inside parameter list [enabled by default]
  u64 *cookie)
  ^
/home/petras/Atsiuntimai/rtl8188eu-5b8f5bc69d6a874c32257249ab0c97ce3a075665/os_dep/ioctl_cfg80211.c:4792:2: warning: its scope is only this definition or declaration, which is probably not what you want [enabled by default]
/home/petras/Atsiuntimai/rtl8188eu-5b8f5bc69d6a874c32257249ab0c97ce3a075665/os_dep/ioctl_cfg80211.c: In function ‘cfg80211_rtw_mgmt_tx’:
/home/petras/Atsiuntimai/rtl8188eu-5b8f5bc69d6a874c32257249ab0c97ce3a075665/os_dep/ioctl_cfg80211.c:4831:41: error: dereferencing pointer to incomplete type
  struct ieee80211_channel *chan = params->chan;
                                         ^
/home/petras/Atsiuntimai/rtl8188eu-5b8f5bc69d6a874c32257249ab0c97ce3a075665/os_dep/ioctl_cfg80211.c:4832:24: error: dereferencing pointer to incomplete type
  const u8 *buf = params->buf;
                        ^
/home/petras/Atsiuntimai/rtl8188eu-5b8f5bc69d6a874c32257249ab0c97ce3a075665/os_dep/ioctl_cfg80211.c:4833:21: error: dereferencing pointer to incomplete type
  size_t len = params->len;
                     ^
/home/petras/Atsiuntimai/rtl8188eu-5b8f5bc69d6a874c32257249ab0c97ce3a075665/os_dep/ioctl_cfg80211.c: At top level:
/home/petras/Atsiuntimai/rtl8188eu-5b8f5bc69d6a874c32257249ab0c97ce3a075665/os_dep/ioctl_cfg80211.c:5355:2: warning: initialization from incompatible pointer type [enabled by default]
  .mgmt_tx = cfg80211_rtw_mgmt_tx,
  ^
/home/petras/Atsiuntimai/rtl8188eu-5b8f5bc69d6a874c32257249ab0c97ce3a075665/os_dep/ioctl_cfg80211.c:5355:2: warning: (near initialization for ‘rtw_cfg80211_ops.mgmt_tx’) [enabled by default]
make[2]: *** [/home/petras/Atsiuntimai/rtl8188eu-5b8f5bc69d6a874c32257249ab0c97ce3a075665/os_dep/ioctl_cfg80211.o] Error 1
make[1]: *** [_module_/home/petras/Atsiuntimai/rtl8188eu-5b8f5bc69d6a874c32257249ab0c97ce3a075665] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-3.13.0-48-generic'
make: *** [modules] Error 2

Sticking with 3.13.0-45 kernel for now :/

lwfinger, 2015-03-24:
That is good advice as I have given up on trying to follow Ubuntu's improper changing of the API within a given kernel. Obviously, they are actively preventing users from building out-of-kernel drivers. If I were one of their users, I would complain loudly!

P3tras, 2015-03-24:
Just to clarify, drivers in master branch are v4.1.4_6773.20130222 and in 4.1.8 branch are v4.1.8_9499.20131104 , correct?

lwfinger, 2015-03-24:
Whatever is says in file include/rtw_version.h is correct. For 4.1.8, the version above is what we have. It is not convenient for me to switch to master at the moment.

hs35912, 2015-08-30:
I am suffering from the same issue as the first post suggest here. 
Yes I have compiled the module from the master branch without any error.
-------
uname -a
Linux ProBook 3.19.0-26-generic #28~14.04.1-Ubuntu SMP Wed Aug 12 14:09:17 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

------
lshw -C network
*-network:1
       description: Wireless interface
       physical id: 2
       bus info: usb@1:1.4
       logical name: wlan6
       serial: 00:e0:4c:81:92:e6
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=r8188eu multicast=yes wireless=unassociated
-------
lsmod | grep 8188eu
8188eu                741376  0 
-------
root@ProBook:~/tmp/rtl8188eu# modinfo r8188eu | grep file
filename:       /lib/modules/3.19.0-26-generic/kernel/drivers/staging/rtl8188eu/r8188eu.ko
root@ProBook:~/tmp/rtl8188eu# modinfo 8188eu | grep file
filename:       /lib/modules/3.19.0-26-generic/kernel/drivers/net/wireless/8188eu.ko
------
I have blacklisted the staging driver and also did fakeinstall:
root@ProBook:/etc/modprobe.d# cat 50-8188eu.conf 
blacklist r8188eu
install r8188eu /bin/true
------
I wonder what else I can do to ensure that kernel uses 8188eu and NOT r8188eu.

Please HELP.

lwfinger, 2015-08-30:
Blacklist r8188eu.

hs35912, 2015-08-31:
Hey thanks... I believe I already have done that here:

root@ProBook:/etc/modprobe.d# cat 50-8188eu.conf 
blacklist r8188eu
install r8188eu /bin/true

lwfinger, 2015-08-31:
You should need only the first line. Have you tried that? Then reboot. After you plug in the device, do 'lsmod | grep 8188'. If you think it is wrong, plost that output.

hs35912, 2015-08-31:
Thanks for you help. It WORKS. The issue was not compiling, rather with the Network Manager.

I now have 2 questions:

1. why does the output of  'lshw -C network' tells me that it's using driver=r8188eu. Here is the relevant output:

*-network:1
       description: Wireless interface
       physical id: 2
       bus info: usb@1:1.4
       logical name: wlan6
       serial: 00:e0:4c:81:92:e6
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=r8188eu ip=10.10.10.3 multicast=yes wireless=IEEE 802.11bgn

Yes this driver is blacklisted now in /etc/modprobe.d folder.

2. iw dev does not show this device:
root@ProBook:~# iw dev
phy#0
	Interface wlan0
		ifindex 3
		type managed
this device is wlan6

Thank you once again.

hs35912, 2015-08-31:
Also apologies as I think I am asking far too many questions.
I could not get this card to monitor mode:
________________________________
root@ProBook:~# airmon-ng start wlan6

PHY	Interface	Driver		Chipset

phy0	wlan0		ath9k		Qualcomm Atheros AR9285 Wireless Network Adapter (PCI-Express) (rev 01)
null	wlan6		r8188eu		Realtek Semiconductor Corp. 

For some reason airmon-ng also tells me the driver is r8188eu. Funny that.

lwfinger, 2015-08-31:
Internally, the driver calls itself r8188eu; however, the make file creates 8188eu. You should ignore those user tools. If there is a question as to which driver is loaded, use lsmod.

The driver does not support monitor mode. If you want that feature, you would need to add it yourself. If you do, I would welcome patches.

hs35912, 2015-08-31:
Thank you for this clear instruction. 
Appreciate it.