lwfinger/rtl8188eu #106: R8188EU: ERROR indicate disassoc

nivw, 2015-03-10:
I see this is similar https://github.com/lwfinger/rtl8188eu/issues/35
I am using beaglebone black with debian 8.0 (jessie).
My kernel is Linux beaglebone3 3.19.0-rc7-bone3 #1 Tue Feb 3 07:24:29 UTC 2015 armv7l
I first try the staging driver in kernel  3.19.0-rc7. I got this error. Then downloaded the master zip file from this site at Mar 9 (see my hashes below).
I removed the file: /lib/firmware/rtlwifi/rtl8188eufw.bin
I build the module, then make install.

I am using the connman-1.28 and wpa_supplicant-2.3
I try to enable wifi direct in connmanctl using
```
connmanctl> enable p2p
Error p2p: Method "SetProperty" with signature "sv" on interface "net.connman.Technology" doesn't exist
```
 Then I get the errors in syslog, and wifi stops working.

I still get these errors:
```
Mar 10 11:25:56 beaglebone3 kernel: R8188EU: ERROR indicate disassoc
Mar 10 11:25:56 beaglebone3 kernel: +rtw_indicate_disconnect
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000010,5]
Mar 10 11:25:56 beaglebone3 kernel: R8188EU: HW_VAR_BASIC_RATE: BrateCfg(0x15f)
Mar 10 11:25:56 beaglebone3 kernel: R8188EU: update_mgnt_tx_rate(): rate = 2
Mar 10 11:25:56 beaglebone3 kernel: recvbuf2recvframe: rxdesc=offsset 0:0x0000003c, 4:0x00000000, 8:0x00000000, C:0x00000000
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000004,8]
Mar 10 11:25:56 beaglebone3 kernel: usb_read_port_complete!!!
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x01000000,5]
Mar 10 11:25:56 beaglebone3 kernel: +rtw_disassoc_cmd
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000080,7]
Mar 10 11:25:56 beaglebone3 kernel: R8188EU: (2)88eu_drv - drv_close, bup =1, hw_init_completed =1
Mar 10 11:25:56 beaglebone3 kernel: +88eu_drv - drv_close
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000800,8]
Mar 10 11:25:56 beaglebone3 kernel: mlme_cmd_hdl(): pcmd_callback = 0x  (null), cmdcode = 0x38
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000080,8]
Mar 10 11:25:56 beaglebone3 kernel: _rtw_alloc_network: ptr=e0e70be0
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000010,8]
Mar 10 11:25:56 beaglebone3 kernel: rtw_survey_event_callback, ssid=sasha_2.2
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000010,8]
Mar 10 11:25:56 beaglebone3 kernel: rtw_free_xmitframe():free_xmitframe_cnt=256
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000001,9]
Mar 10 11:25:56 beaglebone3 kernel: rtw_write_port, w_sz=74
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000001,8]
Mar 10 11:25:56 beaglebone3 kernel: -usb_write_port
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x01000000,5]
Mar 10 11:25:56 beaglebone3 kernel: +usb_write_port
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x01000000,5]
Mar 10 11:25:56 beaglebone3 kernel: rtw_dump_xframe()
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000001,8]
Mar 10 11:25:56 beaglebone3 kernel: issuing probe_req, tx_len=42
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000010,7]
Mar 10 11:25:56 beaglebone3 kernel: rtw_alloc_xmitframe():free_xmitframe_cnt=255
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000001,8]
Mar 10 11:25:56 beaglebone3 kernel: +issue_probereq
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000010,7]
Mar 10 11:25:56 beaglebone3 kernel: rtw_free_xmitframe():free_xmitframe_cnt=256
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000001,9]
Mar 10 11:25:56 beaglebone3 kernel: rtw_write_port, w_sz=74
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000001,8]
Mar 10 11:25:56 beaglebone3 kernel: -usb_write_port
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x01000000,5]
Mar 10 11:25:56 beaglebone3 kernel: +usb_write_port
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x01000000,5]
Mar 10 11:25:56 beaglebone3 kernel: rtw_dump_xframe()
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000001,8]
Mar 10 11:25:56 beaglebone3 kernel: issuing probe_req, tx_len=42
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000010,7]
Mar 10 11:25:56 beaglebone3 kernel: rtw_alloc_xmitframe():free_xmitframe_cnt=255
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000001,8]
Mar 10 11:25:56 beaglebone3 kernel: +issue_probereq
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000010,7]
Mar 10 11:25:56 beaglebone3 kernel: rtw_free_xmitframe():free_xmitframe_cnt=256
[... 81 lines cut ...]
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x02000000,8]
Mar 10 11:25:56 beaglebone3 kernel: [62B blob data]
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x02000000,8]
Mar 10 11:25:56 beaglebone3 kernel: [56B blob data]
Mar 10 11:25:56 beaglebone3 kernel: R8188EU: =>rtw_wx_set_essid
Mar 10 11:25:56 beaglebone3 kernel: +rtw_wx_set_essid: fw_state = 0x00000008
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x02000000,8]
Mar 10 11:25:56 beaglebone3 kernel: mlme_cmd_hdl(): pcmd_callback = 0x  (null), cmdcode = 0x13
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000080,8]
Mar 10 11:25:56 beaglebone3 kernel: rtw_set_802_11_bssid: status=0
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00020000,5]
Mar 10 11:25:56 beaglebone3 kernel: R8188EU: ERROR set bssid:00:00:00:00:00:00
Mar 10 11:25:56 beaglebone3 kernel: after enqueue set_auth_cmd, auth_mode=0
Mar 10 11:25:56 beaglebone3 kernel: R8188EU:  [0x00000010,5]
Mar 10 11:25:56 beaglebone3 kernel: rtw_set_802_11_authentication_mode:psecuritypriv->ndisauthtype=0

```

The files used are:
# sha1sum 8188eu.ko 8188eu.mod.c rtl8188eufw.bin 
```
0deada6a093eabe3e92c1e07b7867bc86b9153e4  8188eu.ko
8d026ef6ee023e2d8a8e3578ca554059b8e9365a  8188eu.mod.c
56b8cebbf0a2042916c135df97b8e6c1b7423785  rtl8188eufw.bin
```

lwfinger, 2015-03-10:
Nothing in the dmesg output is an error. The only error that you show came from conman, which apparently forced a disassociation, and then never was able to make a P2P connection. That seems to be a missing part in conman.

I have never tested P2P, and I am not sure how well it works. The only thing I can suggest is that you download the v4.1.8_9499 branch of the repo and try it. Rather than getting the zipped version, I suggest using git. That way you get *all* the branches, and changing from one to the other is trivial.

Jibun-no-Kage, 2022-10-05:
Just FYI... but Pi OS 11 (Debian ARM base for Raspberry Pi) current driver complies, loads, but then dmesg reports 'R8188EU: INFO indicate disassoc' which is interesting because if an explicit association is done, it is successful.

Explicit association example for wlan0 interface...
\# wpa_supplicant -B -D wext -i wlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf

Of course, wpa_supplicant.conf just configures psk SSID and key.