lwfinger/rtl8188eu #107: Kernel panic at recvframe_chk_defrag

nieltg, 2015-03-15:
Hi there!

First, thank you, lwfinger, for maintaining this great driver. I've used it for ~3 weeks and got this error (crash at ```recvframe_chk_defrag```) twice. I've forgotten about the first error, but this one occurred after I re-plugged the device (because the device wasn't blinking) using this command:

```# cd /sys/bus/usb/drivers/hub; echo 1-0:1.0 > unbind; sleep 1; echo 1-0:1.0 > bind```

I usually re-plug the device (via command) after booting because the device won't usually work & blink before I do it. Here, I've uploaded my screen and thank you very much!

[image]

lwfinger, 2015-03-16:
I will check out this oops. I have never used the unbind/bind sequence. I always use rmmod/insmod or modprobe to unload/reload the driver.

As you talk about the "blink", you are probably using the master branch. Is that correct?

nieltg, 2015-03-16:
Thank you! Yeah, I'm on master branch, at b17edc5b73142e0de1698c2adaf3642e20e079b9 to be specific.

Hmm, I think the device (TL-WN785N) always blinks while it's working correctly, doesn't it? Will its blinking behavior changes if I use other branch?