lwfinger/rtl8188eu #109: realtek's v4.1.8 seems to have different firmware data

elpetak, 2015-03-27:
I was testing v4.1.8 from this repo vs the realtek code I was previuosly using. I thought it should perform pretty much the same, but it didn't, which is weird.
I always use the same video to test with Kodi on my raspberry pi.
With my version of realtek's 4.1.8 (right out of the tar file with minor patches taken from this repo to make it compile without warnings) it plays smothly but with the v4.1.8 from this repo it stutters.
Since the codebase is the same and AFAIK the code on this repo has just some fixes and cleanup I started to think what else could be different...
and then it hitted me: the firmware

So I compared the firmware file content against the data in the ./hal/OUTSRC/rtl8188e/HalHWImg8188E_FW.c from realteks v4.1.8
and they are a lot different. A quick look and the third and fifth bytes are already different.
binary firmware file from here starts with:  E1 88 10 00 0B
new file data starts with: 0xE1, 0x88, 0x30, 0x00, 0x0D,
and it goes on...

Could that be what's making a difference between the two v4.1.8 ?

I'll see if I can translate the data from HalHWImg8188E_FW.c into a firmware file and try it, but my C skills are at least rusty.

elpetak, 2015-03-27:
Ok, on a second look, I may be getting it all wrong. I see many files in that subdir with binary data so I'm not sure which one is the actual firmware data so maybe I'm completely wrong.

elpetak, 2015-03-27:
ok, so it's not the firmware... the file O was looking at is for CONFIG_WOWLAN which is not set.
So I'm clueless about why the realtek's version works better for me on the exact same test.

lwfinger, 2015-03-27:
Do you know that your shooting from the hip (Posting before you think) is a PITA.

If you want to see what the differences are, then use the diff utility. The command 'man diff' will show you how to use it.

elpetak, 2015-03-27:
I'm sorry, I posted too early thinking I had found a newer firmware and then I realized I was completely wrong.
There are way too many differences between the two sources now and I honestly can't understand most of it. Dealing with hardware level code is not my area of expertise as it has been clearly established on this thread.
I'll just close this. Sorry about that.

closed 2015-03-27 by elpetak