lwfinger/rtl8188eu #112: Can not work in OpenWrt

sfzy, 2015-04-21:
Hi lwfinger
I compile the packet but cannot work in OpenWrt(in virtual box, x86) both in Attitude Adjustment 12.09 and Barrier Breaker 14.07. Even print the error "Kernel panic - not  syncing: Fatal exception in interrupt. Rebooting in 3 secoonds.."

And I want to know what is the difference between the master and 4.1.8 branch. When I used the 4.1.8 branch, it can not work with many errors with "Unknown symbol ...",like " Unknown symbol wiphy_register (err 0)" "Unknown symbol cfg80211_get_channel (err 0)"

Thank you

lwfinger, 2015-04-21:
The master branch is based on Realtek's version "v4.1.4_6773.20130222".
The v.1.8 version is based on version "v4.1.8_9499.20131104".

I do not know all the differences, but the main one is that 4.1.4 does not implement the cfg80211 calls, which is why cannot work with a recent hostapd. Version 4.1.8 does work with cfg80211; therefore, it needs cfg80211. Your unknown symbol problem is that your system is not configured correctly. I have no idea why you get the kernel panic. The message you show is insufficient for debugging.

All I know is that both versions work, within their limitations, on x86_64 on Linux. I will try to set up an openWRT implementation to help debug. Unfortunately, it will be several days until I get the time.

sfzy, 2015-04-22:
Thank you!
Here is my system I used. links：http://pan.baidu.com/s/1eQtRw5O key：4lql

lwfinger, 2015-04-22:
When you run 'make menuconfig' for openWRT, under the "Wireless Drivers" page, you need to select cfg80211. I think you missed that part.

The opening page of the link above was quite interesting. Looking at a page where I could not read a single thing does not happen that often. Fortunately, I did not need to use Google translate to get it right.

sfzy, 2015-04-23:
I added cfg80211 like this, I dont know if I didnt make the cfg80211 right. In make menuconfig, I got Kernel modules-->Wireless Drivers -->(I find this one, I dont know if it is you want me to add one)kmod-cfg80211. Then I run make. And remake the 8188eu.ko, but I got this ('...'parts looks all right)
"...
Building modules, stage 2.
  MODPOST 1 modules
WARNING: "wiphy_free" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "ieee80211_frequency_to_channel" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "cfg80211_inform_bss_frame" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "cfg80211_rx_mgmt" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "cfg80211_del_sta" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "cfg80211_mgmt_tx_status" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "__ieee80211_get_channel" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "cfg80211_get_bss" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "wiphy_unregister" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "cfg80211_connect_result" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "cfg80211_michael_mic_failure" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "cfg80211_ibss_joined" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "cfg80211_roamed" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "cfg80211_put_bss" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "wiphy_new" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "wiphy_register" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "cfg80211_ready_on_channel" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "cfg80211_disconnected" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "cfg80211_new_sta" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "cfg80211_remain_on_channel_expired" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
WARNING: "cfg80211_scan_done" [/home/ubk/openwrt/8188/8188eu.ko] undefined!
  CC      /home/ubk/openwrt/8188/8188eu.mod.o
  LD [M]  /home/ubk/openwrt/8188/8188eu.ko
"
Did I miss somethings to add or add a wrong cfg80211.
I upload the 8188eu.ko to the system,and in openwrt installed the kmod-cfg80211. I run insmod 8188eu, I got the message:
[image]

lwfinger, 2015-04-23:
The kmod-cfg80211 is the only one I found.

By the way, when the build process gets missing global warnings, they will certainly show when you try to load the module. Some warnings may be ignored, but not those.

I do not know what is happening. I will try to add the 8188eu driver to the standard openWRT package, but that will take more reading to find out how to do it.

Noltari, 2015-04-24:
You need to add "+kmod-mac80211" to rtl8188eu dependencies in order to get it linked to mac80211. With that OpenWrt will build mac80211 before rtl8188eu and will link it correctly.
Example: https://github.com/openwrt-es/openwrt/blob/7dfb1b2dd992c9216c273e1a5e563b81fdf4e9ef/package/kernel/mwlwifi/Makefile#L32

BTW, in order to test if a specific package compiles you can remove the current build and compile it again with:
```
make package/rtl8188eu/clean V=s
make package/rtl8188eu/compile V=s
```
Where "rtl8188eu" can be changed to any other package, with "rtl8188eu" being the folder where the package is located and not the name of the package.
Example: kmod-mac80211 would be:
```
make package/mac80211/clean V=s
make package/mac80211/compile V=s
```
Because it's located in package/kernel/**mac80211**.

sfzy, 2015-04-26:
Thanks Noltari 
I think I can not understand your suggestion correctly, I am a newer to makefile. I just know change the  line about "CROSS_COMPILE :" and "KSRC :". So I add "DEPENDS:=+kmod-mac80211" in line 3. Then remake 8188eu.ko. Put it in openwrt. It shows the same wrong message.

lwfinger, 2015-04-26:
You need cfg80211, not mac80211.

sfzy, 2015-04-27:
I copy the example without thinking. I change it to "DEPENDS:=+kmod-cfg80211", but it looks no difference, I got the same wrong message.

lwfinger, 2015-04-27:
What modules are located in /lib/modules/`uname -r`/?

sfzy, 2015-04-27:
this？
[image]

lwfinger, 2015-04-27:
You are generating cfg80211.ko. Are you loading it before you load 8188eu.ko? On most systems, running the utility named depmod will create the module dependencies so that using modprobe to load a module will pull in all the dependent modules. I do not know enough about openWRT to know how it works.

If you use insmod to load the module, then you must manually load all dependent modules by yourself.

sfzy, 2015-04-27:
Here is no depmod in my openwrt.
I use insmod cfg80211, but failed to insert it.
[image]

lwfinger, 2015-04-28:
Why? The output of dmesg should show the cause.

sfzy, 2015-04-29:
The dmesg shows a lot of message, I don't know which is helpful. (I think it is too much more message (436 lines) to post here).Maybe you can download it from here:links:http://pan.baidu.com/s/1eQtRw5O key:4lql

lwfinger, 2015-04-29:
I saw no indication why cfg80211 failed to load. When you get errors, please post the actual error message.

Could you make the latest x86 vhd available?

sfzy, 2015-04-29:
When I run insmod cfg80211, I got not error except the only message "kmod: failed to insert /lib/modules/3.10.49/cfg80211.ko". Is there some other commands to get some actual error? I really don't know what to do for next step.

The x86 vhd of openwrt I download from http://downloads.openwrt.org/barrier_breaker/14.07/x86/generic/openwrt-x86-generic-combined-ext4.img.gz
and unzip it, get the img file, and put it in ubuntu 14.04. Use dd command, then I got the vhd.
I can not understand "Could you make the latest x86 vhd available?", So I post how I got the vhd.

joshith, 2015-04-29:
I was facing the same problem with 4.1.8 driver. Turned out to be the 8188eu has dependancy over cfg80211 and which inturn is depended on rfkill. Surprisingly the depmod did not find these. insmod rfkill followed by cf80211 followed by 8188eu solved my problem. Can you check the rfkill is loaded in your case

sfzy, 2015-04-29:
There is no rfkill in my openwrt. I run "insmod rfkill", I got "Failed fo find rfkill. Maybe it is a built in module ?"
In update list, there is no rfkill too.

lwfinger, 2015-04-29:
When either modprobe or insmod fail to load a module, the reasons are listed in the output of dmesg. When your "insmod cfg80211" fails with the console message that you showed above, immediately do "dmesg | less" and go to the bottom of the listing. The missing globals should be listed there.

sfzy, 2015-05-03:
I tried "dmesg | less",but the output message looks same as I used "dmesg | more", that I post before.

lwfinger, 2015-05-04:
Obviously, the pager used will never change the output, but what were the error messages in dmesg when "modprobe cfg80211" failed. When you use insmod, you need a complete path. You likely got a file not found message.

sfzy, 2015-05-05:
I got something this time. When I run "modprobe cfg80211", I got some failed message, it said compat.ko not found. So I want to opkg it, but I got a problem, "* parse_from_stream_nomalloc: Missing new line character at end of file!" I got this message, I can not "opkg update" "opkg install" ipk. Then I delete my openwrt vhd,and rename my backup vhd.  I  installed cfg80211 and I got someting difference from before.
(I can not post images, I don't know why. Github said "Something went really wrong, and we can't process that file.")
The message like this "cfg80211: (2457000KHz - 2482000 KHz @ 40000)..."
It looks the cfg80211 is working. 
So I run insmod 8188eu or modprobe 8188eu, I got only one "Unknown symbol" message, the full message is "Unknown symbol cfg80211_inform_bss_frame (err 0)".

steczol, 2015-09-14:
Hi everyone,
I`m trying to get the realtek usb-dongle-chip 8188eu to work under openwrt. After doing what @joshith wrote: rmmod rfkill, cfg80211, r8188eu; then insmod rfkill, cfg80211, r8188eu, i always get the kernel panic error. Log output:

```
root@OpenWrt:/# insmod r8188eu debug=5
[   67.370000] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_UNKNOWN_CUT(10)_1T1R_RomVer(0)
[   67.380000] R8188EU: RF_Type is 3!!
[   67.390000] R8188EU: _ConfigNormalChipOutEP_8188E OutEpQueueSel(0x05), OutEpNumber(2)
[   67.410000] R8188EU: ====> _ReadAdapterInfo8188EU
[   67.460000] R8188EU: Boot from EFUSE, Autoload OK !
[   67.490000] R8188EU: bFWReady == false call reset 8051...
[   67.520000] R8188EU: =====> _8051Reset88E(): 8051 reset success .
[   67.580000] R8188EU: efuse_read_phymap_from_txpktbuf bcnhead:0
[   67.610000] R8188EU: efuse_read_phymap_from_txpktbuf len:123, lenbak:123, aaa:123, aaabak:123
[   67.720000] R8188EU: efuse_read_phymap_from_txpktbuf read count:121
[   67.740000] R8188EU: EEPROM ID = 0x8129
[   67.740000] R8188EU: VID = 0x0BDA, PID = 0x8179
[   67.750000] R8188EU: Customer ID: 0x00, SubCustomer ID: 0xCD
[   67.760000] R8188EU: Hal_ReadPowerSavingMode88E...bHWPwrPindetect(0)-bHWPowerdown(0) , bSupportRemoteWakeup(1)
[   67.780000] R8188EU: ### PS params =>  power_mgnt(1), usbss_enable(0) ###
[   67.800000] R8188EU: ======= Path 0, Channel 0 =======
[   67.810000] R8188EU: Index24G_CCK_Base[0][0] = 0x38
[   67.820000] R8188EU: Index24G_BW40_Base[0][0] = 0x35
[   67.830000] R8188EU: ======= Path 0, Channel 1 =======
[   67.840000] R8188EU: Index24G_CCK_Base[0][1] = 0x38
[   67.850000] R8188EU: Index24G_BW40_Base[0][1] = 0x35
[   67.860000] R8188EU: ======= Path 0, Channel 2 =======
[   67.870000] R8188EU: Index24G_CCK_Base[0][2] = 0x38
[   67.880000] R8188EU: Index24G_BW40_Base[0][2] = 0x35
[   67.890000] R8188EU: ======= Path 0, Channel 3 =======
[   67.900000] R8188EU: Index24G_CCK_Base[0][3] = 0x36
[   67.910000] R8188EU: Index24G_BW40_Base[0][3] = 0x33
[   67.920000] R8188EU: ======= Path 0, Channel 4 =======
[   67.930000] R8188EU: Index24G_CCK_Base[0][4] = 0x36
[   67.940000] R8188EU: Index24G_BW40_Base[0][4] = 0x33
[   67.950000] R8188EU: ======= Path 0, Channel 5 =======
[   67.960000] R8188EU: Index24G_CCK_Base[0][5] = 0x36
[   67.970000] R8188EU: Index24G_BW40_Base[0][5] = 0x33
[   67.980000] R8188EU: ======= Path 0, Channel 6 =======
[   67.990000] R8188EU: Index24G_CCK_Base[0][6] = 0x35
[   68.000000] R8188EU: Index24G_BW40_Base[0][6] = 0x32
[   68.010000] R8188EU: ======= Path 0, Channel 7 =======
[   68.020000] R8188EU: Index24G_CCK_Base[0][7] = 0x35
[   68.030000] R8188EU: Index24G_BW40_Base[0][7] = 0x32
[   68.040000] R8188EU: ======= Path 0, Channel 8 =======
[   68.050000] R8188EU: Index24G_CCK_Base[0][8] = 0x35
[   68.060000] R8188EU: Index24G_BW40_Base[0][8] = 0x32
[   68.070000] R8188EU: ======= Path 0, Channel 9 =======
[   68.080000] R8188EU: Index24G_CCK_Base[0][9] = 0x33
[   68.090000] R8188EU: Index24G_BW40_Base[0][9] = 0x30
[   68.100000] R8188EU: ======= Path 0, Channel 10 =======
[   68.110000] R8188EU: Index24G_CCK_Base[0][10] = 0x33
[   68.120000] R8188EU: Index24G_BW40_Base[0][10] = 0x30
[   68.130000] R8188EU: ======= Path 0, Channel 11 =======
[   68.140000] R8188EU: Index24G_CCK_Base[0][11] = 0x33
[   68.150000] R8188EU: Index24G_BW40_Base[0][11] = 0x30
[   68.160000] R8188EU: ======= Path 0, Channel 12 =======
[   68.170000] R8188EU: Index24G_CCK_Base[0][12] = 0x34
[   68.180000] R8188EU: Index24G_BW40_Base[0][12] = 0x31
[   68.190000] R8188EU: ======= Path 0, Channel 13 =======
[   68.200000] R8188EU: Index24G_CCK_Base[0][13] = 0x34
[   68.210000] R8188EU: Index24G_BW40_Base[0][13] = 0x31
[   68.220000] R8188EU: ======= TxCount 0 =======
[   68.230000] R8188EU: CCK_24G_Diff[0][0] = 0
[   68.240000] R8188EU: OFDM_24G_Diff[0][0] = 2
[   68.250000] R8188EU: BW20_24G_Diff[0][0] = 0
[   68.250000] R8188EU: BW40_24G_Diff[0][0] = 0
[   68.260000] R8188EU: ======= TxCount 1 =======
[   68.270000] R8188EU: CCK_24G_Diff[0][1] = -2
[   68.280000] R8188EU: OFDM_24G_Diff[0][1] = -2
[... 46 lines cut ...]
[   69.500000] PrId  : 0001964c (MIPS 24KEc)
[   69.500000] Modules linked in: r8188eu pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT spidev slhc nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat_ftp nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_ftp nf_conntrack lib80211_crypt_wep lib80211_crypt_tkip lib80211_crypt_ccmp lib80211 iptable_raw iptable_mangle iptable_filter ip_tables crc_itu_t crc_ccitt compat cdc_wdm i2c_ralink i2c_dev i2c_core ledtrig_usbdev ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables rfkill ipv6 eeprom_93cx6 arc4 crypto_blkcipher ohci_platform ohci_hcd ehci_platform ehci_hcd gpio_button_hotplug button_hotplug input_core usbcore nls_base usb_common [last unloaded: r8188eu]
[   69.500000] Process dropbearkey (pid: 1145, threadinfo=80e80000, task=80f02220, tls=77328644)
[   69.500000] Stack : 00000020 802e06e4 802e0000 00000000 00000007 00000001 80344be0 802e0000
	  80e81ea0 8004f400 8028e1d0 80048b40 00000000 802db570 80e81ea0 80e81ea0
	  7f8efc60 00000002 80323864 0000000a 00000002 00000100 80320000 800264cc
[... cut ...]
[   69.500000] [<8004b694>] handle_percpu_irq+0x54/0x84
[   69.500000] [<80026838>] irq_exit+0x54/0x70
[   69.500000] [<80004430>] ret_from_irq+0x0/0x4
[   69.500000] 
[   69.500000] 
Code: (Bad address in epc)
[   69.500000] 
[   70.000000] ---[ end trace 8870e92054d1c2c3 ]---
[   70.020000] Kernel panic - not syncing: Fatal exception in interrupt
[   70.020000] Rebooting in 3 seconds..
```
I`m gonna try to find out what is happened, but perhaps you are quicker and more accurate in handling this problem. Greetings!

lwfinger, 2015-09-15:
Unfortunately, I cannot reproduce the problem on my x86_64 laptop. In addition, the traceback does not list any 8188eu routines, thus the problem is deeply hidden. I'm not sure how much help I can be.

lwfinger, 2015-09-15:
As your crash happens in a timer callback, perhaps one of them is not being cancelled on exit. I just pushed a change that cancels several more timers. Perhaps it will fix your problem.

sfzy, 2015-10-29:
I find the openwrt 15.05 has the 8188eu driver now. And I find the file TODO has "Please send any patches to Greg Kroah-Hartman <gregkh@linux.com>,and Larry Finger <Larry.Finger@lwfinger.net>."
But I don't know the driver is from master branch or 4.1.8.

lwfinger, 2015-10-29:
As I did not put the driver into openWRT, I do not know either. If the driver knows about the cfg80211 or nl80211 commands, then it is the 4.1.8 branch.