lwfinger/rtl8188eu #114: rtl8188eu compiles under Mint 13 but freezes on browser launch

elstevester, 2015-04-21:
I have a TP-LINK WN725N wireless usb adapter on my elderly Dell Dimension PC. I've successfully compiled the master branch rtl8188eu driver for my Maya 32-bit install (4189499 and kernel-code do not compile, generating a warning that a label can only be part of statement and declarations are not ... etc.).

steve@ElDell ~ $ uname -a
Linux ElDell 3.2.0-33-generic #52-Ubuntu SMP Thu Oct 18 16:19:45 UTC 2012 i686 i686 i386 GNU/Linux

So, all looks good. I get the device recognized and can ping external addresses:

steve@ElDell ~ $ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 004: ID 0bda:8179 Realtek Semiconductor Corp.
Bus 002 Device 002: ID 413c:2003 Dell Computer Corp. Keyboard
Bus 003 Device 002: ID 046d:c016 Logitech, Inc. Optical Wheel Mouse

steve@ElDell ~ $ iwconfig
lo no wireless extensions.

wlan0 IEEE 802.11bgn ESSID:"TALKTALKXXXXXX" Nickname:"<WIFI@REALTEK>"
Mode:Managed Frequency:2.437 GHz Access Point: 68:A0:F6:5F:8D:11
Bit Rate:150 Mb/s Sensitivity:0/0
Retry:off RTS thr:off Fragment thr:off
Power Management:off
Link Quality=45/100 Signal level=-58 dBm Noise level=0 dBm
Rx invalid nwid:0 Rx invalid crypt:0 Rx invalid frag:0
Tx excessive retries:0 Invalid misc:0 Missed beacon:0

eth0 no wireless extensions.

steve@ElDell ~/rtl8188eu-master $ lsmod
Module Size Used by
...
8188eu 672091 0
...

steve@ElDell ~ $ ping http://www.google.co.uk
PING http://www.google.co.uk (216.58.211.163) 56(84) bytes of data.
64 bytes from dub08s01-in-f3.1e100.net (216.58.211.163): icmp_req=1 ttl=53 time=133 ms
64 bytes from dub08s01-in-f3.1e100.net (216.58.211.163): icmp_req=2 ttl=53 time=53.2 ms
64 bytes from dub08s01-in-f3.1e100.net (216.58.211.163): icmp_req=3 ttl=53 time=47.9 ms
64 bytes from dub08s01-in-f3.1e100.net (216.58.211.163): icmp_req=4 ttl=53 time=26.7 ms
^C
--- http://www.google.co.uk ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3003ms
rtt min/avg/max/mdev = 26.712/65.330/133.365/40.518 ms

However, if I open a browser, the system immediately freezes. If I leave the machine running, then on some occasions, after a short period (10-30 mins), I either get a system panic or freeze. No obviously attributable errors in syslog/kern.log or dmesg

Since it has compiled and installed OK, not really sure if this is an issue for here, or more of a Mint/Ubuntu problem. I have posted on a Mint forum, but no advice has been forthcoming. I could upgrade to a newer Mint, but 13 has been working fine for me until I needed to relocate my PC and go wireless.

Any advice? Any specific diagnostics that would assist?

Thanks.

lwfinger, 2015-04-22:
The kernel version branch is for a current kernel, which 3.2 is *clearly* not. If you post the errors you get for 4.1.8, I can try to fix them.

Kernel 3.2 is so old, it is hard to know what might be wrong. I know that there have been lots of fixes to the USB core and that may be part of your problem. Unless you can post the output from the kernel oops, there is nothing I can do. That information will *not* be logged. The best you can do is either implement netconsole, which is a PITA, or select the console logging screen (CTRL-ALT-F10 on openSUSE) and take a photo of the crash when it happens. Even then, it may not be helpful.

elstevester, 2015-04-22:
Thanks for that. I was posting more in hope than expectation as yes, it is a bit of an antique setup. I'll have a look at Netconsole and see if I can get any useful output.  If I get nothing helpful, I'll close the issue.
I had previously been happy with Mint's upgrade motto "As a general rule... unless you need to, or unless you really want to, there's no reason for you to upgrade." but  this issue may well be that need.
Thanks again.
Steve

Date: Tue, 21 Apr 2015 18:58:44 -0700