lwfinger/rtl8188eu #115: Missing 8188eu.ko under centos 6.6

mlazarov, 2015-04-21:
I'm trying to compile the kernel module but no luck.

````
# uname -a
Linux test.lazarov.org 2.6.32-504.12.2.el6.centos.plus.x86_64 #1 SMP Thu Mar 12 18:39:03 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

# cat /etc/redhat-release
CentOS release 6.6 (Final)
```

```
# git clone git@github.com:lwfinger/rtl8188eu.git
# cd rtl8188eu
# make all
....
# make install
install -p -m 644 8188eu.ko  /lib/modules/2.6.32-504.12.2.el6.centos.plus.x86_64/kernel/drivers/net/wireless
install: cannot stat `8188eu.ko': No such file or directory
make: *** [install] Error 1
```
Full output from last command:
https://gist.github.com/mlazarov/8496617be65925097c09

```
# yum install kernel kernel-headers kernel-devel kernel-firmware
Package kernel-2.6.32-504.12.2.el6.centos.plus.x86_64 already installed and latest version
Package kernel-headers-2.6.32-504.12.2.el6.centos.plus.x86_64 already installed and latest version
Package kernel-devel-2.6.32-504.12.2.el6.centos.plus.x86_64 already installed and latest version
Package kernel-firmware-2.6.32-504.12.2.el6.centos.plus.noarch already installed and latest version
```

What is wrong? Why i'm not getting this 8188eu.ko module and there is no build error?

lwfinger, 2015-04-22:
No idea. What does 'ls' in the rtl8188eu/ directory show?

Are you aware that one should never, never, never build anything while logged in as root? A few years ago, running a kernel make as root caused /dev/null to be deleted. You can imagine how badly the screwed up systems. To prevent a bug in the make file from destroying your system, always run make as a normal user. Once the is done, the 'sudo make install' as that step requires privilege. Nothing else does!!

mlazarov, 2015-09-18:
Using latest version I have the same issue. Here is the list of files/directories inside rtl8188eu:

```
[mlazarov@test rtl8188eu]$ ls -la
total 21956
drwxrwxr-x 9 mlazarov mlazarov     4096 Sep 18 16:13 .
drwxrwxr-x 3 mlazarov mlazarov     4096 Sep 18 16:13 ..
-rw-rw-r-- 1 mlazarov mlazarov     1026 Sep 18 16:13 8188eu.mod.c
-rw-rw-r-- 1 mlazarov mlazarov 22367317 Sep 18 16:13 8188eu.o
-rw-rw-r-- 1 mlazarov mlazarov     3143 Sep 18 16:13 .8188eu.o.cmd
-rwxrwxr-x 1 mlazarov mlazarov     4199 Sep 18 16:12 control_ap
-rw-rw-r-- 1 mlazarov mlazarov    18693 Sep 18 16:12 COPYING
drwxrwxr-x 2 mlazarov mlazarov     4096 Sep 18 16:13 core
-rw-rw-r-- 1 mlazarov mlazarov      201 Sep 18 16:12 dkms.conf
drwxrwxr-x 8 mlazarov mlazarov     4096 Sep 18 16:12 .git
-rw-rw-r-- 1 mlazarov mlazarov     1106 Sep 18 16:12 .gitignore
drwxrwxr-x 2 mlazarov mlazarov     4096 Sep 18 16:13 hal
drwxrwxr-x 4 mlazarov mlazarov     4096 Sep 18 16:12 hostapd-0.8
drwxrwxr-x 2 mlazarov mlazarov     4096 Sep 18 16:12 include
-rw-rw-r-- 1 mlazarov mlazarov     3975 Sep 18 16:12 Makefile
-rw-rw-r-- 1 mlazarov mlazarov       46 Sep 18 16:13 modules.order
-rw-rw-r-- 1 mlazarov mlazarov        0 Sep 18 16:13 Module.symvers
drwxrwxr-x 2 mlazarov mlazarov     4096 Sep 18 16:13 os_dep
-rw-rw-r-- 1 mlazarov mlazarov      775 Sep 18 16:12 README.md
-rw-rw-r-- 1 mlazarov mlazarov    13904 Sep 18 16:12 rtl8188eufw.bin
-rw-rw-r-- 1 mlazarov mlazarov     1820 Sep 18 16:12 rtl_hostapd.conf
drwxrwxr-x 2 mlazarov mlazarov     4096 Sep 18 16:13 .tmp_versions
[mlazarov@test rtl8188eu]$ make install
install -p -m 644 8188eu.ko  /lib/modules/2.6.32-573.3.1.el6.centos.plus.x86_64/kernel/drivers/net/wireless
install: cannot stat `8188eu.ko': No such file or directory
make: *** [install] Error 1
[mlazarov@test rtl8188eu]$
```

lwfinger, 2015-09-18:
The command 'make install' will always fail. You must use 'sudo make install'.

closed 2015-09-18 by lwfinger

mlazarov, 2015-09-18:
No, that's not the problem.

```
[root@test rtl8188eu]# make install
install -p -m 644 8188eu.ko  /lib/modules/2.6.32-573.3.1.el6.centos.plus.x86_64/kernel/drivers/net/wireless
install: cannot stat `8188eu.ko': No such file or directory
make: *** [install] Error 1
[root@test rtl8188eu]#
``

lwfinger, 2015-09-18:
What does 'make' as the regular user show? Forget 'make all'.

mlazarov, 2015-09-18:
Here is all the output

https://gist.github.com/mlazarov/48bac8c519a2781fa5c7

lwfinger, 2015-09-18:
The problem is in these two lines:
  Building modules, stage 2.
  MODPOST 0 modules

As I have no instance of any kernel or make that is as old as yours, I have no means of finding what is wrong. I guess I will have to install Centos 6.6 in a VM; however, that will take a while.

lwfinger, 2015-09-18:
I cannot duplicate your result on Centos 6.6 either. The driver built normally and installed correctly.

My only suggestion is to 'sudo yum reinstall make' in case yours is corrupted.