lwfinger/rtl8188eu #116: Unable to use TP-LINK WN725N with Xbian

phuongngoc, 2015-05-11:
I lost 2 days to search and try to make it works, but no success.
So I put here my situation, can you please have a look and give me your solution if any:

I'm running latest Xbian version (XBian_2015.04.04_rpi2.img) on Raspberry Pi 2
Here is the information:

> xbian@xbian ~ $ uname -a
Linux xbian 3.18.8+ #1 SMP PREEMPT Thu Mar 5 03:13:36 CET 2015 armv7l GNU/Linux

> xbian@xbian ~ $ cat /proc/version
Linux version 3.18.8+ (root@vps01) (gcc version 4.7.3 (Ubuntu/Linaro 4.7.3-12ubuntu1) ) #1 SMP PREEMPT Thu Mar 5 03:13:36 CET 2015

> xbian@xbian ~ $ gcc --version
-bash: gcc: command not found

> xbian@xbian ~ $ lsusb
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp.
Bus 001 Device 004: ID 0bda:8179 Realtek Semiconductor Corp.

`dmesg` said: "8188eu: disagrees about version of symbol module_layout"

Do you need other information?

This issue happens with Xbian only, it's automatically detected by OSMC and OpenELEC

Thank you.

lwfinger, 2015-05-11:
A disagreement of module_layout means that the driver was built against the wrong kernel headers. As you do not say much about what you are doing, I cannot tell you what you need to do.

kure94, 2015-06-12:
I would like to know answer too. I have exactly the same problem. I´ve installed fresh xbian on my raspberry 2 b with wifi dongle TP-LINK tl-wn725n V2, but it cant allocate it. when i go to network setting there is only eth0 that cant be changed. I found a installation guide but not for my "uname -a" respons: Linux xbian 3.18.8+ #1. Can somebody give me advice how to sovle it? Thank you very much!

lwfinger, 2015-06-12:
For kernel 3.18, there is a driver in the standard kernel. Perhaps xbian does not enable any staging drivers. In that case, you will need to build your own kernel.

Linux is finicky with regard to the compilation of out-of-kernel modules. You need to build against the exact set of headers that were used to build the kernel. In addition, you need to use the same compiler version. That makes it difficult to build the kernel on one machine, and the module on another. Either build a new kernel on the RPi, or set up a cross build arrangement on some other machine and build both the kernel and module there.

Your problem is not allocating the device - you are unable to load the driver.

kure94, 2015-06-12:
emmmm... OK. I didn´t understand almost a single word. I bit newbie in this. Is here any possibility you could give me a step-by-step tutorial? Thank you very much.

lwfinger, 2015-06-12:
Install the latest version of Raspbian. It has the driver for RTL8188EU already built into the distribution.

MrEngman, 2015-06-12:
Yes, but it will not work as there is no firmware for it.

lwfinger, 2015-06-13:
Copy the firmware from this repo. That will work.

sajohn, 2015-06-13:
Hello lwfinger,
Off the topic request: 
I don't know how to send email for individual request but I am wondering if you have a firmware source code/documentation for RTL8188EU (11n) chip and what are the plants to have 11ac firmware code release if any? 

Thanks

lwfinger, 2015-06-13:
The radio is 802.11n, not 802.11ac. There is no way to change the hardware with new firmware.

Realtek releases no firmware source or documentation. They have only binaries with a right to distribute/copy license.