lwfinger/rtl8188eu #119: my system completely freeze when load 8188eu.ko sometimes

adals, 2015-05-20:
hello
my system completely freeze sometimes
I tested it with latest version of  fedora and ubuntu 
i leave my compute running for days when  8188eu.ko is loadable  it will freeze and when it does not it will not freeze at all 

please help me and give me any command if you want to see any results to help in fixing it

thanks

lwfinger, 2015-05-20:
Without some indication of why the kernel crashed, there is nothing I can do. It does not freeze my computer.

What architecture are you using? With the latest versions, why are you not using the driver in the kernel? Does it also freeze?

adals, 2015-05-21:
hello
I am using the driver comes with kernel and same result

when I plugged usb (wireless) my system sometimes freezes 
when I plugged-off usb (wireless) my system will not freeze

[adals@localhost ~]$ uname -r
3.19.7-200.fc21.i686

[adals@localhost ~]$ cat /proc/cpuinfo 
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 42
model name	: Intel(R) Core(TM) i7-2600K CPU @ 3.40GHz
stepping	: 7
microcode	: 0x29
cpu MHz		: 1699.734
cache size	: 8192 KB
physical id	: 0
siblings	: 8
core id		: 0
cpu cores	: 4
apicid		: 0
initial apicid	: 0
fdiv_bug	: no
f00f_bug	: no
coma_bug	: no
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe nx rdtscp lm constant_tsc arch_perfmon pebs bts xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 popcnt tsc_deadline_timer aes xsave avx lahf_lm ida arat epb pln pts dtherm tpr_shadow vnmi flexpriority ept vpid xsaveopt
bugs		:
bogomips	: 6820.27
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 42
model name	: Intel(R) Core(TM) i7-2600K CPU @ 3.40GHz
stepping	: 7
microcode	: 0x29
cpu MHz		: 1684.328
cache size	: 8192 KB
physical id	: 0
siblings	: 8
core id		: 1
cpu cores	: 4
apicid		: 2
initial apicid	: 2
fdiv_bug	: no
f00f_bug	: no
coma_bug	: no
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe nx rdtscp lm constant_tsc arch_perfmon pebs bts xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 popcnt tsc_deadline_timer aes xsave avx lahf_lm ida arat epb pln pts dtherm tpr_shadow vnmi flexpriority ept vpid xsaveopt
bugs		:
bogomips	: 6820.27
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

[... 224 lines cut ...]
parm:           rtw_ht_enable:int
parm:           rtw_cbw40_enable:int
parm:           rtw_ampdu_enable:int
parm:           rtw_rx_stbc:int
parm:           rtw_ampdu_amsdu:int
parm:           rtw_lowrate_two_xmit:int
parm:           rtw_rf_config:int
parm:           rtw_power_mgnt:int
parm:           rtw_smart_ps:int
parm:           rtw_low_power:int
parm:           rtw_wifi_spec:int
parm:           rtw_antdiv_cfg:int
parm:           rtw_antdiv_type:int
parm:           rtw_enusbss:int
parm:           rtw_hwpdn_mode:int
parm:           rtw_hwpwrp_detect:int
parm:           rtw_hw_wps_pbc:int
parm:           rtw_max_roaming_times:The max roaming times to try (uint)
parm:           rtw_fw_iol:FW IOL (int)
parm:           rtw_mc2u_disable:int
parm:           rtw_80211d:Enable 802.11d mechanism (int)
parm:           rtw_notch_filter:0:Disable, 1:Enable, 2:Enable only for P2P (uint)
parm:           debug:Set debug level (1-9) (default 1) (int)

thanks

lwfinger, 2015-05-21:
I am a little surprised that you are using a 32-bit kernel, but I installed F21 on one of my laptops and compiled and installed 8188eu.ko on that system. It works perfectly. I have done a 'sudo yum update' to get the latest versions of everything. My kernel matches yours.

To get some info on your freezes, reboot with the RTL8188EU removed. Make sure that you have built and installed the latest version of 8188eu.ko from the v4.1.8_9499 branch. Then use CTRL-ALT-F2 to get a console. After logging in, enter the command 'journalctl -f'. That will show the end of the log file. Then plug in the RTL8188EU. If you get an oops logged, take and post a photo of it. To check for freezes, type CTRL-ALT-F1 to return to the desktop; however, immediately return to the log console with CTRL-ALT-F2.

adals, 2015-05-23:
I downloaded the last version of driver and install it and up to now everything is ok

thanks

closed 2023-05-05 by adals