lwfinger/rtl8188eu #120: not able to keep a connection with v4.1.8_9499

tisdall, 2015-05-20:
With rtl8188EUS_rtl8189ES_linux_v4.1.8_9499.20131104 (that I got from https://github.com/LazyZhu/myblog/raw/gh-pages/file/RTL8188EUS_RTL8189ES_linux_v4.1.8_9499.20131104.zip ) I'm able to connect to my wifi ap with no issue:

	Successfully initialized wpa_supplicant
	==> rtl8188e_iol_efuse_patch 
	RTL871X: set bssid:00:00:00:00:00:00
	ioctl[SIOCSIWAP]: Operation not permitted
	RTL871X: set ssid [g�isQ�J�)ͺ����F|�T���vZ.c3�ɚ] fw_state=0x00000008
	ioctl[SIOCSIWENCODEEXT]: Invalid argument
	ioctl[SIOCSIWENCODEEXT]: Invalid argument
	root@linaro-developer:~# RTL871X: indicate disassoc
	RTL871X: set ssid [Engage Labs] fw_state=0x00000008
	RTL871X: set bssid:b8:a3:86:50:59:c6
	RTL871X: start auth
	RTL871X: auth success, start assoc
	RTL871X: assoc success
	UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:0, rate_bitmap:0x000ff005
	RTL871X: send eapol packet
	RTL871X: send eapol packet
	RTL871X: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) camid:4
	[CCMP]  GTK key_len=16 @@@@@@@@@@@@@@@@@@
	RTL871X: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:2
	UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:1, rate_bitmap:0x000f0000

But with the v4.1.8_9499 branch (of this repo) I get constant disconnects right after connecting as follows:

	...
	
	R8188EU: set bssid:00:00:00:00:00:00
	<6>cfg80211: Calling CRDA to update world regulatory domain
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:0, rate_bitmap:0x000ff005
	R8188EU: nolinked power save enter
	==> rtl8188e_iol_efuse_patch 
	R8188EU: nolinked power save leave
	R8188EU: set bssid:b8:a3:86:50:59:c6
	R8188EU: set ssid [Engage Labs] fw_state=0x00000088
	R8188EU: start auth
	R8188EU: auth success, start assoc
	R8188EU: assoc success
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:0, rate_bitmap:0x000ff005
	R8188EU: rtw_aes_decrypt(wlan0) no_gkey_bc_cnt:3, no_gkey_mc_cnt:0
	R8188EU: rtw_aes_decrypt(wlan0) no_gkey_bc_cnt:3, no_gkey_mc_cnt:0
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:2, rate_bitmap:0x000ff000
	R8188EU: rtw_aes_decrypt(wlan0) no_gkey_bc_cnt:3, no_gkey_mc_cnt:0
	R8188EU: sta recv deauth reason code(2) sta:b8:a3:86:50:59:c6, ignore = 0
	R8188EU: set bssid:00:00:00:00:00:00
	R8188EU: nolinked power save enter
	==> rtl8188e_iol_efuse_patch 
	R8188EU: nolinked power save leave
	R8188EU: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
	R8188EU: start auth
	R8188EU: auth success, start assoc
	R8188EU: assoc success
	<6>cfg80211: Calling CRDA for country: CA
	R8188EU: set bssid:00:00:00:00:00:00
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:0, rate_bitmap:0x000ff005
	R8188EU: nolinked power save enter
	==> rtl8188e_iol_efuse_patch 
	R8188EU: nolinked power save leave
	R8188EU: nolinked power save enter
	==> rtl8188e_iol_efuse_patch 
	R8188EU: nolinked power save leave
	R8188EU: set bssid:b8:a3:86:50:59:c6
	R8188EU: set ssid [Engage Labs] fw_state=0x00000088
	R8188EU: start auth
	R8188EU: auth success, start assoc
	R8188EU: assoc success
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:0, rate_bitmap:0x000ff005
	R8188EU: rtw_aes_decrypt(wlan0) no_gkey_bc_cnt:3, no_gkey_mc_cnt:0
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:1, rate_bitmap:0x000f0000
	R8188EU: sta recv deauth reason code(2) sta:b8:a3:86:50:59:c6, ignore = 0
	R8188EU: set bssid:00:00:00:00:00:00
	R8188EU: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
	R8188EU: start auth
	R8188EU: auth success, start assoc
	R8188EU: assoc success
	<6>cfg80211: Calling CRDA to update world regulatory domain
	R8188EU: set bssid:00:00:00:00:00:00
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:0, rate_bitmap:0x000ff005
	R8188EU: nolinked power save enter
	==> rtl8188e_iol_efuse_patch 
	R8188EU: nolinked power save leave
	R8188EU: nolinked power save enter
	==> rtl8188e_iol_efuse_patch 
	R8188EU: nolinked power save leave
	R8188EU: nolinked power save enter
	==> rtl8188e_iol_efuse_patch 
	R8188EU: nolinked power save leave
	R8188EU: set bssid:b8:a3:86:50:59:c6
	R8188EU: set ssid [Engage Labs] fw_state=0x00000088
	R8188EU: start auth
	R8188EU: auth success, start assoc
	R8188EU: assoc success
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:0, rate_bitmap:0x000ff005
	R8188EU: rtw_aes_decrypt(wlan0) no_gkey_bc_cnt:2, no_gkey_mc_cnt:0
	R8188EU: rtw_aes_decrypt(wlan0) no_gkey_bc_cnt:3, no_gkey_mc_cnt:0
	R8188EU: rtw_aes_decrypt(wlan0) no_gkey_bc_cnt:1, no_gkey_mc_cnt:0
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:2, rate_bitmap:0x000ff000
	R8188EU: rtw_aes_decrypt(wlan0) no_gkey_bc_cnt:3, no_gkey_mc_cnt:0
	R8188EU: sta recv deauth reason code(2) sta:b8:a3:86:50:59:c6, ignore = 0

	...
	

This is with a 3.4 kernel fork.  The top one is using the "wext" driver for wpa_supplicant and the bottom gave the same result whether I used "nl80211" or "wext".

I also tried a connection with no encryption and I got this (over and over again):

	...

	R8188EU: nolinked power save leave
	R8188EU: set bssid:84:7a:88:2d:7a:29
	R8188EU: set ssid [TimmyHTC] fw_state=0x00000088
	R8188EU: set bssid:84:7a:88:2d:7a:29
	R8188EU: set ssid [TimmyHTC] fw_state=0x00000088
	R8188EU: start auth
	R8188EU: auth success, start assoc
	R8188EU: assoc success
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:0, rate_bitmap:0x000ff005
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:1, rate_bitmap:0x000f0000
	R8188EU: set bssid:00:00:00:00:00:00
	R8188EU: set bssid:00:00:00:00:00:00
	R8188EU: nolinked power save enter
	==> rtl8188e_iol_efuse_patch 
	R8188EU: nolinked power save leave
	R8188EU: nolinked power save enter
	==> rtl8188e_iol_efuse_patch 
	R8188EU: nolinked power save leave
	R8188EU: set bssid:84:7a:88:2d:7a:29
	R8188EU: set bssid:84:7a:88:2d:7a:29
	R8188EU: set ssid [TimmyHTC] fw_state=0x00000088
	R8188EU: set ssid [TimmyHTC] fw_state=0x00000088
	R8188EU: start auth
	R8188EU: auth success, start assoc
	R8188EU: assoc success
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:0, rate_bitmap:0x000ff005
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:2, rate_bitmap:0x000ff000
	R8188EU: set bssid:00:00:00:00:00:00
	R8188EU: set bssid:00:00:00:00:00:00
	R8188EU: nolinked power save enter
	==> rtl8188e_iol_efuse_patch 
	R8188EU: nolinked power save leave
	R8188EU: nolinked power save enter
	==> rtl8188e_iol_efuse_patch 
	R8188EU: nolinked power save leave
	R8188EU: set bssid:84:7a:88:2d:7a:29
	R8188EU: set bssid:84:7a:88:2d:7a:29
	R8188EU: set ssid [TimmyHTC] fw_state=0x00000088
	R8188EU: set ssid [TimmyHTC] fw_state=0x00000088
	R8188EU: start auth
	R8188EU: auth success, start assoc
	R8188EU: assoc success
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:0, rate_bitmap:0x000ff005
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:1, rate_bitmap:0x000f0000
	R8188EU: set bssid:00:00:00:00:00:00
	R8188EU: set bssid:00:00:00:00:00:00
	R8188EU: nolinked power save enter
	==> rtl8188e_iol_efuse_patch 
	R8188EU: nolinked power save leave
	R8188EU: nolinked power save enter
	==> rtl8188e_iol_efuse_patch 
	R8188EU: nolinked power save leave
	R8188EU: nolinked power save enter
	==> rtl8188e_iol_efuse_patch 
	R8188EU: nolinked power save leave
	R8188EU: set bssid:84:7a:88:2d:7a:29
	R8188EU: set bssid:84:7a:88:2d:7a:29
	R8188EU: set ssid [TimmyHTC] fw_state=0x00000088
	R8188EU: set ssid [TimmyHTC] fw_state=0x00000088
	R8188EU: start auth
	R8188EU: auth success, start assoc
	R8188EU: assoc success
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:0, rate_bitmap:0x000ff005
	<6>UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
		     ==> rssi_level:1, rate_bitmap:0x000f0000
	R8188EU: set bssid:00:00:00:00:00:00
	R8188EU: set bssid:00:00:00:00:00:00
	R8188EU: nolinked power save enter
	==> rtl8188e_iol_efuse_patch 

	...

lwfinger, 2015-05-20:
On my system using V4.1.8 from my repo, I get:

[92729.337503] usbcore: registered new interface driver rtl8188eu
[92729.344533] rtl8188eu 3-14:1.0 wlp0s20u14: renamed from wlan0
[92729.401193] R8188EU: Loaded firmware file rtlwifi/rtl8188eufw.bin
[92729.401197] R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1
[92729.676633] ==> rtl8188e_iol_efuse_patch 
[92731.410543] R8188EU: rtw_set_802_11_connect(wlp0s20u14)  fw_state=0x00000008
[92731.485181] R8188EU: start auth
[92731.487182] R8188EU: auth success, start assoc
[92731.489919] R8188EU: rtw_cfg80211_indicate_connect(wlp0s20u14) BSS not found !!
[92731.489945] R8188EU: assoc success
[92731.493190] UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x03, mask:0x00000fff
         ==> rssi_level:0, rate_bitmap:0x00000ff5
[92731.495721] R8188EU: send eapol packet
[92731.505563] R8188EU: send eapol packet
[92731.505883] R8188EU: set pairwise key to hw: alg:2(WEP40-1 WEP104-5 TKIP-2 AES-4) camid:4
[92731.534471] R8188EU: set group key to hw: alg:2(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:2
[92731.534530] R8188EU: send eapol packet
[92754.319253] R8188EU: rtw_set_802_11_connect(wlp0s20u14)  fw_state=0x00000008
[92754.528836] R8188EU: start auth
[92754.532362] R8188EU: auth success, start assoc
[92754.538316] R8188EU: rtw_cfg80211_indicate_connect(wlp0s20u14) BSS not found !!
[92754.538354] R8188EU: assoc success
[92754.546948] UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
         ==> rssi_level:0, rate_bitmap:0x000ff005
[92754.549713] R8188EU: send eapol packet
[92754.558132] R8188EU: send eapol packet
[92754.558336] R8188EU: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) camid:4
[92754.559726] R8188EU: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:1
[92756.013210] UpdateHalRAMask8188EUsb => mac_id:0, networkType:0x0b, mask:0x000fffff
         ==> rssi_level:2, rate_bitmap:0x000ff000

The above is with a WPA2 connection to a Netgear WNDR3400 V2 AP. What are you using?

tisdall, 2015-05-20:
The first log snippet is WPA2 on D-Link DIR-825 rev. B1 with OpenWrt Attitude Adjustment 12.09-rc1.  The second is no encryption using my cell phone as a wifi "hotspot".

This is running on an ARM processor (I saw in another github issue that there may be issues there).

lwfinger, 2015-05-20:
The ARM CPUs do have different alignment requirements. That could certainly make a difference.

tisdall, 2015-05-20:
The default setting on my kernel is to trap alignment issues and fix them.
```
# cat /proc/cpu/alignment 
User:           0
System:         1
Skipped:        0
Half:           1
Word:           0
DWord:          0
Multi:          0
User faults:    2 (fixup)
```

(for any one else reading this)  This can be set with `echo 2 > /proc/cpu/alignment`

I guess I could try your oldest version in the 4.1.8 branch and if that works try bisecting until I find the commit causing the issue...    always a fun exercise.  Unless you can think of something else to try.

lwfinger, 2015-05-20:
Bisecting is what I would do. At least the symptoms are not subtle.

BTW, running with some other setting than "fixup" would let us find the alignment problems.

tisdall, 2015-05-21:
mode "3" is fixup with logging...  is the logging sufficient to figure out
where the bugs are?  If so, do those logs appear in syslog?

lwfinger, 2015-05-21:
Any logging will be in syslog; however, I have never been able to fix any bugs using Realtek's higher levels of logging. On occasion, their PCI group has been able to interpret them, but they know the internals of the chips. I'm not sure about the USB group, which is entirely separate. It is possible that you will be able to see what is wrong, particularly as you will have the logs from working and failing code.

tisdall, 2015-05-21:
In your own test environment, are you able to connect to unencrypted wifi
with 4.1.8?

lwfinger, 2015-05-21:
Absolutely.

BTW, why are you using a 32-bit OS? I am currently setting up a 32-bit Fedora test box to see if that makes a difference.

tisdall, 2015-05-21:
I'm using a 32-bit processor.  http://en.wikipedia.org/wiki/Allwinner_A1X

lwfinger, 2015-05-21:
Sorry, I crossed threads. In the other one, the guy has an i7 CPU, but is using a 32-bit system.

erwinkendo, 2015-05-29:
Same behaviour using 4.1.8 branch in ARM cpu, specially in AP mode.

R8188EU: ap recv disassoc reason code(1) sta:f8:e0:79:0f:04:fb                  
[  607.198444] R8188EU: set pairwise key to hw: alg:0(WEP40-1 WEP104-5 TKIP-2 AES-4) camid:4

lwfinger, 2015-05-30:
I have no ARM CPUs on which to test. My experience is that this does not happen on i686 or x86-64. Unfortunately, you will need to debug this. Sorry.

EdRiffo, 2015-07-03:
I have the same issue, any findings?

EdRiffo, 2015-07-03:
In AP mode that is...

tisdall, 2015-07-03:
@EdRiffo, sorry, I haven't had a chance to work on this and the way things are going it's looking like I won't be able to either.  I didn't use AP mode, though.