lwfinger/rtl8188eu #121: Problem with Hostapd 2.2 (kernel 3.10.17 Yocto linux)

blackdiamondute, 2015-05-27:
I have the v4.1.8_9499 version of the driver cross compiled and running on my device.  I am using hostapd v2.2.  My usb device is the TPlink TL-wn725N v2.  I can connect an access point and it works well.  But trying to setup my device as a host I am not allowed to connect.

My conf file is the following:
interface=wlan0
driver=nl80211
ssid=Setup
hw_mode=g
channel=7
ieee80211n=1
wme_enabled=1

macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0

#Sets WPA and WPA2 authentication
wpa=3
wpa_passphrase=12345678
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
max_num_sta=8
wpa_group_rekey=86400

when I run "hostapd -Bdd /etc/hostapd.conf" I get the following return:

random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd.conf
nl80211: Could not add multicast membership for vendor events: -2 (No such file or directory)
rfkill: initial event: idx=0 type=2 op=0 soft=0 hard=0
rfkill: initial event: idx=1 type=1 op=0 soft=0 hard=0
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Using driver-based off-channel TX
nl80211: interface wlan0 in phy phy0
nl80211: Set mode ifindex 4 iftype 3 (AP)
==> rtl8188e_iol_efuse_patch
R8188EU: nolinked power save leave
nl80211: Setup AP(wlan0) - device_ap_sme=1 use_monitor=1
nl80211: Enable Probe Request reporting nl_preq=0xd01938
nl80211: Register frame type=0x40 nl_handle=0xd01938 match=
nl80211: Add own interface ifindex 4
nl80211: if_indices[16]: 4
phy: phy0
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
nl80211: Regulatory information - country=00
nl80211: 2402-2472 @ 40 MHz 20 mBm
nl80211: 2457-2482 @ 40 MHz 20 mBm (no IR)
nl80211: 2474-2494 @ 20 MHz 20 mBm (no OFDM) (no IR)
nl80211: 5170-5250 @ 80 MHz 20 mBm (no IR)
nl80211: 5735-5835 @ 80 MHz 20 mBm (no IR)
nl80211: 57240-63720 @ 2160 MHz 0 mBm
nl80211: Added 802.11b mode based on 802.11g information
Allowed channel: mode=1 chan=1 freq=2412 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=2 freq=2417 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=3 freq=2422 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=4 freq=2427 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=5 freq=2432 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=6 freq=2437 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=7 freq=2442 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=8 freq=2447 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=9 freq=2452 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=10 freq=2457 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=11 freq=2462 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=1 freq=2412 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=2 freq=2417 MHz max_tx_power=20 dBm
AlloweR8188EU: assoc success
d channel: mode=0 chan=3 freq=242R8188EU: set group key to hw: alg:2(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:1
2 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=4 freq=2427 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=5 freq=2432 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=6 freq=2437 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=7 freq=2442 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=8 freq=2447 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=9 freq=2452 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=10 freq=2457 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=11 freq=2462 MHz max_tx_power=20 dBm
Completing interface initialization
Mode: IEEE 802.11g  Channel: 7  Frequency: 2442 MHz
DFS 0 channels required radar detection
nl80211: Set freq 2442 (ht_enabled=1, vht_enabled=0, bandwidth=20 MHz, cf1=2442 MHz, cf2=0 MHz)
RATE[0] rate=10 flags=0x1
RATE[1] rate=20 flags=0x1
RATE[2] rate=55 flags=0x1
RATE[3] rate=110 flags=0x1
RATE[4] rate=60 flags=0x0
RATE[5] rate=90 flags=0x0
RATE[6] rate=120 flags=0x0
RATE[7] rate=180 flags=0x0
RATE[8] rate=240 flags=0x0
RATE[9] rate=360 flags=0x0
RATE[10] rate=480 flags=0x0
RATE[11] rate=540 flags=0x0
hostapd_setup_bss(hapd=0xd00ed8 (wlan0), first=1)
wlan0: Flushing old station entries
nl80211: flush -> DEL_STATION wlan0 (all)
wlan0: Deauthenticate all stations
nl80211: sta_remove -> DEL_STATION wlan0 ff:ff:ff:ff:ff:ff --> -22 (Invalid argument)
wpa_driver_nl80211_set_key: ifindex=4 (wlan0) alg=0 addr=(nil) key_idx=0 set_tx=0 seq_len=0 key_len=0
wpa_driver_nl80211_set_key: ifindex=4 (wlan0) alg=0 addr=(nil) key_idx=1 set_tx=0 seq_len=0 key_len=0
wpa_driver_nl80211_set_key: ifindex=4 (wlan0) alg=0 addr=(nil) key_idx=2 set_tx=0 seq_len=0 key_len=0
wpa_driver_nl80211_set_key: ifindex=4 (wlan0) alg=0 addr=(nil) key_idx=3 set_tx=0 seq_len=0 key_len=0
Using interface wlan0 with hwaddr e8:de:27:a2:cd:81 and ssid "Setup"
Deriving WPA PSK based on passphrase
SSID - hexdump_ascii(len=5):
     53 65 74 75 70                                    Setup
PSK (ASCII passphrase) - hexdump_ascii(len=8): [REMOVED]
PSK (from passphrase) - hexdump(len=32): [REMOVED]
random: Got 20/20 bytes from /dev/random
Get randomness: len=32 entropy=0
GMK - hexdump(len=32): [REMOVED]
Get randomness: len=32 entropy=0
Key Counter - hexdump(len=32): [REMOVED]
WPA: Delay group state machine start until Beacon frames have been configured
nl80211: Set beacon (beacon_set=0)
nl80211: Beacon head - hexdump(len=56): 80 00 00 00 ff ff ff ff ff ff e8 de 27 a2 cd 81 e8 de 27 a2 cd 81 00 00 00 00 00 00 00 00 00 00 64 00 11 04 00 05 53 65 74 75 70 01 08 82 84 8b 96 0c 12 18 24 03 01 07
nl80211: Beacon tail - hexdump(len=133): 2a 01 04 32 04 30 48 60 6c 30 14 01 00 00 0f ac 02 01 00 00 0f ac 04 01 00 00 0f ac 02 0c 00 dd 16 00 50 f2 01 01 00 00 50 f2 02 01 00 00 50 f2 02 01 00 00 50 f2 02 2d 1a 0c 00 1f ff 00 00 00 01 00 00 00 00 00 96 00 01 00 00 00 00 00 00 00 00 00 00 3d 16 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 dd 18 00 50 f2 02 01 01 00 00 03 a4 00 00 27 a4 00 00 42 43 5e 00 62 32 2f 00
nl80211: ifindex=4
nl80211: beacon_int=100
nl80211: dtim_period=2
nl80211: ssid - hexdump_ascii(len=5):
     53 65 74 75 70                                    Setup
nl80211: hidden SSID not in use
nl80211: privacy=1
nl80211: auth_algs=0x1
nl80211: wpa_version=0x3
nl80211: key_mgmt_suites=0x2
nl80211: pairwise_ciphers=0x18
nl80211: group_cipher=0x8
WPA: Start group state machine to set initial keys
WPA: group state machine entering state GTK_INIT (VLAN-ID 0)
Get randomness: len=16 entropy=0
GTK - hexdump(len=32): [REMOVED]
WPA: group state machine entering state SETKEYSDONE (VLAN-ID 0)
wpa_driver_nl80211_set_key: ifindex=4 (wlan0) alg=2 addr=0x86298 key_idx=1 set_tx=1 seq_len=0 key_len=32
nl80211: KEY_DATA - hexdump(len=32): [REMOVED]
   broadcast key
nl80211: Set wlan0 operstate 0->1 (UP)
netlink: Operstate: ifindex=4 linkmode=-1 (no change), operstate=6 (IF_OPER_UP)
Failed to set TX queue parameters for queue 0.
Failed to set TX queue parameters for queue 1.
Failed to set TX queue parameters for queue 2.
Failed to set TX queue parameters for queue 3.
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: Setup of interface done.
ctrl_iface not configured!

After this It appears to function but every time I try to connect to my SSID I get the following results.
My client attempts to connect and the stops and I get the following debug.
R8188EU: ap recv disassoc reason code(8) sta:a8:06:00:c6:41:5f
R8188EU: set pairwise key to hw: alg:0(WEP40-1 WEP104-5 TKIP-2 AES-4) camid:4

richiejp, 2015-06-19:
Do you have DHCP set up? I found that some Android devices would disassociate if they didn't receive an IP address.

lwfinger, 2015-06-19:
Reason code 8 is "Disassociated because sending station is leaving (or has left) BSS". That is exactly the response that you would get if the client is trying to get an IP via DHCP, and there is no server running. My script for starting hostapd is as follows (Note the section for dhcpd):

#!/bin/sh
# Script to start/stop a hostapd-based access point
#
# Sample start call "control_ap start wlan0 eth0 encrypt"
# Stop with "control_ap stop"
#

ENCRYPT=$4
case "$1" in
start)
        if [ $# -le 2 ]
        then
          echo "Usage: $0 start AP_iface NET_iface <encrypt>"
          echo "       Optional encrypt is wpa, wpa2, wep, or none. If not given, wpa2 is assumed"
          exit 1
        fi
        if [ $# -eq 3 ]
        then
          ENCRYPT="wpa2"
        fi
;;
stop)
        if [ $# -ne 1 ]
        then
          echo "Usage: $0 stop"
          exit 1
        fi
;;
*)
        echo "Usage:"
        echo "$0 start AP-iface net_iface <encrypt>"
        echo "or"
        echo "$0 stop"
        exit 1
        ;;
esac

# Symbols for needed programs

IPTABLES=/usr/sbin/iptables
IFCONFIG=/sbin/ifconfig
DHCPD=/usr/sbin/dhcpd
HOSTAPD=/usr/sbin/hostapd
#HOSTAPD=/home/finger/hostapd-0.8/hostapd/hostapd
OPT=-ddK

# Symbols for AP and external interfaces

NET_AP=$2
NET_EXT=$3

# First 3 octets of IP address for the AP

AP_ADDR=192.168.0

# IP address for nameserver

NAME_SERVER=8.8.8.8

# AP Channel, SSID, Encryption method,  and Encryption secret

AP_CHANNEL=3
AP_SSID=test
WPA_SECRET=TheSecret
WEP_SECRET=123456789a

case "$1" in
start)
[... 97 lines cut ...]
        echo "wmm_ac_vo_aifs=2" >> ~/hostapd.conf
        echo "wmm_ac_vo_cwmin=2" >> ~/hostapd.conf
        echo "wmm_ac_vo_cwmax=3" >> ~/hostapd.conf
        echo "wmm_ac_vo_txop_limit=47" >> ~/hostapd.conf
        echo "wmm_ac_vo_acm=0" >> ~/hostapd.conf
        echo "ieee80211n=1" >> ~/hostapd.conf
        echo "ht_capab=[HT40+][SHORT-GI-40][DSSS_CCK-40]" >> ~/hostapd.conf
        echo "logger_syslog=-1" >> ~/hostapd.conf
        echo "logger_syslog_level=0" >> ~/hostapd.conf
        echo "logger_stdout=-1" >> ~/hostapd.conf
        echo "logger_stdout_level=0" >> ~/hostapd.conf
        echo "ctrl_interface=/var/run/hostapd" >> ~/hostapd.conf
        echo "ctrl_interface_group=0" >> ~/hostapd.conf
        # Bring up hostapd
        $HOSTAPD $OPT -B ~/hostapd.conf
        ;;
stop)
        echo "Stopping AP mode"
        # Stop hostapd and dhcpd daemons
        killproc hostapd
        killproc dhcpd
        rm -f ~/hostapd.conf
        rm -f ~/dhcpd.conf
        ;;
esac

blackdiamondute, 2015-06-22:
Thank you I will give this a try.

 

- Michael Diamond