lwfinger/rtl8188eu #122: version magic '4.0.4-301.fc22.i686 SMP mod_unload 686 ' should be '4.0.4-301.fc22.i686+debug SMP mod_unload 686 '

adals, 2015-05-29:
I installed driver and run ok I did 
 
make all
sudo make install 
sudo modprobe 8188eu 

but when reboot my desktop it does not work and dmesg gave me this msg

8188eu: version magic '4.0.4-301.fc22.i686 SMP mod_unload 686 ' should be '4.0.4-301.fc22.i686+debug SMP mod_unload 686 '

and when I did 

sudo modprobe 8188eu 
modprobe: ERROR: could not insert '8188eu': Exec format error

please help me

adals, 2015-05-29:
sorry 
solved it

thanks

Yoendric, 2018-05-10:
Hi.
I have the same problem. I could comment how you could solve it. Thank you

lwfinger, 2018-05-10:
Messages like these arise from having kernel headers different from the kernel booted version. They must be the same. That is a rule that cannot be broken without introducing the likeliness  of system crashes.

closed 2023-05-05 by adals