lwfinger/rtl8188eu #124: Scanning for networks... ioctl[SIOCSIWAP]: Operation not permitted

GAZ082, 2015-06-19:
Hi there. I have a cubox-i4 pro and cant get a TPLINK wifi dongle to work.

When running wifi-menu i get:

Scanning for networks... ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
done

Then the wifi-menu shows the available networks but with zero strenght signals.
[image]

I wonder if this is a driver issue or may be something else. Tips?

Thanks!

lwfinger, 2015-06-19:
Since you posted here, I assume you are running a driver from this repo. You should, however, specify what branch you are using. I recommend v4.1.8_9499 in most cases, but it is known to fail on big-endian machines. That should not affect your Cortex A9 as I think it is little endian.

I am not familiar with the wifi-menu utility that you posted. Can you open a terminal on that box? If so, please run the command 'sudo iwlist scan' and post the output.

GAZ082, 2015-06-19:
Sure, I will post more info
- I compiled the main branch, i believe is the version you said.
- 3.14.44-1-ARCH

here is the output with my wifi SSID:

wlan0     Scan completed :
          Cell 01 - Address: 48:F8:B3:35:AA:54
                    ESSID:"mO_Om"
                    Protocol:IEEE 802.11bgn
                    Mode:Master
                    Frequency:2.452 GHz (Channel 9)
                    Encryption key:on
                    Bit Rates:144 Mb/s
                    Extra:rsn_ie =30140100000fac040100000fac040100000fac020c00
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    Quality:0  Signal level:0  Noise level:0

Thank you for your support.

lwfinger, 2015-06-19:
It looks as if you got the "master" branch. That is not the one I said. The one you have does not return any signal strength. Perhaps that confuses your user space.

GAZ082, 2015-06-19:
DAMN IT WORKS! I'm SO SO HAPPY! Thanks a lot man, been struggling with this for months!

GAZ082, 2015-06-19:
Oh wait, i'm getting "connecting failed" with wifi-menu, but at least i do not have the initial error and do have wifi signal levels:

 Cell 04 - Address: 48:F8:B3:35:AA:54
                    ESSID:"mO_Om"
                    Protocol:IEEE 802.11bgn
                    Mode:Master
                    Frequency:2.452 GHz (Channel 9)
                    Encryption key:on
                    Bit Rates:144 Mb/s
                    Extra:rsn_ie=30140100000fac040100000fac040100000fac020c00
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    Quality=0/100  Signal level=60/100
                    Extra:fm=0003

GAZ082, 2015-06-19:
Never mind, this is not related. Again, thank you!!

closed 2015-06-19 by GAZ082