lwfinger/rtl8188eu #125: RTL8188EUS stopped working on debian testing (kernel 3.16.0)

jochym, 2015-06-21:
My TL-WN725Nv2 stopped working recently on debian testing with 3.16.0 kernel.
I have refreshed the source (from master) and recompiled (dkms) modules. Still no connection.
The driver and firmware loads and interface shows up but cannot connect to my AP.
It also does not show the signal strength.
How can I help to debug this issue?

lwfinger, 2015-06-21:
The "master" branch has never shown other than zero signal strength. If that hurts your user-space connection machinery, then switch to the "v4.1.8_9499" branch. That one provides signal info.

jochym, 2015-06-22:
Ok. I will try the other branch, but the problem was no connection - the signal strength was indeed always absent but this is not a huge problem. I have tried to connect to open and WPA2 APs - no connection. I have no precise records but it used to work a month or two ago.

jochym, 2015-06-22:
I have tried the 4.1 branch. It showed the signal level but still no connection at all. 
This is the same on recent fedora 22 (kernel 4.0.5-300).
How can I help?