lwfinger/rtl8188eu #129: Oops/Segfault on Raspberry Pi / Raspbian / Debian Wheezy

sebastianbergmann, 2015-07-09:
## Relevant section

    [   74.249606] Unable to handle kernel paging request at virtual address 7e32d254
    [   74.261993] pgd = dd18c000
    [   74.270985] [7e32d254] *pgd=00000000
    [   74.280001] Internal error: Oops: 5 [#1] PREEMPT ARM
    [   74.289880] Modules linked in: 8188eu(O+) cfg80211 rfkill snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd evdev joydev uio_pdrv_genirq uio
    [   74.308698] CPU: 0 PID: 2329 Comm: modprobe Tainted: G           O   3.18.11+ #781
    [   74.321564] task: dd245100 ti: ddfd4000 task.ti: ddfd4000
    [   74.332279] PC is at load_module+0x1948/0x1f1c
    [   74.341811] LR is at load_module+0x1934/0x1f1c
    [   74.351490] pc : [<c00886d0>]    lr : [<c00886bc>]    psr: 30000013
    [   74.351490] sp : ddfd5e88  ip : bf1a81c0  fp : ddfd5f44
    [   74.373345] r10: bf1a8058  r9 : 00000000  r8 : bf1a8064
    [   74.383843] r7 : c0557314  r6 : dd3ef4e0  r5 : 7e32d240  r4 : ddfd5f48
    [   74.395550] r3 : 00000000  r2 : 00000000  r1 : da5b7dc0  r0 : c0814280
    [   74.407339] Flags: nzCV  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
    [   74.419872] Control: 00c5387d  Table: 1d18c008  DAC: 00000015
    [   74.431055] Process modprobe (pid: 2329, stack limit = 0xddfd41b0)
    [   74.442519] Stack: (0xddfd5e88 to 0xddfd6000)
    [   74.452377] 5e80:                   bf1a8064 00007fff c0085714 c02fcd88 00000013 00000000
    [   74.466125] 5ea0: df9ae000 ddfd5f7c ddfd5f50 ddfd5eb8 00000000 bf1a8064 ddfd4008 bf1a80a0
    [   74.479747] 5ec0: bf1a81c0 00000000 b6cd0000 ddfd4000 00002db0 00000000 00000000 00000000
    [   74.493403] 5ee0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
    [   74.507108] 5f00: 00000000 00000000 00000000 00000000 00000000 00000000 00000080 000e9725
    [   74.520767] 5f20: b6bec000 b6f02948 00000080 c000eaa8 ddfd4000 00000000 ddfd5fa4 ddfd5f48
    [   74.534393] 5f40: c0088d90 c0086d94 df9ae000 000e9725 dfa4f344 dfa4f1d6 dfa8d044 000a1ccc
    [   74.547937] 5f60: 000b245c 00000000 00000000 00000000 00000023 00000024 00000018 0000001c
    [   74.561425] 5f80: 0000000c 00000000 00000000 00000000 00060000 b74df608 00000000 ddfd5fa8
    [   74.574972] 5fa0: c000e800 c0088cb0 00000000 00060000 b6bec000 000e9725 b6f02948 b6bec000
    [   74.588535] 5fc0: 00000000 00060000 b74df608 00000080 b74ded58 000e9725 b6f02948 00000000
    [   74.602120] 5fe0: 00000000 befe090c b6ef9fb4 b6e64e94 60000010 b6bec000 00000000 00000000
    [   74.615733] [<c00886d0>] (load_module) from [<c0088d90>] (SyS_init_module+0xec/0x100)
    [   74.628903] [<c0088d90>] (SyS_init_module) from [<c000e800>] (ret_fast_syscall+0x0/0x48)
    [   74.642434] Code: e51bc084 e15c0005 e2455008 0a000009 (e5953014) 
    [   74.659323] ---[ end trace 9b95a96a348bea3c ]---

## Full log

    [    0.000000] Booting Linux on physical CPU 0x0
    [    0.000000] Initializing cgroup subsys cpu
    [    0.000000] Initializing cgroup subsys cpuacct
    [    0.000000] Linux version 3.18.11+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #781 PREEMPT Tue Apr 21 18:02:18 BST 2015
    [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
    [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
    [    0.000000] Machine model: Raspberry Pi Model B Plus
    [    0.000000] cma: Reserved 8 MiB at 0x1e800000
    [    0.000000] Memory policy: Data cache writeback
    [    0.000000] On node 0 totalpages: 126976
    [    0.000000] free_area_init_node: node 0, pgdat c083d364, node_mem_map de414000
    [    0.000000]   Normal zone: 992 pages used for memmap
    [    0.000000]   Normal zone: 0 pages reserved
    [    0.000000]   Normal zone: 126976 pages, LIFO batch:31
    [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
    [    0.000000] pcpu-alloc: [0] 0 
    [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 125984
    [    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1776 bcm2708_fb.fbheight=952 bcm2708.boardrev=0x10 bcm2708.serial=0x7e814aa1 smsc95xx.macaddr=B8:27:EB:81:4A:A1 bcm2708_fb.fbswap=1 bcm2708.disk_led_gpio=47 bcm2708.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p6 rootfstype=ext4 elevator=deadline rootwait
    [    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
    [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
    [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
    [    0.000000] Memory: 486044K/507904K available (5881K kernel code, 348K rwdata, 1868K rodata, 336K init, 733K bss, 21860K reserved)
    [    0.000000] Virtual kernel memory layout:
    [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    [    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
    [    0.000000]     vmalloc : 0xdf800000 - 0xff000000   ( 504 MB)
    [    0.000000]     lowmem  : 0xc0000000 - 0xdf000000   ( 496 MB)
    [    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
    [    0.000000]       .text : 0xc0008000 - 0xc07997a0   (7750 kB)
    [    0.000000]       .init : 0xc079a000 - 0xc07ee000   ( 336 kB)
    [    0.000000]       .data : 0xc07ee000 - 0xc084513c   ( 349 kB)
[... 200 lines cut ...]
    [   74.351490] pc : [<c00886d0>]    lr : [<c00886bc>]    psr: 30000013
    [   74.351490] sp : ddfd5e88  ip : bf1a81c0  fp : ddfd5f44
    [   74.373345] r10: bf1a8058  r9 : 00000000  r8 : bf1a8064
    [   74.383843] r7 : c0557314  r6 : dd3ef4e0  r5 : 7e32d240  r4 : ddfd5f48
    [   74.395550] r3 : 00000000  r2 : 00000000  r1 : da5b7dc0  r0 : c0814280
    [   74.407339] Flags: nzCV  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
    [   74.419872] Control: 00c5387d  Table: 1d18c008  DAC: 00000015
    [   74.431055] Process modprobe (pid: 2329, stack limit = 0xddfd41b0)
    [   74.442519] Stack: (0xddfd5e88 to 0xddfd6000)
    [   74.452377] 5e80:                   bf1a8064 00007fff c0085714 c02fcd88 00000013 00000000
    [   74.466125] 5ea0: df9ae000 ddfd5f7c ddfd5f50 ddfd5eb8 00000000 bf1a8064 ddfd4008 bf1a80a0
    [   74.479747] 5ec0: bf1a81c0 00000000 b6cd0000 ddfd4000 00002db0 00000000 00000000 00000000
    [   74.493403] 5ee0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
    [   74.507108] 5f00: 00000000 00000000 00000000 00000000 00000000 00000000 00000080 000e9725
    [   74.520767] 5f20: b6bec000 b6f02948 00000080 c000eaa8 ddfd4000 00000000 ddfd5fa4 ddfd5f48
    [   74.534393] 5f40: c0088d90 c0086d94 df9ae000 000e9725 dfa4f344 dfa4f1d6 dfa8d044 000a1ccc
    [   74.547937] 5f60: 000b245c 00000000 00000000 00000000 00000023 00000024 00000018 0000001c
    [   74.561425] 5f80: 0000000c 00000000 00000000 00000000 00060000 b74df608 00000000 ddfd5fa8
    [   74.574972] 5fa0: c000e800 c0088cb0 00000000 00060000 b6bec000 000e9725 b6f02948 b6bec000
    [   74.588535] 5fc0: 00000000 00060000 b74df608 00000080 b74ded58 000e9725 b6f02948 00000000
    [   74.602120] 5fe0: 00000000 befe090c b6ef9fb4 b6e64e94 60000010 b6bec000 00000000 00000000
    [   74.615733] [<c00886d0>] (load_module) from [<c0088d90>] (SyS_init_module+0xec/0x100)
    [   74.628903] [<c0088d90>] (SyS_init_module) from [<c000e800>] (ret_fast_syscall+0x0/0x48)
    [   74.642434] Code: e51bc084 e15c0005 e2455008 0a000009 (e5953014) 
    [   74.659323] ---[ end trace 9b95a96a348bea3c ]---

sebastianbergmann, 2015-07-09:
Closing because I neither use the OS version nor the USB WLAN dongle anymore.

closed 2015-07-09 by sebastianbergmann

lwfinger, 2015-07-09:
If you are not using the OS version or the dongle any more, why did you open this issue?

sebastianbergmann, 2015-07-09:
Six hours ago I was still using them, sorry for the noise.