lwfinger/rtl8188eu #131: DKMS driver does not work, manual compiling yes,

GAZ082, 2015-07-20:
Tried with DMKS the 4.1.88, did not work. Throw several errors i had with the master branch, now i recompiled by had (make, make install) and worked as expected.

Perhaps i should DMKS another version and not 1.0?

Puzzled.

drencrom, 2015-08-28:
Have you ffound the problem? I compiled the master branch without problems with dkms. Then tried the 4.1.8_9499 and failed (that is ok, it also fails manually) but now the master also fails with dkms.
Compiling manualy works ok

lwfinger, 2015-08-28:
If you are asking me, I have never learned the details of dkms configuration files. Someone else will need to figure that out.

If you are getting failures of a manual build, then please post the error messages.

GAZ082, 2015-08-28:
My current situation is the same, DKMS does not work, manual yes.

drencrom, 2015-08-28:
I solved it by editing the dkms config file and changing the version to 1.1. I suspect this must be a dkms bug. I also disabled the REMAKE_INITRD option as the manual installation works without adding the module to the initrd.

drencrom, 2015-08-28:
Its a bit off topic but since @lwfinger asked these are the errors I get with the 4.1.8_9499 branch. I suppose it does not work with kernel 3.10.

```
/root/rtl8188eu-4.1.8_9499/os_dep/ioctl_cfg80211.c:736:12: error: ‘struct wireless_dev’ has no member named ‘sme_state’
   if (pwdev->sme_state==CFG80211_SME_CONNECTING)

/root/rtl8188eu-4.1.8_9499/os_dep/ioctl_cfg80211.c:3761:2: error: too few arguments to function ‘cfg80211_rx_mgmt’
  rtw_cfg80211_rx_mgmt(padapter, freq, 0, pmgmt_frame, frame_len, GFP_ATOMIC);`
```

lwfinger, 2015-08-28:
Are you a Ubuntu user? These errors look like the ones that have developed because some distros (mainly Ubuntu) are backporting API changes to older kernels. After wasting a lot of time trying to understand this issue, I decided not to try to keep up with them.

drencrom, 2015-08-28:
This is centos 7.

lwfinger, 2015-08-28:
On my vanilla 3.10 source, both branches build without error. Apparently, CentOS also backports API changes the same as Ubuntu does. The master branch is OK because the API changes affect cfg80211 code, which the master branch does not use.

GAZ082, 2015-08-28:
I'm using Archlinux x64.