lwfinger/rtl8188eu #132: Unable to build under fedora22

ofayans, 2015-07-22:
[13:33:47]ofayans@localhost:~/Realtek/rtl8188eu/rtl8188eu-master]$ make all
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.0.8-300.fc22.x86_64/build M=/home/ofayans/Realtek/rtl8188eu/rtl8188eu-master  modules
make[1]: Entering directory '/usr/src/kernels/linux-4.0.8'

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /home/ofayans/Realtek/rtl8188eu/rtl8188eu-master/core/rtw_ap.o
/bin/bash: ./scripts/recordmcount: No such file or directory
scripts/Makefile.build:258: recipe for target '/home/ofayans/Realtek/rtl8188eu/rtl8188eu-master/core/rtw_ap.o' failed
make[2]: *** [/home/ofayans/Realtek/rtl8188eu/rtl8188eu-master/core/rtw_ap.o] Error 1
Makefile:1390: recipe for target '_module_/home/ofayans/Realtek/rtl8188eu/rtl8188eu-master' failed
make[1]: *** [_module_/home/ofayans/Realtek/rtl8188eu/rtl8188eu-master] Error 2
make[1]: Leaving directory '/usr/src/kernels/linux-4.0.8'
Makefile:147: recipe for target 'modules' failed
make: *** [modules] Error 2

============================================================
There is a scripts/recordmcount.pl file under the linux source folder, so I created a symbolic link to it:
[13:34:48]ofayans@localhost:~/Realtek/rtl8188eu]$ cd /usr/src/kernels/linux-4.0.8/scripts/
[13:35:01]ofayans@localhost:/usr/src/kernels/linux-4.0.8/scripts]$ ls -s recordmcount.pl recordmcount
And tried again, but the build failed anyway:
=============================================================
[13:35:34]ofayans@localhost:/usr/src/kernels/linux-4.0.8/scripts]$ cd -
/home/ofayans/Realtek/rtl8188eu
[13:35:38]ofayans@localhost:~/Realtek/rtl8188eu]$ make all
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.0.8-300.fc22.x86_64/build M=/home/ofayans/Realtek/rtl8188eu  modules
make[1]: Entering directory '/usr/src/kernels/linux-4.0.8'

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /home/ofayans/Realtek/rtl8188eu/core/rtw_ap.o
usage: recordmcount arch endian bits objdump objcopy cc ld nm rm mv is_module inputfile
version: 0.1
scripts/Makefile.build:258: recipe for target '/home/ofayans/Realtek/rtl8188eu/core/rtw_ap.o' failed
make[2]: *** [/home/ofayans/Realtek/rtl8188eu/core/rtw_ap.o] Error 1
Makefile:1390: recipe for target '_module_/home/ofayans/Realtek/rtl8188eu' failed
make[1]: *** [_module_/home/ofayans/Realtek/rtl8188eu] Error 2
make[1]: Leaving directory '/usr/src/kernels/linux-4.0.8'
Makefile:147: recipe for target 'modules' failed
make: *** [modules] Error 2

lwfinger, 2015-07-22:
Have you installed the kernel headers? These kinds of errors usually mean that the answer is "no".

As a final check, does the command "ls /lib/modules/`uname -r`/build" show a directory named "include"? Those ` characters are the back tick, usually found next to the number 1.

ofayans, 2015-07-22:
I have kernel headers installed. However, /lib/modules/4.0.7-300.fc22.x86_64/build initially was a symlink pointing to a nonexistent folder /usr/src/kernels/4.0.7-300.fc22.x86_64
I tried two things that did not help:
1. Removed this symlink and created a new one pointing to /usr/include/linux (a folder belonging to kernel-headers package), then tried to build and failed:
make[1]: *** No rule to make target 'modules'.  Stop.
2. downloaded a vanilla kernel of the same version as my running kernel, unpacked it to /usr/src/kernels, created a 'build' symlink to it in /lib/modules/4.0.7-300.fc22.x86_64/ folder, then in the kernel source tree issued `make oldconfig && make modules_prepare`
Then tried to build again which resulted in the error I described in initial message

lwfinger, 2015-07-22:
Are you downloading the zip file? That is the worst way to do it. Use the "git clone" to get the source. That way, you only need to "git pull" to get any updates, and you can switch branches. In fact, the v4.1.8_9499 branch is better than master. Try that and see if the error persists.

It seems as if your headers are properly installed. I have no idea where those error messages are coming from. I do not have any experience with F22. What version of gcc is being used?