lwfinger/rtl8188eu #136: Kernel panic in recv_chk_defrag

jdyatsin, 2015-08-12:
Hi, 
Thank you lwfinger for providing this driver.
We are occasionally seeing a crash, and would be very grateful if you could help us with this.

Flags: Nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Control: 10c53c7d  Table: 866c8059  DAC: 00000015
Process sshd (pid: 520, stack limit = 0xc66c22e8)
Stack: (0xc66c3e40 to 0xc66c4000)
3e40: c059f2c0 c6117cb8 39ac18ac 00000000 c6615300 c66d7c44 c66d7c30 c665c460
3e60: c88e5000 c8929ae0 c88e5000 c66d7c28 c66d7c28 00000000 00000b10 c88e6518
3e80: 00004000 c023c278 c6615300 c88e5000 00000000 00000000 c8929ae0 c88e5000
3ea0: c6240000 c8929b44 00000000 c62d0040 c62d0028 c025fe80 008e5000 c62d0040
3ec0: 006d7cb0 c8929ae0 c6615300 c88e5000 c059d974 00000082 00000100 c620c600
3ee0: c62d0028 c026565c 00000020 c053c464 c056e620 00003e58 00000000 c88e6518
3f00: 0p000090 00000001 00000100 00000100 c6240000 c62d0040 c88e6500 c6242cd8
3f20: 00000121 c05439f8 c058f660 00000000 00000000 c058f660 00000102 c058f684
3f40: 00000000 c004844c c00483d0 00000001 00000018 c66c2000 00000009 c0048560
3f60: c0548550 c66c2000 c61e2040 00000006 00000000 0000006f 00000000 0000006f
3f80: 10000001 00000000 c66c2000 e72acf39 2a1c1598 c00489e0 00000000 c0028064
3fa0: 000100;0 ffffffff f2022100 c002d5e8 e913874a 6f727420 2aba6c3b be9141e0
3fc0: be9141e4 e72acf39 b00327c8 153b27ea 247c8955 16588ede e72acf39 2a1c1598
3fe0: 4b3cc91= be9140d8 401fbe81 401fb8b2 00010030 ffffffff 00000000 00000000
[<ã023ad00>] (recvframe_chk_defrag+0x1a0/0xa38) from [<c023c278>] (rtw_recv_entry+0x3f0/0x1520)
[<c023c278>] (rtw_recv_entry+0x3f0/0x1520) from [<c026565c>] (rtl8188eu_recv_tasklet+0x434/0x608)
[<c026565c>] (rtl8188eu_recv_tasklet+0x434/0x608) from [<c00486c>] (tasklet_action+0x7c/0xc0)
[<c004844c>] (tasklet_action+0x7c/0xc0) from [<c0048560>] (__do_softirq+0x84/0x110)
[<c0048560>] (__do_softirq+0x84/0x110) from [<c00489e0>] (irq_exit+0x44/0xac)
[<c00489e0>] (irq_exit+0x44/0xac) from [<c0028064>] (asm_do_IRQ+0x64/0x84)
[<c0028064>] (asm_do_IRQ+0x64/0x84) from [<c002d5e8>] (__irq_usr+0x48/0xc0)
3fa0: e913874a 6f727420 2aba6c3b be9141e0
3fc0: be9141e4 e72acf39 b00327c8 153b27ea 247c8955 16588ede e72acf39 2a1c1598
3fe0: 4b3ac919 be9140d8 401fbe81 401fb8b2 00010030 ffffffff
Code: e3a01000 e1a02008 eb06aef0 e3a03000 (e8930006)
---[ end trace 11adfb351172f51c ]---
Kernel panic - not syncing: Fatal exception in interrupt

We are running your master branch, git #:3091828c8f4b4a01cbec6025128bf77e6e7b9f97

Running in AP mode, with hostapd-v2.2. 
Linux 3.0.15 Kernel
ARM board (Freescale Vybrid)

We have not been able to find a way to trigger the crash. The units can run for weeks before this happens.

This is my hostapd.conf:

interface=wlan0
ssid=DevelWifi
channel=7
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=thisIsATest
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
driver=rtl871xdrv
hw_mode=g
wpa_group_rekey=3600
eap_server=1
ctrl_interface=/firmware/data/hostapd_ctrl

Let me know if you need additional information.

Thank you in advance.

David

lwfinger, 2015-08-12:
Any chance of converting that stack dump into a traceback? Without this info, it is impossible to know what is going wrong.

Have you tried the v4.1.8_9499 branch?

netllama, 2015-08-12:
i'm lurking here, but wondering how 'main' differs from 'v4.1.8_9499'?  The
README doesn't specifically recommend using any branch, so everyone
defaults to main.

jdyatsin, 2015-08-12:
>Any chance of converting that stack dump into a traceback? Without this info, it is impossible to know >what is going wrong.
>Have you tried the v4.1.8_9499 branch?

I will try both, and get let you know

jdyatsin, 2015-08-12:
Updated original post because function names were obfuscated

lwfinger, 2015-08-12:
Thanks for the extra traceback info. I wondered about those empty "from:"  lines.

If possible, it would be helpful to run gdb on the .ko file, and show the results for
l *recvframe_chk_defrag+0x1a0. That should show the line that actually faulted.

@netllama - The v4.1.8_9499 branch is a later release from Realtek. I think its implementation of cfg80211 calls is better, but otherwise, I do not know what they changed.

jdyatsin, 2015-08-12:
This is what I got from gdb:

/opt/cross-gcc-4.7.2-armv7-hard/bin/arm-linux-gnueabihf-gdb core/rtw_recv.o
GNU gdb (crosstool-NG linaro-1.13.1-2012.09-20120921 - Linaro GCC 2012.09) 7.5-2012.09
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-build_pc-linux-gnu --target=arm-linux-gnueabihf".
For bug reporting instructions, please see:
<https://bugs.launchpad.net/gcc-linaro>...
Reading symbols from /data/builds/ECU-WM/trunk/sources/kernel/drivers/net/wireless/rtl8188eu/core/rtw_recv.o...done.
(gdb) list *(recvframe_ck_defrag+0x1a0)
No symbol "recvframe_ck_defrag" in current context.
(gdb) list *(recvframe_chk_defrag+0x1a0)
0x1394 is in recvframe_chk_defrag (include/linux/list.h:101).
96       * in an undefined state.
97       */
98      #ifndef CONFIG_DEBUG_LIST
99      static inline void __list_del_entry(struct list_head *entry)
100     {
101             __list_del(entry->prev, entry->next);
102     }
103
104     static inline void list_del(struct list_head *entry)
105     {
(gdb) list *(rtw_recv_entry+0x3f0)
0x290c is in rtw_recv_entry (drivers/net/wireless/rtl8188eu/core/rtw_recv.c:1139).
1134            struct sta_info *psta;
1135
1136            RT_TRACE(_module_rtl871x_recv_c_, _drv_info_, ("+validate_recv_mgnt_frame\n"));
1137
1138            precv_frame = recvframe_chk_defrag(padapter, precv_frame);
1139            if (precv_frame == NULL) {
1140                    RT_TRACE(_module_rtl871x_recv_c_, _drv_notice_, ("%s: fragment packet\n", __func__));
1141                    return _SUCCESS;
1142            }
1143
(gdb)

The kernel was compiled with -Os. I will try to re-compile without optimizations to see if we can get the actual line number in recvframe_chk_defrag on the next crash. But it might take 1-2 weeks before we get a crash

lwfinger, 2015-08-12:
I can wait. :)

jdyatsin, 2015-08-18:
Hi lwfinger,
We just had a few coredumps in the past 1 hour:
[ 1118.398715] Stack: (0xc6435e08 to 0xc6436000)
[ 1118.403120] 5e00:                   00000000 c62bfe00 c61dee00 c64488c0 ffdfd0f8 c023e608
[ 1118.411370] 5e20: c62bfe00 c02dbef4 00000302 c892e00c c6435e3a c8923390 c88af000 c88b0518
[ 1118.419619] 5e40: c05dfa50 00000001 000072b9 0000000b c88b0500 c025d434 ffff88cc ffffffff
[ 1118.427869] 5e60: c88b222c c025c5e8 000082f7 c88af000 c8923390 00000001 c88b0518 0000ffff
[ 1118.436117] 5e80: 000072b9 0000000b c88b0500 c025e760 c88af000 c892e004 c89233f4 c62e8040
[ 1118.444368] 5ea0: c660322e c6220000 c66032ac c660322e 0000ffff c661c0c0 00000020 c88af09c
[ 1118.452616] 5ec0: 0034899c c892e004 ffffffff c8923390 c661c0c0 0000007e c62e8028 000000b6
[ 1118.460866] 5ee0: 00000100 00000001 c66032ac c028d298 c6434000 c02eeee8 00003e58 00000006
[ 1118.469114] 5f00: 00000100 c88af000 c620f9c0 c88b0518 c6220000 c62e8040 c88b05c0 c88b0500
[ 1118.477364] 5f20: 00000000 00000000 c0585a78 c05d1720 00000000 c6434000 c05d1720 00000102
[ 1118.485613] 5f40: 00000006 c004adbc c05858e0 00000001 00000018 c05d1760 c05d1744 c004aefc
[ 1118.493861] 5f60: c6434000 40036000 00000000 0000000a 00010319 c6434000 0000006c 00000000
[ 1118.502111] 5f80: 00000001 00000000 c6434000 40036000 00000000 c004b3b4 c0595c2c c0029034
[ 1118.510361] 5fa0: ffffffff f2022100 0000006c c002ebe8 003afb9c 003afadc fffffffc 003fee20
[ 1118.518610] 5fc0: 003afa98 00000080 00000003 00000000 00000000 00000000 40036000 00000000
[ 1118.526859] 5fe0: 003afbc0 beb68c84 0015708f 404625cc 800f0010 ffffffff 00000000 00000000
[ 1118.535160] [<c025c934>] (recvframe_chk_defrag+0xf0/0xa6c) from [<c025d434>] (recv_func_posthandle+0xf4/0x7d8)
[ 1118.545270] [<c025d434>] (recv_func_posthandle+0xf4/0x7d8) from [<c025e760>] (rtw_recv_entry+0xbb8/0x1648)
[ 1118.555025] [<c025e760>] (rtw_recv_entry+0xbb8/0x1648) from [<c028d298>] (rtl8188eu_recv_tasklet+0x254/0x65c)
[ 1118.565036] [<c028d298>] (rtl8188eu_recv_tasklet+0x254/0x65c) from [<c004adbc>] (tasklet_action+0x60/0xc0)
[ 1118.574786] [<c004adbc>] (tasklet_action+0x60/0xc0) from [<c004aefc>] (__do_softirq+0x94/0x128)
[ 1118.583572] [<c004aefc>] (__do_softirq+0x94/0x128) from [<c004b3b4>] (irq_exit+0x90/0x98)
[ 1118.591833] [<c004b3b4>] (irq_exit+0x90/0x98) from [<c0029034>] (asm_do_IRQ+0x34/0x84)
[ 1118.599833] [<c0029034>] (asm_do_IRQ+0x34/0x84) from [<c002ebe8>] (__irq_usr+0x48/0xc0)
[ 1118.607899] Exception stack(0xc6435fb0 to 0xc6435ff8)

They are all at:(recvframe_chk_defrag+0xf0)

Reading symbols from /data/builds/ECU-WM/branches/test-rtl8188-master/sources/kernel/drivers/net/wireless/rtl8188eu/core/rtw_recv.o...done.
(gdb) list *(recvframe_chk_defrag+0xf0)
0x138c is in recvframe_chk_defrag (drivers/net/wireless/rtl8188eu/core/rtw_recv.c:1480).
1475            plist = phead->next;
1476            prframe = container_of(plist, struct recv_frame, list);
1477            prframe = (struct recv_frame *)pfhdr;
1478            list_del_init(&(prframe->list));
1479
1480            if (curfragnum != pfhdr->attrib.frag_num) {
1481                    /* the first fragment number must be 0 */
1482                    /* free the whole queue */
1483                    rtw_free_recvframe(prframe, pfree_recv_queue);
1484                    rtw_free_recvframe_queue(defrag_q, pfree_recv_queue);

It looks like pfhdr is NULL or invalid?

lwfinger, 2015-08-19:
I expect that it will be NULL. I just committed a fix that will protect against that. Let me know what happens.

jdyatsin, 2015-08-19:
I was using the master branch, because the v4.1.8_9499 does not compile on 3.0.15 kernel (struct station_info in cfg80211.h does not have members assoc_req_ies and assoc_req_ies_len).
Is it ok if I just add this 2 members to my struct station_info?

For the master branch in recvframe_defrag, I think this line:
```
prframe = (struct recv_frame *)pfhdr;
```
should be:
```
pfhdr = (struct recv_frame *)prframe;
```

lwfinger, 2015-08-19:
No,  the mistake was one line earlier. It should be

        pfhdr = container_of(plist, struct recv_frame, list);
        prframe = (struct recv_frame *)pfhdr;

I pushed this fix just now after recovering from a typo.

jdyatsin, 2015-09-11:
We have not seen any kernel panics so far, so I am assuming this issue is closed for now.
Thank you

closed 2015-09-11 by jdyatsin