lwfinger/rtl8188eu #137: tx hang after REG_TXDMA_STATUS:0x00004200 on ARM platform

bsansoni, 2015-08-22:
Hi,
I am currently using the current rtl8188eu for the USB wifi dongle TL-WN725N V2 on Cubieboard (A10 ARM allwinner platform).
I used it as AP. It works but after a while (somtime 20 minutes, sometime 1h), the accesses to the AP is not possible. (by ping, etc ...).
I activated some debug and I got this

<6>R8188EU: rtl8188e_sreset_xmit_status_check REG_TXDMA_STATUS:0x00003000
[  742.091842] R8188EU: rtl8188e_sreset_xmit_status_check REG_TXDMA_STATUS:0x00003000
<6>R8188EU: rtl8188e_sreset_xmit_status_check REG_TXDMA_STATUS (re-read):0x00003000
[  742.112451] R8188EU: rtl8188e_sreset_xmit_status_check REG_TXDMA_STATUS (re-read):0x00003000
<6>R8188EU: rtl8188e_sreset_xmit_status_check REG_TXDMA_STATUS:0x00003000
[  918.090466] R8188EU: rtl8188e_sreset_xmit_status_check REG_TXDMA_STATUS:0x00003000
<6>R8188EU: rtl8188e_sreset_xmit_status_check REG_TXDMA_STATUS (re-read):0x00003000
[  918.105573] R8188EU: rtl8188e_sreset_xmit_status_check REG_TXDMA_STATUS (re-read):0x00003000
<6>R8188EU: rtl8188e_sreset_xmit_status_check REG_TXDMA_STATUS:0x00004200
[ 1048.089588] R8188EU: rtl8188e_sreset_xmit_status_check REG_TXDMA_STATUS:0x00004200
<6>R8188EU: rtl8188e_sreset_xmit_status_check REG_TXDMA_STATUS (re-read):0x00004200
[ 1048.109694] R8188EU: rtl8188e_sreset_xmit_status_check REG_TXDMA_STATUS (re-read):0x00004200
<6>R8188EU: OnAction_back
[ 1052.059867] R8188EU: OnAction_back
<6>R8188EU: OnAction_back, action=2
[ 1052.066492] R8188EU: OnAction_back, action=2
<6>R8188EU: OnAction_back(): DELBA: 0(27)
[ 1052.074495] R8188EU: OnAction_back(): DELBA: 0(27)
<6>R8188EU: rtl8188e_sreset_xmit_status_check tx hang
[ 1068.087616] R8188EU: rtl8188e_sreset_xmit_status_check tx hang
<6>R8188EU: issue_nulldata(wlan0) to fc:f8:ae:0a:9b:c3, ch:11, 1/1 in 0 ms
...

FYI, I deactivate the power management in the module as follow 

options 8188eu rtw_power_mgnt=0 rtw_enusbss=0

It is very strange because when the REG_TXDMA_STATUS = 0x4200 everything hang up. I don't really know what is the signification of this value.

Thank in advance for any suggestion

lwfinger, 2015-08-23:
At the moment, I have no idea what that value in TXDMA_STATUS means. I know no more about the internals of the chip than you do, but given the name of the register, it is obvious that something has gone wrong with DMA.

Which branch of the repo are you using? On any repo with multiple branches, please supply this info.

bsansoni, 2015-08-23:
I agree with you it is a TX DMA issue.
This is the branch I am currently using
benoit@nuc:~/Projects/Build/cubieboard/build/rtl8188eu$ git branch -v
* master 3091828 rtl8188eu: On kernel 2.6.32, a NULL dereference was found

I already had the case where TXDMA_STATUS=0x3. But the system was alive. So I think that my issue depends on the value.

lwfinger, 2015-08-25:
This driver has had considerable usage on x86 (both 32- and 64-bit) without this problem appearing, thus I suspect an alignment problem on the ARM cpu. At least, I cannot think of any other differences between the two architectures that would not be handled by the compiler.

Is your kernel reporting any alignment fixups that are needed?

bsansoni, 2015-08-25:
No, the kernel did not report any alignment fixups.
I checked it out in the dmesg.
It is the latest stable one 3.4.103 (managed on sunxi github repo).
I tried also with the 3.4.43. But same issue.
This is below the dump of /proc/cpu/alignment.
root@cubie:~# cat /proc/cpu/alignment 
User:           0
System:         0
Skipped:        0
Half:           0
Word:           0
DWord:          0
Multi:          0
User faults:    2 (fixup)

I tried also with an other usb wifi dongle based on rtl8192 and by activating the debug log, I found out the same issue.
Perhaps, you are right the problem can be around the alignment on this architecture. I am trying to continue to investigate.

bsansoni, 2015-08-26:
Sorry again,

I checked out one of your answer for the driver rtl8723au talking about RX gain modification done on RTL8188eu .
The issue https://github.com/lwfinger/rtl8723au/issues/4 is exactly what I have.
I saw this modification present at this link location https://github.com/lwfinger/rtl8723au/commit/c08a716d78bc9e45e9c147a65dbfb730f37e66b8 .

I think that for the RTL8188eu in non-AP mode the RX max/min gain is represented by these values 

#define		DM_DIG_MAX_NIC			0x4e  <-- This value was modified
#define		DM_DIG_MIN_NIC			0x1e /* 0x22/0x1c */

And or AP mode by these value 

#define		DM_DIG_MAX_AP			0x32 <-- can it be increased ???
#define		DM_DIG_MIN_AP			0x20

Don't you think that for AP the max value must be increased also ?
How to increase the TX gain ?