lwfinger/rtl8188eu #140: Fedora 22 freezes when plug on the tp-link wn725n

TJProgrammer, 2015-09-03:
I run the comand journalctl -f and the PC do something about network (3 of 5 steps) until stops/freezes.
I just install this driver today.

lwfinger, 2015-09-03:
Your statement that the OS just stops/freezes is not sufficient for me to tell anything about what is happening.

Are you running the driver from this repo? If so, which branch? What architecture?

In general, the driver in the kernel works other than for big-endian architectures, but it is unlikely that you have one of these.

TJProgrammer, 2015-09-03:
[root@localhost ~]# uname -a
Linux localhost.localdomain 4.1.6-200.fc22.i686 #1 SMP Mon Aug 17 20:27:54 UTC 2015 i686 i686 i386 GNU/Linux

Repo/branch: rtl8188eu-master.zip

TJProgrammer, 2015-09-03:
When I tested the driver, the terminal was on in jounalctl -f, but on reboot I can't go there (on the information that displayed before)...

lwfinger, 2015-09-03:
I do not have F22 running on any of my hardware, but I installed the 32-bit version in a VirtualBox virtual machine. The kernel driver is not enabled for the standard Fedora kernel, thus I cloned the git repo, added the necessary parts to build and install the driver, and rebooted. My local APs showed in the list of networks, and the network connected with no problems.

The v4.1.8_9499 branch also works. I am unable to duplicate your problem.

By the way, it is better to clone a git repo, rather than download a zipped version of it. Upgrading the repo is as simple as 'git pull'. There is never a need to grab the entire source again. In addition, in case a change has gone bad, you can always checkout an older version.

TJProgrammer, 2015-09-03:
Thanks, but how can I do the clone in terminal?

lwfinger, 2015-09-03:
git clone http://github.com/lwfinger/rtl8188eu.git

TJProgrammer, 2015-09-04:
Thanks. When you have news will can you tell me? Thanks.

lwfinger, 2015-09-04:
There will be no news until you provide the result of a kernel oops, or some other diagnostic output. The driver works here, and at the moment, I have nothing on which to base a fix.

TJProgrammer, 2015-09-04:
Some photos when the journalctl -f freeze:
[image]
[image]
[image]
[image]

lwfinger, 2015-09-04:
There is no crash here. You need to switch to the logging console (CTRL-ALT-F10 on my system) to get that information.

TJProgrammer, 2015-09-04:
It's true, there isn't no crash because the pc is freezed...
So, go on. After reboot the PC, If I do CTRL-ALT-F10, can I this information?
Anyway, only in Monday I can do this, but its strange that in your PC it's all ok, and in my don't.
When I can I will put more information here...

TJProgrammer, 2015-09-07:
Hello again. Today, I tested again with a clean SO Fedora 22. So the results:
```
[root@chatroom-informatica Transferências]# git clone http://github.com/lwfinger/rtl8188eu.git
Clonar em 'rtl8188eu'...
remote: Counting objects: 9894, done.
remote: Total 9894 (delta 0), reused 0 (delta 0), pack-reused 9894
Receiving objects: 100% (9894/9894), 12.03 MiB | 5.54 MiB/s, done.
Resolvendo deltas: 100% (7294/7294), done.
Checking connectivity... terminado.
[root@chatroom-informatica Transferências]# cd rtl8188eu/
[root@chatroom-informatica rtl8188eu]# make all
make ARCH=i386 CROSS_COMPILE= -C /lib/modules/4.1.6-200.fc22.i686+PAE/build M=/home/informatica/Transferências/rtl8188eu  modules
make[1]: *** /lib/modules/4.1.6-200.fc22.i686+PAE/build: No such file or directory.  Stop.
Makefile:147: recipe for target 'modules' failed
make: *** [modules] Error 2
[root@chatroom-informatica rtl8188eu]# dnf install kernel-headers kernel-devel
Last metadata expiration check performed 4:02:20 ago on Mon Sep  7 10:20:10 2015.
Package kernel-headers-4.1.6-200.fc22.i686 is already installed, skipping.
Package kernel-devel-4.1.6-200.fc22.i686 is already installed, skipping.
Dependencies resolved.
Nothing to do.
Complete!
[root@chatroom-informatica rtl8188eu]# uname -a
Linux chatroom-informatica 4.1.6-200.fc22.i686+PAE #1 SMP Mon Aug 17 20:12:14 UTC 2015 i686 i686 i386 GNU/Linux
[root@chatroom-informatica rtl8188eu]# dnf update
Last metadata expiration check performed 4:02:48 ago on Mon Sep  7 10:20:10 2015.
Dependencies resolved.
Nothing to do.
Complete!
```
PS: I've already installed kernel-headers, kernel-devel, gcc-c++, group 'Development Tools', ...

TJProgrammer, 2015-09-07:
I need a solution, because the tp link usb don't show the light green.
TP-Link Model: TL-WN725N

lwfinger, 2015-09-07:
Sorry, but I cannot help you. On every Fedora system where I have needed to compile out-of-kernel drivers, adding kernel-headers, kernel-devel, and gcc have been enough. Perhaps the PAE versions are different.

When you run the command "ls -l /lib/modules/`uname -r`/build", what do you get? Those funny symbols are back-tics, not the single quotation mark.

TJProgrammer, 2015-09-08:
I get this:
```
[root@chatroom-informatica ~]# ls -l /lib/modules/ uname -r /build
ls: impossível aceder a uname: No such file or directory
ls: impossível aceder a /build: No such file or directory
/lib/modules/:
total 12
drwxr-xr-x. 6 root root 4096 Set  7 10:09 4.1.6-200.fc22.i686+PAE
drwxr-xr-x. 5 root root 4096 Set  7 12:26 4.1.6-200.fc22.i686
drwxr-xr-x. 5 root root 4096 Set  4 22:40 4.0.4-301.fc22.i686
```

lwfinger, 2015-09-08:
Uck. GitHub mangled my command. Use
ls -l /lib/modules/$(uname -r)
ls -l /lib/modules/$(uname -r)/build

The first one is new. GitHub will not mess up that version.

TJProgrammer, 2015-09-08:
```
[informatica@chatroom-informatica ~]$ ls -l /lib/modules/$(uname -r)
total 3240
lrwxrwxrwx.  1 root root     40 Ago 17 21:26 build -> /usr/src/kernels/4.1.6-200.fc22.i686+PAE
drwxr-xr-x.  5 root root   4096 Set  7 10:05 extra
drwxr-xr-x. 12 root root   4096 Set  7 10:05 kernel
-rw-r--r--.  1 root root 849171 Set  7 10:09 modules.alias
-rw-r--r--.  1 root root 829867 Set  7 10:09 modules.alias.bin
-rw-r--r--.  1 root root   1773 Ago 17 21:26 modules.block
-rw-r--r--.  1 root root   7320 Ago 17 21:26 modules.builtin
-rw-r--r--.  1 root root   9779 Set  7 10:09 modules.builtin.bin
-rw-r--r--.  1 root root 291601 Set  7 10:09 modules.dep
-rw-r--r--.  1 root root 417304 Set  7 10:09 modules.dep.bin
-rw-r--r--.  1 root root    349 Set  7 10:09 modules.devname
-rw-r--r--.  1 root root    123 Ago 17 21:26 modules.drm
-rw-r--r--.  1 root root    100 Ago 17 21:26 modules.modesetting
-rw-r--r--.  1 root root   2498 Ago 17 21:26 modules.networking
-rw-r--r--.  1 root root 108866 Ago 17 21:26 modules.order
-rw-r--r--.  1 root root     84 Set  7 10:09 modules.softdep
-rw-r--r--.  1 root root 332173 Set  7 10:09 modules.symbols
-rw-r--r--.  1 root root 412627 Set  7 10:09 modules.symbols.bin
lrwxrwxrwx.  1 root root      5 Ago 17 21:26 source -> build
drwxr-xr-x.  2 root root   4096 Ago 17 21:25 updates
drwxr-xr-x.  2 root root   4096 Set  7 10:05 vdso
[informatica@chatroom-informatica ~]$ 
```

TJProgrammer, 2015-09-08:
```
[informatica@chatroom-informatica ~]$ ls -l /lib/modules/$(uname -r)/build
lrwxrwxrwx. 1 root root 40 Ago 17 21:26 /lib/modules/4.1.6-200.fc22.i686+PAE/build -> /usr/src/kernels/4.1.6-200.fc22.i686+PAE
[informatica@chatroom-informatica ~]$ 
```

lwfinger, 2015-09-08:
I also need 2 more
ls /lib/modules/$(uname -r)/build
ls -l /lib/modules

TJProgrammer, 2015-09-08:
```
[informatica@chatroom-informatica ~]$ ls /lib/modules/$(uname -r)/build
/lib/modules/4.1.6-200.fc22.i686+PAE/build
[informatica@chatroom-informatica ~]$ ls -l /lib/modules
total 12
drwxr-xr-x. 5 root root 4096 Set  4 22:40 4.0.4-301.fc22.i686
drwxr-xr-x. 5 root root 4096 Set  7 12:26 4.1.6-200.fc22.i686
drwxr-xr-x. 6 root root 4096 Set  7 10:09 4.1.6-200.fc22.i686+PAE
```

lwfinger, 2015-09-09:
Your result is the clue.
[informatica@chatroom-informatica ~]$ ls /lib/modules/$(uname -r)/build
/lib/modules/4.1.6-200.fc22.i686+PAE/build

That indicates that the kernel headers that you installed do not match the kernel that you are running. I think I recall that Fedora gives you the latest package, not the one that matches your kernel. You probably got headers for 4.1.6-200.fc22.i686, not 4.1.6-200.fc22.i686+PAE. If there is a way for dnf to force the latter, use it.

It may work if you "cp /lib/modules/4.1.6-200.fc22.i686/build 4.1.6-200.fc22.i686+PAE/build". That will need to be done as root.

TJProgrammer, 2015-09-09:
I understand you, but does the PC? Here are the results:
```
[root@chatroom-informatica ~]# cp /lib/modules/4.1.6-200.fc22.i686/build 4.1.6-200.fc22.i686+PAE/build
cp: a omitir a pasta «/lib/modules/4.1.6-200.fc22.i686/build»
```
Now I will try to install the driver...
```
[root@chatroom-informatica rtl8188eu]# make all
make ARCH=i386 CROSS_COMPILE= -C /lib/modules/4.1.6-200.fc22.i686+PAE/build M=/home/informatica/Transferências/rtl8188eu  modules
make[1]: *** /lib/modules/4.1.6-200.fc22.i686+PAE/build: No such file or directory.  Stop.
Makefile:147: recipe for target 'modules' failed
make: *** [modules] Error 2
```

lwfinger, 2015-09-09:
I was hurrying too much. That should have been
cp /lib/modules/4.1.6-200.fc22.i686/build /lib/modules/4.1.6-200.fc22.i686+PAE/build

If this does not work, you need to go to the Fedora help forum to find out how to install kernel-headers for the PAE kernel. By the way, how much RAM is in your computer?

TJProgrammer, 2015-09-09:
I've 3,9GB of RAM.
Results:
```
[root@chatroom-informatica rtl8188eu]# cp /lib/modules/4.1.6-200.fc22.i686/build /lib/modules/4.1.6-200.fc22.i686+PAE/build
cp: a omitir a pasta «/lib/modules/4.1.6-200.fc22.i686/build»
[root@chatroom-informatica rtl8188eu]# make all
make ARCH=i386 CROSS_COMPILE= -C /lib/modules/4.1.6-200.fc22.i686+PAE/build M=/home/informatica/Transferências/rtl8188eu  modules
make[1]: *** /lib/modules/4.1.6-200.fc22.i686+PAE/build: No such file or directory.  Stop.
Makefile:147: recipe for target 'modules' failed
make: *** [modules] Error 2
```
That seems doesn't work. So ... if you haven't any idea, can you say me one fedora forum?

lwfinger, 2015-09-09:
With that amount of RAM, the PAE kernel does nothing for you. It is only of value when you have a 32-bit system, and more than 4 GB RAM. I suggest you switch to the 4.1.6-200.fc22.i686 kernel.

TJProgrammer, 2015-09-10:
I reboot the Pc and change the kernel, in boot, to 4.1.6-200.fc22.i686
After that I rum this:
```
[root@chatroom-informatica rtl8188eu]# make all
make ARCH=i386 CROSS_COMPILE= -C /lib/modules/4.1.6-200.fc22.i686/build M=/home/informatica/Transferências/rtl8188eu  modules
make[1]: Entering directory '/usr/src/kernels/4.1.6-200.fc22.i686'
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_ap.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_br_ext.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_cmd.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_debug.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_efuse.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_ieee80211.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_io.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_ioctl_set.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_iol.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_led.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_mlme.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_mlme_ext.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_mp.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_mp_ioctl.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_pwrctrl.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_p2p.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_recv.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_rf.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_security.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_sreset.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_sta_mgt.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_wlan_util.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/core/rtw_xmit.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/HalHWImg8188E_MAC.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/HalHWImg8188E_BB.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/HalHWImg8188E_RF.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/HalPhyRf.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/HalPhyRf_8188e.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/HalPwrSeqCmd.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/Hal8188EPwrSeq.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/Hal8188ERateAdaptive.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/hal_intf.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/hal_com.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/odm.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/odm_debug.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/odm_interface.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/odm_HWConfig.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/odm_RegConfig8188E.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/odm_RTL8188E.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/rtl8188e_cmd.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/rtl8188e_dm.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/rtl8188e_hal_init.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/rtl8188e_mp.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/rtl8188e_phycfg.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/rtl8188e_rf6052.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/rtl8188e_rxdesc.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/rtl8188e_sreset.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/rtl8188e_xmit.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/rtl8188eu_led.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/rtl8188eu_recv.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/rtl8188eu_xmit.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/usb_halinit.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/hal/usb_ops_linux.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/os_dep/ioctl_linux.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/os_dep/mlme_linux.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/os_dep/os_intfs.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/os_dep/osdep_service.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/os_dep/recv_linux.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/os_dep/rtw_android.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/os_dep/usb_intf.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/os_dep/usb_ops_linux.o
  CC [M]  /home/informatica/Transferências/rtl8188eu/os_dep/xmit_linux.o
  LD [M]  /home/informatica/Transferências/rtl8188eu/8188eu.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/informatica/Transferências/rtl8188eu/8188eu.mod.o
  LD [M]  /home/informatica/Transferências/rtl8188eu/8188eu.ko
make[1]: Leaving directory '/usr/src/kernels/4.1.6-200.fc22.i686'
[root@chatroom-informatica rtl8188eu]# make install
install -p -m 644 8188eu.ko  /lib/modules/4.1.6-200.fc22.i686/kernel/drivers/net/wireless
cp rtl8188eufw.bin /lib/firmware/.
/sbin/depmod -a 4.1.6-200.fc22.i686
mkdir -p /lib/firmware/rtlwifi
cp -n rtl8188eufw.bin /lib/firmware/rtlwifi/.
```
Now, reboot the PC or isn't necessary?
And the other kernel version (with PAE), unistall it?

lwfinger, 2015-09-10:
A reboot is not necessary. Just 'sudo  modprobe -v 8188eu'.

You can uninstall the other kernel if you want, or you can simply not boot it.

TJProgrammer, 2015-09-11:
And it worked, after a first test unsuccessful. A question: always when I turn on the PC is necessary do the 'make all' and 'make install' commands?

lwfinger, 2015-09-11:
No. As long as you have a wifi device, it will be fine. The only time you need to do remake the driver is when the kernel is updated.

closed 2015-09-11 by lwfinger

TJProgrammer, 2015-09-11:
Thanks a lot!
by the way, when is it that this driver is built in fedora?