lwfinger/rtl8188eu #142: firmware not working

tisdall, 2015-09-11:
I'm getting `Request firmware failed with error 0xfffffffe` but I'm not sure why.

I think that's the error you get when you forget to put the firmware .bin in the right place, but I _did_ copy it over.
```
root@linaro-developer:~# ls -l /lib/firmware/rtlwifi/rtl8188eufw.bin 
-rw-r--r-- 1 root root 13904 Sep 11  2015 /lib/firmware/rtlwifi/rtl8188eufw.bin
root@linaro-developer:~# /sbin/depmod 
root@linaro-developer:~# modprobe -r 8188eu
<6>usbcore: deregistering interface driver rtl8188eu
<6>usb 2-1: reset high-speed USB device number 2 using sw-ehci
root@linaro-developer:~# modprobe 8188eu
<6>cfg80211: Calling CRDA to update world regulatory domain
bFWReady == false call reset 8051...
<6>usbcore: registered new interface driver rtl8188eu
root@linaro-developer:~# ifconfig
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:16436  Metric:1
          RX packets:4168 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4168 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:337552 (337.5 KB)  TX bytes:337552 (337.5 KB)

root@linaro-developer:~# ifconfig wlan0 up
<3>Request firmware failed with error 0xfffffffe
[  796.025107] Request firmware failed with error 0xfffffffe
SIOCSIFFLAGS: Operation not permitted
root@linaro-developer:~# modinfo 8188eu
filename:       /lib/modules/3.4.104-00136-gd47d367-dirty/kernel/drivers/net/wireless/rtl8188eu/8188eu.ko
version:        v4.1.8_9499.20131104
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     90632A8D1E91F916242A3B5
alias:          usb:v07B8p8179d*dc*dsc*dp*ic*isc*ip*
alias:          usb:v0BDAp0179d*dc*dsc*dp*ic*isc*ip*
alias:          usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*
depends:        cfg80211
vermagic:       3.4.104-00136-gd47d367-dirty preempt mod_unload modversions ARMv7 p2v8 
parm:           rtw_ips_mode:The default IPS mode (int)
parm:           rtw_regulatory_id:int
parm:           rtw_qos_opt_enable:int
parm:           ifname:The default name to allocate for first interface (charp)
parm:           if2name:The default name to allocate for second interface (charp)
parm:           rtw_initmac:charp
parm:           rtw_channel_plan:int
parm:           rtw_chip_version:int
parm:           rtw_rfintfs:int
parm:           rtw_lbkmode:int
parm:           rtw_network_mode:int
parm:           rtw_channel:int
parm:           rtw_mp_mode:int
parm:           rtw_wmm_enable:int
parm:           rtw_vrtl_carrier_sense:int
parm:           rtw_vcs_type:int
parm:           rtw_busy_thresh:int
parm:           rtw_ht_enable:int
parm:           rtw_cbw40_enable:int
parm:           rtw_ampdu_enable:int
parm:           rtw_rx_stbc:int
parm:           rtw_ampdu_amsdu:int
parm:           rtw_lowrate_two_xmit:int
parm:           rtw_rf_config:int
parm:           rtw_power_mgnt:int
parm:           rtw_smart_ps:int
parm:           rtw_low_power:int
parm:           rtw_wifi_spec:int
parm:           rtw_antdiv_cfg:int
parm:           rtw_antdiv_type:int
parm:           rtw_enusbss:int
parm:           rtw_hwpdn_mode:int
parm:           rtw_hwpwrp_detect:int
parm:           rtw_hw_wps_pbc:int
parm:           rtw_max_roaming_times:The max roaming times to try (uint)
parm:           rtw_fw_iol:FW IOL (int)
parm:           rtw_mc2u_disable:int
parm:           rtw_80211d:Enable 802.11d mechanism (int)
parm:           rtw_notch_filter:0:Disable, 1:Enable, 2:Enable only for P2P (uint)
parm:           debug:Set debug level (1-9) (default 1) (int)
```

Is there something obvious I'm missing?

Alex66955, 2017-08-15:
I got the same error.
Any updates?

lwfinger, 2017-08-15:
When you reply to a two-year old message, you can hardly appear to have the same problem.

There are two reasons for missing firmware: (1) The listed file does not exist in /lib/firmware. (2) The driver is included in your initrd, but the firmware is not. The test for #2 is to unload/reload the driver.

Alex66955, 2017-08-15:
Thank for fast reply.
Like in your documentation mentioned the firmware was missing.