lwfinger/rtl8188eu #143: Fedora 22 Update

TJProgrammer, 2015-09-15:
Hello!
I had a problem with that driver that has been resolved.

Despite having problems, as freeze Pc by anything and not pass the screen to the Fedora's logo at the restart, things were going well.

But there was an update the kernel. After the updates, I made this comands as follows:
```
[root@chatroom-informatica ~]# uname --all
Linux chatroom-informatica 4.1.6-201.fc22.i686 #1 SMP Fri Sep 4 18:22:45 UTC 2015 i686 i686 i386 GNU/Linux
[root@chatroom-informatica ~]# modprobe -v 8188eu
insmod /lib/modules/4.1.6-201.fc22.i686/kernel/drivers/net/wireless/8188eu.ko 
modprobe: ERROR: could not insert '8188eu': Exec format error
[root@chatroom-informatica ~]# 
[root@chatroom-informatica ~]# cd /home/informatica/Transferências/rtl8188eu/
[root@chatroom-informatica rtl8188eu]# git clone -b v4.1.8_9499 https://github.com/lwfinger/rtl8188eu.git
Clonar em 'rtl8188eu'...
remote: Counting objects: 9910, done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 9910 (delta 0), reused 0 (delta 0), pack-reused 9905
Receiving objects: 100% (9910/9910), 12.06 MiB | 486.00 KiB/s, done.
Resolvendo deltas: 100% (7304/7304), done.
Checking connectivity... terminado.
[root@chatroom-informatica rtl8188eu]# make al
make: *** No rule to make target 'al'.  Stop.
[root@chatroom-informatica rtl8188eu]# make all
make ARCH=i386 CROSS_COMPILE= -C /lib/modules/4.1.6-201.fc22.i686/build M=/home/informatica/Transferências/rtl8188eu  modules
make[1]: Entering directory '/usr/src/kernels/4.1.6-201.fc22.i686'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/kernels/4.1.6-201.fc22.i686'
[root@chatroom-informatica rtl8188eu]# make install 
install -p -m 644 8188eu.ko  /lib/modules/4.1.6-201.fc22.i686/kernel/drivers/net/wireless
cp rtl8188eufw.bin /lib/firmware/.
/sbin/depmod -a 4.1.6-201.fc22.i686
mkdir -p /lib/firmware/rtlwifi
cp -n rtl8188eufw.bin /lib/firmware/rtlwifi/.
[root@chatroom-informatica rtl8188eu]# modprobe -v 8188eu
insmod /lib/modules/4.1.6-201.fc22.i686/kernel/drivers/net/wireless/8188eu.ko 
modprobe: ERROR: could not insert '8188eu': Exec format error
[root@chatroom-informatica rtl8188eu]# 
```
I do not know what I can do.
Does the solution is to continue to use another kernel?

lwfinger, 2015-09-16:
What kernel are you actually running? 'uname -r' will tell you that. I bet that it is not 4.1.6-201.fc22.i686.

TJProgrammer, 2015-09-16:
```
4.1.6-201.fc22.i686
```

lwfinger, 2015-09-16:
Of course you would get errors for the driver from the TP-Link site. If you want to use that, you would need a much older kernel.

I have fixed the problem. There must not be many people running with this branch.

TJProgrammer, 2015-09-17:
I will do git pull, but I do from this branch or from master?

lwfinger, 2015-09-17:
From whatever branch you are using!!!

TJProgrammer, 2015-09-18:
Thank you! The wireless internet was working!

closed 2015-11-20 by TJProgrammer