lwfinger/rtl8188eu #145: Missing aliases in compiled driver

marthellius, 2015-09-24:
I'm trying to compile the rtl8188eu module on a Raspberry Pi B running kernel 3.12.9 (from the Raspbian repository).

The compilation process seems to work as expected:

```
make all
make ARCH=arm CROSS_COMPILE= -C /lib/modules/3.12-1-rpi/build M=/tmp/rtl8188eu-master  modules
make[1]: Entering directory '/usr/src/linux-headers-3.12-1-rpi'
Makefile:10: *** mixed implicit and normal rules: deprecated syntax
  CC [M]  /tmp/rtl8188eu-master/core/rtw_ap.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_br_ext.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_cmd.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_debug.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_efuse.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_ieee80211.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_io.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_ioctl_set.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_iol.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_led.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_mlme.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_mlme_ext.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_mp.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_mp_ioctl.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_pwrctrl.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_p2p.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_recv.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_rf.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_security.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_sreset.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_sta_mgt.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_wlan_util.o
  CC [M]  /tmp/rtl8188eu-master/core/rtw_xmit.o
  CC [M]  /tmp/rtl8188eu-master/hal/HalHWImg8188E_MAC.o
  CC [M]  /tmp/rtl8188eu-master/hal/HalHWImg8188E_BB.o
  CC [M]  /tmp/rtl8188eu-master/hal/HalHWImg8188E_RF.o
  CC [M]  /tmp/rtl8188eu-master/hal/HalPhyRf.o
  CC [M]  /tmp/rtl8188eu-master/hal/HalPhyRf_8188e.o
  CC [M]  /tmp/rtl8188eu-master/hal/HalPwrSeqCmd.o
  CC [M]  /tmp/rtl8188eu-master/hal/Hal8188EPwrSeq.o
  CC [M]  /tmp/rtl8188eu-master/hal/Hal8188ERateAdaptive.o
  CC [M]  /tmp/rtl8188eu-master/hal/hal_intf.o
  CC [M]  /tmp/rtl8188eu-master/hal/hal_com.o
  CC [M]  /tmp/rtl8188eu-master/hal/odm.o
  CC [M]  /tmp/rtl8188eu-master/hal/odm_debug.o
  CC [M]  /tmp/rtl8188eu-master/hal/odm_interface.o
  CC [M]  /tmp/rtl8188eu-master/hal/odm_HWConfig.o
  CC [M]  /tmp/rtl8188eu-master/hal/odm_RegConfig8188E.o
  CC [M]  /tmp/rtl8188eu-master/hal/odm_RTL8188E.o
  CC [M]  /tmp/rtl8188eu-master/hal/rtl8188e_cmd.o
  CC [M]  /tmp/rtl8188eu-master/hal/rtl8188e_dm.o
  CC [M]  /tmp/rtl8188eu-master/hal/rtl8188e_hal_init.o
  CC [M]  /tmp/rtl8188eu-master/hal/rtl8188e_mp.o
  CC [M]  /tmp/rtl8188eu-master/hal/rtl8188e_phycfg.o
  CC [M]  /tmp/rtl8188eu-master/hal/rtl8188e_rf6052.o
  CC [M]  /tmp/rtl8188eu-master/hal/rtl8188e_rxdesc.o
  CC [M]  /tmp/rtl8188eu-master/hal/rtl8188e_sreset.o
  CC [M]  /tmp/rtl8188eu-master/hal/rtl8188e_xmit.o
  CC [M]  /tmp/rtl8188eu-master/hal/rtl8188eu_led.o
  CC [M]  /tmp/rtl8188eu-master/hal/rtl8188eu_recv.o
  CC [M]  /tmp/rtl8188eu-master/hal/rtl8188eu_xmit.o
  CC [M]  /tmp/rtl8188eu-master/hal/usb_halinit.o
  CC [M]  /tmp/rtl8188eu-master/hal/usb_ops_linux.o
  CC [M]  /tmp/rtl8188eu-master/os_dep/ioctl_linux.o
  CC [M]  /tmp/rtl8188eu-master/os_dep/mlme_linux.o
  CC [M]  /tmp/rtl8188eu-master/os_dep/os_intfs.o
  CC [M]  /tmp/rtl8188eu-master/os_dep/osdep_service.o
  CC [M]  /tmp/rtl8188eu-master/os_dep/recv_linux.o
  CC [M]  /tmp/rtl8188eu-master/os_dep/rtw_android.o
  CC [M]  /tmp/rtl8188eu-master/os_dep/usb_intf.o
  CC [M]  /tmp/rtl8188eu-master/os_dep/usb_ops_linux.o
  CC [M]  /tmp/rtl8188eu-master/os_dep/xmit_linux.o
  LD [M]  /tmp/rtl8188eu-master/8188eu.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /tmp/rtl8188eu-master/8188eu.mod.o
  LD [M]  /tmp/rtl8188eu-master/8188eu.ko
make[1]: Leaving directory '/usr/src/linux-headers-3.12-1-rpi'
```
However, the module is missing all the aliases:

```
modinfo ./8188eu.ko 
filename:       /tmp/rtl8188eu-master/./8188eu.ko
version:        v4.1.4_6773.20130222
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     BB97F7E3F4F970628EC7EA3
depends:        
vermagic:       3.12-1-rpi mod_unload modversions ARMv6 p2v8 
parm:           rtw_ips_mode:The default IPS mode (int)
parm:           ifname:The default name to allocate for first interface (charp)
parm:           if2name:The default name to allocate for second interface (charp)
parm:           rtw_initmac:charp
parm:           rtw_channel_plan:int
parm:           rtw_chip_version:int
parm:           rtw_rfintfs:int
parm:           rtw_lbkmode:int
parm:           rtw_network_mode:int
parm:           rtw_channel:int
parm:           rtw_mp_mode:int
parm:           rtw_wmm_enable:int
parm:           rtw_vrtl_carrier_sense:int
parm:           rtw_vcs_type:int
parm:           rtw_busy_thresh:int
parm:           rtw_ht_enable:int
parm:           rtw_cbw40_enable:int
parm:           rtw_ampdu_enable:int
parm:           rtw_rx_stbc:int
parm:           rtw_ampdu_amsdu:int
parm:           rtw_lowrate_two_xmit:int
parm:           rtw_rf_config:int
parm:           rtw_power_mgnt:int
parm:           rtw_smart_ps:int
parm:           rtw_low_power:int
parm:           rtw_wifi_spec:int
parm:           rtw_antdiv_cfg:int
parm:           rtw_antdiv_type:int
parm:           rtw_enusbss:int
parm:           rtw_hwpdn_mode:int
parm:           rtw_hwpwrp_detect:int
parm:           rtw_hw_wps_pbc:int
parm:           rtw_max_roaming_times:The max roaming times to try (uint)
parm:           rtw_fw_iol:FW IOL (int)
parm:           rtw_mc2u_disable:int
parm:           rtw_80211d:Enable 802.11d mechanism (int)
parm:           rtw_notch_filter:0:Disable, 1:Enable, 2:Enable only for P2P (uint)
parm:           debug:Set debug level (1-9) (default 1) (int)
```

If I compile the driver on Debian 7 with kernel 3.2 (i686), the resulting ko file has all the expected aliases. 

Am I doing something wrong when compiling the driver on the Pi?

lwfinger, 2015-09-25:
Those aliases are created by the USB_DEVICE macro. Perhaps there is something wrong with the C compiler on the ARM. There is not anything that you could do that would cause this problem.

marthellius, 2015-10-08:
However, if I compile the whole 3.18 kernel (which includes the r8188eu module), all the aliases are there:

```
filename:       /tmp/r8188eu.ko
version:        v4.1.4_6773.20130222
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     78730ECEEDF3DEA0B52452E
alias:          usb:v0DF6p0076d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3311d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3310d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p330Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p8179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep4008d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*in*
depends:        
staging:        Y
intree:         Y
vermagic:       3.18.16-rpi preempt mod_unload modversions ARMv6 
parm:           rtw_ips_mode:The default IPS mode (int)
parm:           ifname:The default name to allocate for first interface (charp)
parm:           if2name:The default name to allocate for second interface (charp)
parm:           rtw_initmac:charp
parm:           rtw_channel_plan:int
parm:           rtw_chip_version:int
parm:           rtw_rfintfs:int
parm:           rtw_lbkmode:int
parm:           rtw_network_mode:int
parm:           rtw_channel:int
parm:           rtw_wmm_enable:int
parm:           rtw_vrtl_carrier_sense:int
parm:           rtw_vcs_type:int
parm:           rtw_busy_thresh:int
parm:           rtw_ht_enable:int
parm:           rtw_cbw40_enable:int
parm:           rtw_ampdu_enable:int
parm:           rtw_rx_stbc:int
parm:           rtw_ampdu_amsdu:int
parm:           rtw_lowrate_two_xmit:int
parm:           rtw_rf_config:int
parm:           rtw_power_mgnt:int
parm:           rtw_smart_ps:int
parm:           rtw_low_power:int
parm:           rtw_wifi_spec:int
parm:           rtw_antdiv_cfg:int
parm:           rtw_antdiv_type:int
parm:           rtw_enusbss:int
parm:           rtw_hwpdn_mode:int
parm:           rtw_hwpwrp_detect:int
parm:           rtw_hw_wps_pbc:int
parm:           rtw_max_roaming_times:The max roaming times to try (uint)
parm:           rtw_fw_iol:FW IOL (int)
parm:           rtw_mc2u_disable:int
parm:           rtw_80211d:Enable 802.11d mechanism (int)
parm:           rtw_notch_filter:0:Disable, 1:Enable, 2:Enable only for P2P (uint)
parm:           debug:Set debug level (1-9) (default 1) (int)
```

The compiler I'm using is gcc 4.7.2