lwfinger/rtl8188eu #148: lag spikes

choikwa, 2015-10-11:
I am getting lag spikes every 20-30 pings.

64 bytes from yyz08s13-in-f20.1e100.net (74.125.226.116): icmp_seq=1894 ttl=58 time=217 ms
64 bytes from yyz08s13-in-f20.1e100.net (74.125.226.116): icmp_seq=1895 ttl=58 time=12.0 ms
64 bytes from yyz08s13-in-f20.1e100.net (74.125.226.116): icmp_seq=1896 ttl=58 time=12.4 ms
64 bytes from yyz08s13-in-f20.1e100.net (74.125.226.116): icmp_seq=1897 ttl=58 time=12.1 ms
64 bytes from yyz08s13-in-f20.1e100.net (74.125.226.116): icmp_seq=1898 ttl=58 time=1015 ms
64 bytes from yyz08s13-in-f20.1e100.net (74.125.226.116): icmp_seq=1899 ttl=58 time=20.1 ms
64 bytes from yyz08s13-in-f20.1e100.net (74.125.226.116): icmp_seq=1900 ttl=58 time=11.9 ms
64 bytes from yyz08s13-in-f20.1e100.net (74.125.226.116): icmp_seq=1901 ttl=58 time=12.5 ms
64 bytes from yyz08s13-in-f20.1e100.net (74.125.226.116): icmp_seq=1902 ttl=58 time=12.3 ms

...
64 bytes from yyz08s13-in-f20.1e100.net (74.125.226.116): icmp_seq=2077 ttl=58 time=11.8 ms
64 bytes from yyz08s13-in-f20.1e100.net (74.125.226.116): icmp_seq=2078 ttl=58 time=12.8 ms
64 bytes from yyz08s13-in-f20.1e100.net (74.125.226.116): icmp_seq=2079 ttl=58 time=978 ms
64 bytes from yyz08s13-in-f20.1e100.net (74.125.226.116): icmp_seq=2080 ttl=58 time=12.3 ms
64 bytes from yyz08s13-in-f20.1e100.net (74.125.226.116): icmp_seq=2081 ttl=58 time=11.9 ms

This is making it difficult in gaming. Does anyone know how to start debugging this? I'd like to confirm if this is the limitation of the wireless adapter I'm using TL-WN725n.
cat /etc/*release
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=17
DISTRIB_CODENAME=qiana
DISTRIB_DESCRIPTION="Linux Mint 17 Qiana"
NAME="Ubuntu"
VERSION="14.04.3 LTS, Trusty Tahr"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 14.04.3 LTS"
VERSION_ID="14.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"

choikwa, 2015-10-11:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=62371&sid=afeb1d232e836be4128ceec4fef182ed&start=25

this thread seems to be relevant in describing similar issue.

choikwa, 2015-10-11:
Hmm... this adapter doesn't seem to be in the kernel..

https://wikidevi.com/wiki/TP-LINK_TL-WN725N_v2

lwfinger, 2015-10-11:
If you want it in the kernel, you absolutely must supply the USB IDs for your device.

choikwa, 2015-10-23:
found somewhere that setting BSSID disables this periodic scanning which seems to be the source of the spikes.

closed 2015-10-23 by choikwa

netllama, 2015-10-23:
@choikwa how did you disable BSSID?

choikwa, 2015-10-30:
Go left-click nm-applet (wifi thingy) on Mint panel, network connections, edit your wifi, click dropdown on BSSID, and it already had it for me so I just clicked it and lag disappeared