lwfinger/rtl8188eu #150: How to cross-compiled?

luisfpinto, 2015-10-16:
Hi guys!

Could you please me, how can i cross compile this code? What environment variables do I need to define? Or what have I to do?  

Thank you!

lwfinger, 2015-10-16:
It is like any other out-of-kernel driver. You need *exactly* the same compiler used to build the kernel.

If the target architecture is different than the host, you will need the appropriate cross compiler.

Your environment variables will need to define ARCH and CROSS_COMPILE, or you can define them in the make line below:

make ARCH=XXXXX CROSS_COMPILE=YYYYY -C ZZZZZ M=WWWWW  modules

ZZZZZ is the patch of the headers for the target kernel. WWWWW is the path to the source of the driver source. If you are in that directory when you issue the make, you can ignore that part.

I have not searched, but there should be lots of examples on how to get the appropriate cross compiler, etc.

luisfpinto, 2015-10-16:
And what YYYY does it means? 

Thank you!

lwfinger, 2015-10-16:
Web page http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.faqs/14409.html shows how to set it up for one flavor of ARM. If that does not meet your requirement, google for CROSS_COMPILE.

TheBlackRabbit, 2015-11-10:
Personally, I might suggest (with no offense intended of course) that perhaps you start with educating yourself on the more basic concepts of Unix/Linux. In this case package managers either common or specific to various distributions such as apt-get, aptitude, yum (Debian) & pacman (ArchLinux) specifically concerning their critical role regarding dependencies; especially those required for development & compiling GCC/G++/TCC etc... If you decide to head my advice and take a dive into java and python as well I highly doubt you would regret it! The "newinstaller" nethuner build branch utilises python build.py significantly decreasing time,  effort, bandwidth, computing resources and dependency issues typical work the older, more traditional (?) GNU Make approach.

In an effort to make my advice more concise, I would suffice to say that until you make the effort to get a GOOD grasp of you're Unix/Linux basics; your going to end up spending more time asking questions rather than getting anything done...

Personally I study "The Linux Bible" (eighth edition atm) on practically a daily basis.I suggest anyone do the same.

Because once you know Linux and how to utilize it you not only realize there are a million different ways to do any thing but that this also applies to getting answers. You won't need to ask many, if any, questions. You would realize that perhaps you have missing build dependencies for GCC/G++/TCC/etc..? Missing $PATH appendages? Perhaps you simply don't meet the hardware/resource or even the skill level for the project your attempting.

README(.md) -/info / man / ESC+? / --help / -v(vv) / apt search 'x' / apt-cache search 'x' / dmesg / ps -aux (and so many more!) are our friends!

Arun22, 2016-01-27:
if you still have issue in cross compile: here you go: I build this driver for Raspberry Pi from Ubuntu Host 14.04_LTS.
make ARCH=arm CROSS_COMPILE=$RASP_CROSS_COMPILE KSRC=/home/pi/rasp/source/linux-raspbian-3.18.y

closed 2016-04-22 by luisfpinto