lwfinger/rtl8188eu #152: New driver for RTL8188EU

jeffdelk, 2015-10-29:
Tp Link has a new linux  driver that work great for me after slight modification (http://www.tp-link.us/res/down/soft/TL-WN725N_V2_150911.zip)  

The Tp-link files would not compile on my 4.0.0+  kernel so I modified the files. Here is a link to the modified files:
 https://www.dropbox.com/s/znp5munr03oow0b/rtl8188eu-linux-driver.zip?dl=0  
 I have compiled this against 4.1.9,  4.1.8, and 3.14.7 x86 and x64 kernels

This driver solves 4 problems for me: 1) random hanging of the interface, 2) low reported signal strength, 3) Nop phy(x) created (iw list returns nothing),  and 4) Excessive cpu load.

I had tried the staging diver, your master branch and  Daber's master branch. these preformed poorly for me.

I hope this helps,

CJD

lwfinger, 2015-10-30:
It would be helpful if you found the differences between this "wonderful" driver and the one in the kernel, and fixed the latter. Me, I have no more interest in working with the crappy code that the USB group at Realtek foists on the public!!!

trust199, 2015-11-23:
on 4.2 didn't work

callegar, 2015-12-11:
Same here. Cannot compile on Ubuntu Wily.

callegar, 2015-12-11:
But it compiles with a couple of trivial fixes. And the big difference between this and the original driver is that this one works while the one shipped with ubuntu is totally broken for me on linux 4.2 (continuous disconnections).

lwfinger, 2015-12-11:
OK, why not send me those trivial fixes. As long as they do not rely on some of the questionable practices of Ubuntu, then I will incorporate them.

callegar, 2015-12-11:
No, they are not ubuntu specific. They are only due to changes in some interfaces in 4.2.

First fix:

diff -r rtl8188eu-linux-driver/os_dep/linux/ioctl_cfg80211.c rtl8188eu-linux-driver2/os_dep/linux/ioctl_cfg80211.c
907c907
<                       cfg80211_disconnected(padapter->pnetdev, 0, NULL, 0, GFP_ATOMIC);
---
>                       cfg80211_disconnected(padapter->pnetdev, 0, NULL, 0, false, GFP_ATOMIC);

This is pretty obvious. However, to make it general, you should include it into an #ifdef so that you call cfg80211_disconnected with 5 params on kernels up to 4.1 and with 6 params from 4.2 on.

Second fix

diff -r rtl8188eu-linux-driver/hal/hal_com_phycfg.c rtl8188eu-linux-driver2/hal/hal_com_phycfg.c
2578c2578
< char  file_path[PATH_LENGTH_MAX];
---
> //char        file_path[PATH_LENGTH_MAX];

Not totally sure about this one. Don't know if the kernel has gained a definition that was missing or if file_path is now used for something totally unrelated so it needs to be renamed here.

Hope this helps.

Any chance that your tree is pushed to Greg for inclusion in his kernel staging tree?

lwfinger, 2015-12-11:
I have no idea what source you are using. The "file_path" problem was fixed in the v4.1.8_9499 branch nearly a year ago, and the cfg80211_disconnected() change is not needed there.

No, this tree will not be pushed to Greg. Getting this stuff into a form that the kernel would be acceptable would be about 6-8 months of work, and I have no interest in that kind of commitment.

callegar, 2015-12-12:
I am referring to the jeffdelk tree (see above), that seems to be the only one working on my system without constantly disconnecting.

lwfinger, 2015-12-12:
Then you should mention that at the beginning before you waste my time!!!!

closed 2015-12-12 by lwfinger

callegar, 2015-12-12:
Sorry about that, but it was in an issue about this new driver from TPLINK "New driver for RTL8188EU" so I thought it was sort of implicit. In fact, initially I got surprised by your request for the "trivial compilation fixes" (originally I did not even post them, because I thought they were not relevant for you being against the TPLINK Code)...