lwfinger/rtl8188eu #155: Ubuntu 15.10 - Hotspot with hostapd 0.8

abarcellona, 2015-12-30:
I have compiled and installed driver and firmware from this repository.
I have compiled hostapd from this repository.
I have connected a TP-Link tl-wn725n V2 USB wifi dongle. I works in client mode (I can connect to an AP).
But I need to setup an AP, so I started hostapd apparently without any error (see output below), but I can't see the SSID from an android smartphone...
    
andreabarcellona@hp:~/rtl8188eu$ sudo ./control_ap start wlx60e3271f0534 eno1

Starting AP mode for wlx60e3271f0534 at address 192.168.0.1
Internet Systems Consortium DHCP Server 4.3.1
Copyright 2004-2014 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/
Config file: /home/andreabarcellona/dhcpd.conf
Database file: /var/lib/dhcp/dhcpd.leases
PID file: /var/run/dhcpd.pid
Wrote 0 leases to leases file.
Listening on LPF/wlx60e3271f0534/60:e3:27:1f:05:34/192.168.0.0/24
Sending on   LPF/wlx60e3271f0534/60:e3:27:1f:05:34/192.168.0.0/24
Sending on   Socket/fallback/fallback-net
random: Trying to read entropy from /dev/random
Configuration file: /home/andreabarcellona/hostapd.conf
ctrl_interface_group=0
drv->ifindex=8
l2_sock_recv==l2_sock_xmit=0x0xe0cc20
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
Allowed channel: mode=1 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=7 freq=2442 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=8 freq=2447 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=12 freq=2467 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=13 freq=2472 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=7 freq=2442 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=8 freq=2447 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=36 freq=5180 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=40 freq=5200 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=44 freq=5220 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=48 freq=5240 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=149 freq=5745 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=153 freq=5765 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=157 freq=5785 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=161 freq=5805 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=165 freq=5825 MHz max_tx_power=0 dBm
Completing interface initialization
Mode: IEEE 802.11g  Channel: 11  Frequency: 2462 MHz
RATE[0] rate=10 flags=0x1
RATE[1] rate=20 flags=0x1
RATE[2] rate=55 flags=0x1
RATE[3] rate=110 flags=0x1
RATE[4] rate=60 flags=0x0
RATE[5] rate=90 flags=0x0
RATE[6] rate=120 flags=0x0
RATE[7] rate=180 flags=0x0
RATE[8] rate=240 flags=0x0
RATE[9] rate=360 flags=0x0
RATE[10] rate=480 flags=0x0
RATE[11] rate=540 flags=0x0
Flushing old station entries
Deauthenticate all stations
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=2
rtl871x_set_key_ops
rtl871x_set_key_ops
rtl871x_set_key_ops
rtl871x_set_key_ops
Using interface wlx60e3271f0534 with hwaddr 60:e3:27:1f:05:34 and ssid 'BANDIT'
Deriving WPA PSK based on passphrase
SSID - hexdump_ascii(len=6):
     42 41 4e 44 49 54                                 BANDIT          
PSK (ASCII passphrase) - hexdump_ascii(len=10): [REMOVED]
PSK (from passphrase) - hexdump(len=32): [REMOVED]
rtl871x_set_wps_assoc_resp_ie
rtl871x_set_wps_beacon_ie
rtl871x_set_wps_probe_resp_ie
random: Got 20/20 bytes from /dev/random
Get randomness: len=32 entropy=0
GMK - hexdump(len=32): [REMOVED]
Get randomness: len=32 entropy=0
Key Counter - hexdump(len=32): [REMOVED]
WPA: group state machine entering state GTK_INIT (VLAN-ID 0)
Get randomness: len=16 entropy=0
GTK - hexdump(len=16): [REMOVED]
WPA: group state machine entering state SETKEYSDONE (VLAN-ID 0)
rtl871x_set_key_ops
rtl871x_set_beacon_ops
wlx60e3271f0534: Setup of interface done.

lwfinger, 2015-12-30:
The only thing I see in your listing is that all channels have max power of 0.

I have given up on hostapd. I now use the light-weight AP of NetworkManager, which works with the v4.1.8_9499 branch of this repo. Use the connection editor to set it up.

abarcellona, 2016-01-03:
Ok, I have installed  driver v4.1.8_9499 and KDE Connection Editor and everything works fine.
Thanks

closed 2016-01-03 by abarcellona