lwfinger/rtl8188eu #156: ioctl_linux.c: unknown field specified in initializer

berezovskyi, 2016-01-08:
rev 396d6c1
kernel `3.10.40-gdacac96` (Ubuntu L4T R21)
arch `armv7l`
config.gz: https://gist.github.com/berezovskyi/817cd97fad2eb085fd48

Hello Larry,

I am having issues compiling the driver under my kernel. Below is the relevant part of the trace:

```c
  CC [M]  /home/andrew/Downloads/rtl8188eu-master/os_dep/ioctl_linux.o
/home/andrew/Downloads/rtl8188eu-master/os_dep/ioctl_linux.c:7798:2: error: unknown field ‘private’ specified in initializer
  .private = rtw_private_handler,
  ^
/home/andrew/Downloads/rtl8188eu-master/os_dep/ioctl_linux.c:7798:2: warning: initialization from incompatible pointer type [enabled by default]
/home/andrew/Downloads/rtl8188eu-master/os_dep/ioctl_linux.c:7798:2: warning: (near initialization for ‘rtw_handlers_def.get_wireless_stats’) [enabled by default]
/home/andrew/Downloads/rtl8188eu-master/os_dep/ioctl_linux.c:7799:2: error: unknown field ‘private_args’ specified in initializer
  .private_args = (struct iw_priv_args *)rtw_private_args,
  ^
/home/andrew/Downloads/rtl8188eu-master/os_dep/ioctl_linux.c:7799:26: warning: excess elements in struct initializer [enabled by default]
  .private_args = (struct iw_priv_args *)rtw_private_args,
                          ^
/home/andrew/Downloads/rtl8188eu-master/os_dep/ioctl_linux.c:7799:26: warning: (near initialization for ‘rtw_handlers_def’) [enabled by default]
/home/andrew/Downloads/rtl8188eu-master/os_dep/ioctl_linux.c:7800:2: error: unknown field ‘num_private’ specified in initializer
  .num_private = sizeof(rtw_private_handler) / sizeof(iw_handler),
  ^
/home/andrew/Downloads/rtl8188eu-master/os_dep/ioctl_linux.c:7800:2: warning: excess elements in struct initializer [enabled by default]
/home/andrew/Downloads/rtl8188eu-master/os_dep/ioctl_linux.c:7800:2: warning: (near initialization for ‘rtw_handlers_def’) [enabled by default]
/home/andrew/Downloads/rtl8188eu-master/os_dep/ioctl_linux.c:7801:2: error: unknown field ‘num_private_args’ specified in initializer
  .num_private_args = sizeof(rtw_private_args) / sizeof(struct iw_priv_args),
  ^
/home/andrew/Downloads/rtl8188eu-master/os_dep/ioctl_linux.c:7801:63: warning: excess elements in struct initializer [enabled by default]
  .num_private_args = sizeof(rtw_private_args) / sizeof(struct iw_priv_args),
                                                               ^
/home/andrew/Downloads/rtl8188eu-master/os_dep/ioctl_linux.c:7801:63: warning: (near initialization for ‘rtw_handlers_def’) [enabled by default]
make[2]: *** [/home/andrew/Downloads/rtl8188eu-master/os_dep/ioctl_linux.o] Error 1
make[1]: *** [_module_/home/andrew/Downloads/rtl8188eu-master] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-3.10.40-gdacac96'
make: *** [modules] Error 2

```

However, after a compilation failure I found that I already seem to have the driver in my system:

```bash
andrew@tegra-ubuntu ~/Downloads/rtl8188eu 14:32
➟  locate rtl8188e
/lib/firmware/rtlwifi/rtl8188efw.bin
/lib/firmware/rtlwifi/rtl8188eufw.bin
/lib/modules/3.10.40-gdacac96/kernel/drivers/net/wireless/rtlwifi/rtl8188ee
/lib/modules/3.10.40-gdacac96/kernel/drivers/net/wireless/rtlwifi/rtl8188ee/rtl8188ee.ko
/usr/src/linux-headers-3.10.40-gdacac96/drivers/net/wireless/rtlwifi/rtl8188ee
/usr/src/linux-headers-3.10.40-gdacac96/drivers/net/wireless/rtlwifi/rtl8188ee/Makefile
/usr/src/linux-headers-3.10.40-gdacac96/include/config/rtl8188ee.h
```

I am curious if the driver got merged into the mainline at some point of time? If so, when? Is this repo targeted at kernel version below a certain version?

Sorry if you find some of these questions quite basic, maybe you might want to add a few lines to README and point me there instead.

Thank you for your time!

lwfinger, 2016-01-08:
The driver for RTL8188EU was merged into the kernel in version 3.12 on Sept. 2013. I have no idea where the one see came from. This repo is maintained to allow older kernels to build the driver and to provide newer drivers that those merged into the kernel.

When compiling the master branch of this repo, your kernel configuration needs to have CONFIG_WEXT_PRIV selected to have the "private" member of struct iw_handler_def defined. That is your problem.

berezovskyi, 2016-01-11:
Thank you very much for your help!

closed 2016-01-11 by berezovskyi