lwfinger/rtl8188eu #157: Kernel panic

goanga, 2016-01-21:
I have the TP-WN725N dongle and I've tested it on Fedora 23 (4.2.7, 4.2.8) and Ubuntu 15.10 (4.2.0).
On both OS's, it first works without issue with your driver (on Ubuntu there already was a r8188eu which I don't think was available on Fedora).
Then (not sure if after a reboot or just after I take it out and reinsert it) as soon as the dongle is initialized, I consistently get a kernel panic (caps lock flashes). I tried looking for logs but am not sure what to look for. /var/log/syslog on Ubuntu only shows messages up to the actual crash.

lwfinger, 2016-01-21:
Before you take the action that causes the kernel panic, switch to the logging console. On many systems, the key combination CTRL+ALT+F10 will show the necessary screen, but it may be different on Ubuntu. When the panic happens, the dump will be on the screen. Take a photo and post it here. The clearer the better. If you cannot get a photo, then write down and post the traceback.

goanga, 2016-01-21:
Here is thedump.
I think that it happens on WPA2 networks - it worked earlier at a Starbucks (open network), but when connecting to my network at home it panics.

[image]

lwfinger, 2016-01-22:
Nice job on the photo. They are seldom that clear.

The backtrace on the panic shows that it started in routine translate_scan() of driver 8188eu. Because the item above that is __stack_chk_fail(), I suspected stack overflow. When I examined routine translate_scan(), I found a number of large arrays being placed on the stack, which is not recommended. These have been moved off the stack and kzalloc() is now used to allocate and clear the space. The driver should now work.

Please be aware that I do not have this hardware, which is why I had not seen and fixed this problem earlier.

rhempel, 2016-01-22:
@lwfinger - thanks for your tireless work on this series of drivers - we use them in the ev3dev.org project so that the LEGO MINDSTORMS EV3 brick can use pretty much any WiFi dongle you have lying around.

goanga, 2016-01-23:
Thanks for the hard and fast work :+1: . I was only able to test the new driver today and I have good news and bad news :)

The good news is that there is no panic anymore, the driver loads and is able to connect to the network. The bad news is that after a brief period, the system freezes.
Here is what I found in syslog at the moment of the crash, it seems to be the same series of messages everytime: http://pastebin.com/raw/uZmVb9nG

lwfinger, 2016-01-24:
Unfortunately, the tracebacks in that posting do not point back to any component of driver r8192eu. It probably is at fault, but I would have no idea where to look.

I am assuming that you are using the "master" branch. If so, please checkout the "v4.1.8_9499" branch and try it.

I did locate my device that uses this driver, and hit it hard with no problems. In fact, I got a Firefox download speed of up to 4.3 MB/s over an 802.11n link.

goanga, 2016-01-24:
You wrote r8192eu in your message while this is about r8188eu, but I suppose it's a typo.

I've checked out the v4.1.8_9499 branch and compiled it, so far it runs without issue.