lwfinger/rtl8188eu #160: Kernel panic

jvoisin, 2016-02-06:
[image]

lwfinger, 2016-02-06:
When you file a bug report for a repo with multiple branches, always state which branch you are using. From the traceback, I think you have the master branch. If that is not correct, let me know.

Routine rtw_free_recvframe() is checking all the pointers to make sure that they are not NULL. As a result, I'm not sure what is causing your OOPS. The only thing I can suggest is that you switch to the v4.1.8_9499 branch. As it implements the full cfg80211 interface, it is more widely used. Perhaps the bug has been fixed there.

jvoisin, 2016-02-06:
My bad: I was on master.