lwfinger/rtl8188eu #161: Lubuntu 15.10: Connected but no internet access

dieforrockvn, 2016-02-29:
After make install and copy the driver file to lib folder, I can connect to wifi network but no internet access. Now even using wire can't access internet.
Thanks.

lwfinger, 2016-02-29:
Did you copy some files, or did you do 'sudo make install'? Installation does more than copy files.

Nothing you do with wireless drivers should affect the wired network. It seems likely that you user space network configuration is messed up.

dieforrockvn, 2016-02-29:
I did 'sudo make install' and manually copied rtl8188eufw.bin
After reboot I can connect to wifi but can't access to anything.
If you want any more information I will be pleased to give you, just tell me what commands I have to run.

allaudet, 2016-02-29:
Seems like you are having problems with DHCP. That's my guess.
Check if an IP address is assigned to your interface.

dieforrockvn, 2016-03-02:
I have found problem is wrong DNS nameserver.
Thanks.

closed 2016-03-02 by dieforrockvn