lwfinger/rtl8188eu #162: NetworkManager doesn't work with 8188eu driver in CentOS7

kuntan2237, 2016-03-01:
Hi,
I'm installing 8188eu driver in CentOS7, WIFI can be connected successfully but NetworkManager 
starting failed. 
Any help/suggestions would be appreciated!

systemctl logs:
```
[root@localhost network-scripts]# systemctl start NetworkManager
[root@localhost network-scripts]# systemctl status NetworkManager -l
● NetworkManager.service - Network Manager
   Loaded: loaded (/usr/lib/systemd/system/NetworkManager.service; disabled; vendor preset: enabled)
   Active: failed (Result: start-limit) since Tue 2016-03-01 15:25:55 CST; 6s ago
  Process: 5629 ExecStart=/usr/sbin/NetworkManager --no-daemon (code=dumped, signal=SEGV)
 Main PID: 5629 (code=dumped, signal=SEGV)

Mar 01 15:25:54 localhost.localdomain systemd[1]: Unit NetworkManager.service entered failed state.
Mar 01 15:25:54 localhost.localdomain systemd[1]: NetworkManager.service failed.
Mar 01 15:25:55 localhost.localdomain systemd[1]: NetworkManager.service holdoff time over, scheduling restart.
Mar 01 15:25:55 localhost.localdomain systemd[1]: start request repeated too quickly for NetworkManager.service
Mar 01 15:25:55 localhost.localdomain systemd[1]: Failed to start Network Manager.
Mar 01 15:25:55 localhost.localdomain systemd[1]: Unit NetworkManager.service entered failed state.
Mar 01 15:25:55 localhost.localdomain systemd[1]: NetworkManager.service failed.
```
dmesg | grep NetworkManager
```
[root@localhost network-scripts]# dmesg | grep NetworkManager
[ 1069.006437] NetworkManager[5560]: segfault at 8 ip 00007f6bc303d4df sp 00007ffd0da65b90 error 4 in NetworkManager[7f6bc2f60000+1cb000]
[ 1069.466456] NetworkManager[5581]: segfault at 8 ip 00007ff09da474df sp 00007ffe48a35450 error 4 in NetworkManager[7ff09d96a000+1cb000]
[ 1069.704264] NetworkManager[5590]: segfault at 8 ip 00007f80b3efb4df sp 00007ffd29ee1250 error 4 in NetworkManager[7f80b3e1e000+1cb000]
[ 1069.948151] NetworkManager[5614]: segfault at 8 ip 00007fbf8fbfe4df sp 00007fffb3d7ce70 error 4 in NetworkManager[7fbf8fb21000+1cb000]
[ 1070.204881] NetworkManager[5629]: segfault at 8 ip 00007fa90b8b94df sp 00007ffee4c94a50 error 4 in NetworkManager[7fa90b7dc000+1cb000]
```
dmesg | grep 8188eu
```
[root@localhost network-scripts]# dmesg | grep 8188eu
[   11.040443] 8188eu: module verification failed: signature and/or required key missing - tainting kernel
[   11.077759] usbcore: registered new interface driver r8188eu
```
The wireless can be configured and connected successfully, however there's some warnning/errors with wpa_supplicant commands:
`ioctl[SIOCSIWAP]: Operation not permitted`

lwfinger, 2016-03-01:
Your NetworkManager is segfaulting. As far as I know, that cannot be caused by a driver. As the driver works for me on openSUSE 13.2 with NM, my suspicion is that your problem is with Centos.

The ioctl failure is due to lack of implementation of a call that is not necessary.

closed 2016-03-01 by lwfinger

kuntan2237, 2016-03-02:
Hi lwfinger,
Yes you are right, I tried Fedora 23, everything works fine.
Thanks.

lucumt, 2018-03-24:
@lwfinger  so if I want to use it in Centos7,what shall I do now? Thanks!

lwfinger, 2018-03-24:
As I have no idea what Centos did to NetworkManager, I have no idea what to propose. I do not even know what other network control software that they use, but you should try it. I do know that NetworkManager version 1.10 as provided by openSUSE works.

lucumt, 2018-03-25:
@lwfinger thank you very much,I just found it doesn't matter when have this issue,I using dhclient wlan1 to get the ip address and I can connect to the wireless network!