lwfinger/rtl8188eu #164: Signal strength suddenly drops down

allaudet, 2016-03-07:
Hello Larry,

I'm working in one project where I use a **Raspberry Pi2B** with one **Dongle 8188eu** as wireless device and I'm experiencing some problems with this driver.

The Linux distribution I'm using is OpenWrt 15.05 with **kernel 3.18**. The hostapd is v0.8 with the rtl871xdrv REALTEK driver, so the driver ends using **WEXT**.
I'm using your **v4.1.4 (master)** driver version, I've also tried to use the v4.1.8 version but it's not compatible with 3.18 kernel because it lacks two kernel wireless functions that your drivers uses.
I've also turned off the power saver option of the driver (**_rtw_power_mgnt=0 rtw_enusbss=0_**).

- When I connect the wireless as **Client**, the ping delay is very high with **min=5.507**/avg=56.945/max=114.328 (with 39 packets sent). After a while of inactivity, numbers improves slightly to **min=4.305**/avg=10.533/max=54.559 (with 36 packets sent), but still being high.

- But the main problem is with the **Access Point mode**, where appears some signal strength problems which sometimes strongly decreases and prevents any kind of connection with the device and it doesn't recover back.
[image]

I've tried to reset de wireless connection but nothing happens, also tried to restart the hostapd service but neither solves the problem.
The only solution that I found its restarting the 8188eu kernel module (_rmmod 8188eu_ + _insmod 8188eu_) and then setup again the wireless configuration, but it does not guarantee me that it wont happen again.

An additional problem is that I can't identify programmatically the error, "_dmesc_" doesn't show anything and "_ifconfig_" or "_iwconfig_" don't seem to be sufficiently accurate with the power that they show.

Adrià

lwfinger, 2016-03-07:
Please be aware that this is not my driver. What I do is make available the drivers provided by Realtek. As I know nothing of the internals of the wifi chips, my debugging is essentially limited to fixing kernel oops and warnings. Rate setting, etc. is beyond my knowledge.

It seems as though the RX gain setting is changing in a drastic fashion. Unfortunately, I'm not quite sure which register is being used to set that. At the moment, I think that the firmware is given a gain range, and that it sets the gain appropriately.

mxjoe, 2016-03-31:
Same problem here with OpenELEC 6.95.1.

closed 2023-07-03 by allaudet