lwfinger/rtl8188eu #165: WARNING Trace from v4.1.8_9499: Driver non working

allaudet, 2016-03-07:
Since I experience some problems with the other version, I've tried to move on the v4.1.8_9499 compiling new Kernel that supports it.

**Driver version**: _v4.1.8_9499_
**Kernel**: _4.4_
**Platform**: _Raspberry Pi 2B_

Related _dmesg_ output:
```
[    2.627055] usb 1-1.5: new high-speed USB device number 4 using dwc_otg
[    2.747674] usb 1-1.5: New USB device found, idVendor=0bda, idProduct=8179
[    2.754571] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.761933] usb 1-1.5: Product: 802.11n NIC
[    2.766131] usb 1-1.5: Manufacturer: Realtek
[    2.770519] usb 1-1.5: SerialNumber: 243C200CCD26
[    2.864169] init: - preinit -
[    3.009217] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    3.016657] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    3.047606] random: procd urandom read with 74 bits of entropy available
[    4.716304] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    4.723550] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x45E1
[    6.175205] mount_root: mounting /dev/root
[    6.182802] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    6.231998] FAT-fs (mmcblk0p1): codepage cp437 not found
[    6.258228] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    6.273473] procd: - early -
[    6.276589] procd: - watchdog -
[    6.903296] procd: - ubus -
[    7.917304] procd: - init -
[    9.578527] ip6_tables: (C) 2000-2006 Netfilter Core Team
[    9.590144] Loading modules backported from Linux version v4.4-rc5-1913-gc8fdf68
[    9.597616] Backport generated by backports.git backports-20151218-0-g2f58d9d
[    9.646132] ------------[ cut here ]------------
[    9.650915] WARNING: CPU: 2 PID: 501 at compat-wireless-2016-01-10/net/wireless/core.c:361 wiphy_new_nm+0x534/0x654 [cfg80211]()
[    9.662527] Modules linked in: 8188eu(+) cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi snd_seq_device snd_hwdep snd soundcore rfkill vfat fat ohci_platform ohci_hcd ehci_platform ehci_hcd aead crypto_null
[    9.694668] CPU: 2 PID: 501 Comm: kmodloader Not tainted 4.4.4 #9
[    9.700768] Hardware name: BCM2709
[    9.704209] [<80020388>] (unwind_backtrace) from [<8001cde8>] (show_stack+0x10/0x14)
[    9.711987] [<8001cde8>] (show_stack) from [<801b3874>] (dump_stack+0x9c/0xbc)
[    9.719242] [<801b3874>] (dump_stack) from [<800276f8>] (warn_slowpath_common+0x94/0xb0)
[    9.727363] [<800276f8>] (warn_slowpath_common) from [<800277b0>] (warn_slowpath_null+0x1c/0x24)
[    9.736204] [<800277b0>] (warn_slowpath_null) from [<7f0cd68c>] (wiphy_new_nm+0x534/0x654 [cfg80211])
[    9.745659] [<7f0cd68c>] (wiphy_new_nm [cfg80211]) from [<7f176ad0>] (rtw_wdev_alloc+0x28/0x2a0 [8188eu])
[    9.755506] [<7f176ad0>] (rtw_wdev_alloc [8188eu]) from [<7f1663e0>] (rtw_drv_init+0x2ec/0x48c [8188eu])
[    9.765145] [<7f1663e0>] (rtw_drv_init [8188eu]) from [<80277de4>] (usb_probe_interface+0x190/0x24c)
[    9.774314] [<80277de4>] (usb_probe_interface) from [<80224348>] (driver_probe_device+0x140/0x30c)
[    9.783304] [<80224348>] (driver_probe_device) from [<8022457c>] (__driver_attach+0x68/0x8c)
[    9.791772] [<8022457c>] (__driver_attach) from [<802228b4>] (bus_for_each_dev+0x4c/0xa0)
[    9.799979] [<802228b4>] (bus_for_each_dev) from [<80223928>] (bus_add_driver+0xd4/0x210)
[    9.808186] [<80223928>] (bus_add_driver) from [<80224c90>] (driver_register+0xa8/0xec)
[    9.816220] [<80224c90>] (driver_register) from [<80276ae8>] (usb_register_driver+0x70/0x134)
[    9.824898] [<80276ae8>] (usb_register_driver) from [<7f19802c>] (init_module+0x2c/0x30 [8188eu])
[    9.833931] [<7f19802c>] (init_module [8188eu]) from [<80014748>] (do_one_initcall+0x10c/0x1d4)
[    9.842663] [<80014748>] (do_one_initcall) from [<800889a4>] (do_init_module+0x60/0x1c4)
[    9.850781] [<800889a4>] (do_init_module) from [<8008a1f4>] (load_module+0x168c/0x1cb4)
[    9.858812] [<8008a1f4>] (load_module) from [<8008a954>] (SyS_init_module+0x138/0x164)
[    9.866757] [<8008a954>] (SyS_init_module) from [<80009c00>] (ret_fast_syscall+0x0/0x3c)
[    9.874910] ---[ end trace c21f8c764fd4e400 ]---
[    9.879558] ------------[ cut here ]------------
[    9.884226] WARNING: CPU: 2 PID: 501 at compat-wireless-2016-01-10/net/wireless/core.c:371 wiphy_new_nm+0x5fc/0x654 [cfg80211]()
[    9.895837] Modules linked in: 8188eu(+) cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi snd_seq_device snd_hwdep snd soundcore rfkill vfat fat ohci_platform ohci_hcd ehci_platform ehci_hcd aead crypto_null
[    9.927971] CPU: 2 PID: 501 Comm: kmodloader Tainted: G        W       4.4.4 #9
[    9.935299] Hardware name: BCM2709
[    9.938726] [<80020388>] (unwind_backtrace) from [<8001cde8>] (show_stack+0x10/0x14)
[    9.946500] [<8001cde8>] (show_stack) from [<801b3874>] (dump_stack+0x9c/0xbc)
[    9.953753] [<801b3874>] (dump_stack) from [<800276f8>] (warn_slowpath_common+0x94/0xb0)
[    9.961874] [<800276f8>] (warn_slowpath_common) from [<800277b0>] (warn_slowpath_null+0x1c/0x24)
[    9.970712] [<800277b0>] (warn_slowpath_null) from [<7f0cd754>] (wiphy_new_nm+0x5fc/0x654 [cfg80211])
[    9.980117] [<7f0cd754>] (wiphy_new_nm [cfg80211]) from [<7f176ad0>] (rtw_wdev_alloc+0x28/0x2a0 [8188eu])
[    9.989957] [<7f176ad0>] (rtw_wdev_alloc [8188eu]) from [<7f1663e0>] (rtw_drv_init+0x2ec/0x48c [8188eu])
[    9.999593] [<7f1663e0>] (rtw_drv_init [8188eu]) from [<80277de4>] (usb_probe_interface+0x190/0x24c)
[... 172 lines cut ...]
soundcore               4177  1 snd
spi_bitbang             2669  1 spi_gpio
spi_gpio                6191  0 
spidev                  6084  0 
usbhid                 24088  0 
vfat                    8402  0 
x_tables               10153 26 ipt_REJECT
xt_CT                   2837  0 
xt_LOG                  1027  0 
xt_REDIRECT              949  0 
xt_TCPMSS               2792  0 
xt_comment               651 60 
xt_conntrack            2648 16 
xt_id                    646113 
xt_limit                1421 20 
xt_mac                   775  0 
xt_mark                  864  0 
xt_multiport            1492  0 
xt_nat                  1373  0 
xt_state                 969  0 
xt_tcpudp               1944  8 
xt_time                 1830  0 
```

And the commands `ifconfig -a`, `iwinfo`, `iwconfig` and `iw list` don't show any device.

lwfinger, 2016-03-07:
This problem does not show on x86_64 systems, and it is probably due to some incompatibility with ARM architecture. Note that I will need your help to debug this.

Please tell me what statement corresponds to line 361 of compat-wireless-2016-01-10/net/wireless/core.c.

allaudet, 2016-03-07:
Thank you very much for the time,
this is line 361:
```c
	WARN_ON(ops->join_ibss && !ops->leave_ibss);
```

And this is the entire function:
```c
/* exported functions */ <----- Line 337

struct wiphy *wiphy_new_nm(const struct cfg80211_ops *ops, int sizeof_priv,
			   const char *requested_name)
{
	static atomic_t wiphy_counter = ATOMIC_INIT(0);

	struct cfg80211_registered_device *rdev;
	int alloc_size;

	/*
	 * Make sure the padding is >= the rest of the struct so that we
	 * always keep it large enough to pad out the entire original
	 * kernel's struct. We really only need to make sure it's larger
	 * than the kernel compat is compiled against, but since it'll
	 * only increase in size make sure it's larger than the current
	 * version of it. Subtract since it's included.
	 */
	BUILD_BUG_ON(WIPHY_COMPAT_PAD_SIZE <
		     sizeof(struct wiphy) - WIPHY_COMPAT_PAD_SIZE);

	WARN_ON(ops->add_key && (!ops->del_key || !ops->set_default_key));
	WARN_ON(ops->auth && (!ops->assoc || !ops->deauth || !ops->disassoc));
	WARN_ON(ops->connect && !ops->disconnect);
	WARN_ON(ops->join_ibss && !ops->leave_ibss);
	WARN_ON(ops->add_virtual_intf && !ops->del_virtual_intf);
	WARN_ON(ops->add_station && !ops->del_station);
	WARN_ON(ops->add_mpath && !ops->del_mpath);
	WARN_ON(ops->join_mesh && !ops->leave_mesh);
	WARN_ON(ops->start_p2p_device && !ops->stop_p2p_device);
	WARN_ON(ops->start_ap && !ops->stop_ap);
	WARN_ON(ops->join_ocb && !ops->leave_ocb);
	WARN_ON(ops->suspend && !ops->resume);
	WARN_ON(ops->sched_scan_start && !ops->sched_scan_stop);
	WARN_ON(ops->remain_on_channel && !ops->cancel_remain_on_channel);
	WARN_ON(ops->tdls_channel_switch && !ops->tdls_cancel_channel_switch);
	WARN_ON(ops->add_tx_ts && !ops->del_tx_ts);
	WARN_ON(ops->set_tx_power && !ops->get_tx_power);
	WARN_ON(ops->set_antenna && !ops->get_antenna);

	alloc_size = sizeof(*rdev) + sizeof_priv;

	rdev = kzalloc(alloc_size, GFP_KERNEL);
	if (!rdev)
		return NULL;

	rdev->ops = ops;

	rdev->wiphy_idx = atomic_inc_return(&wiphy_counter);

	if (unlikely(rdev->wiphy_idx < 0)) {
		/* ugh, wrapped! */
		atomic_dec(&wiphy_counter);
		kfree(rdev);
		return NULL;
	}

	/* atomic_inc_return makes it start at 1, make it start at 0 */
	rdev->wiphy_idx--;

	/* give it a proper name */
	if (requested_name && requested_name[0]) {
		int rv;

		rtnl_lock();
		rv = cfg80211_dev_check_name(rdev, requested_name);

		if (rv < 0) {
			rtnl_unlock();
			goto use_default_name;
		}

		rv = dev_set_name(&rdev->wiphy.dev, "%s", requested_name);
		rtnl_unlock();
		if (rv)
			goto use_default_name;
	} else {
use_default_name:
		/* NOTE:  This is *probably* safe w/out holding rtnl because of
		 * the restrictions on phy names.  Probably this call could
		 * fail if some other part of the kernel (re)named a device
		 * phyX.  But, might should add some locking and check return
		 * value, and use a different name if this one exists?
		 */
		dev_set_name(&rdev->wiphy.dev, PHY_NAME "%d", rdev->wiphy_idx);
	}

	INIT_LIST_HEAD(&rdev->wdev_list);
	INIT_LIST_HEAD(&rdev->beacon_registrations);
	spin_lock_init(&rdev->beacon_registrations_lock);
	spin_lock_init(&rdev->bss_lock);
	INIT_LIST_HEAD(&rdev->bss_list);
	INIT_WORK(&rdev->scan_done_wk, __cfg80211_scan_done);
	INIT_WORK(&rdev->sched_scan_results_wk, __cfg80211_sched_scan_results);
	INIT_LIST_HEAD(&rdev->mlme_unreg);
	spin_lock_init(&rdev->mlme_unreg_lock);
	INIT_WORK(&rdev->mlme_unreg_wk, cfg80211_mlme_unreg_wk);
	INIT_DELAYED_WORK(&rdev->dfs_update_channels_wk,
			  cfg80211_dfs_channels_update_work);
	device_initialize(&rdev->wiphy.dev);
	rdev->wiphy.dev.class = &ieee80211_class;
	rdev->wiphy.dev.platform_data = rdev;
	device_enable_async_suspend(&rdev->wiphy.dev);

	INIT_LIST_HEAD(&rdev->destroy_list);
	spin_lock_init(&rdev->destroy_list_lock);
	INIT_WORK(&rdev->destroy_work, cfg80211_destroy_iface_wk);
	INIT_WORK(&rdev->sched_scan_stop_wk, cfg80211_sched_scan_stop_wk);

#ifdef CPTCFG_CFG80211_DEFAULT_PS
	rdev->wiphy.flags |= WIPHY_FLAG_PS_ON_BY_DEFAULT;
#endif

	wiphy_net_set(&rdev->wiphy, &init_net);

	rdev->rfkill_ops.set_block = cfg80211_rfkill_set_block;
	rdev->rfkill = rfkill_alloc(dev_name(&rdev->wiphy.dev),
				   &rdev->wiphy.dev, RFKILL_TYPE_WLAN,
				   &rdev->rfkill_ops, rdev);

	if (!rdev->rfkill) {
		kfree(rdev);
		return NULL;
	}

	INIT_WORK(&rdev->rfkill_sync, cfg80211_rfkill_sync_work);
	INIT_WORK(&rdev->conn_work, cfg80211_conn_work);
	INIT_WORK(&rdev->event_work, cfg80211_event_work);

	init_waitqueue_head(&rdev->dev_wait);

	/*
	 * Initialize wiphy parameters to IEEE 802.11 MIB default values.
	 * Fragmentation and RTS threshold are disabled by default with the
	 * special -1 value.
	 */
	rdev->wiphy.retry_short = 7;
	rdev->wiphy.retry_long = 4;
	rdev->wiphy.frag_threshold = (u32) -1;
	rdev->wiphy.rts_threshold = (u32) -1;
	rdev->wiphy.coverage_class = 0;

	rdev->wiphy.max_num_csa_counters = 1;

	rdev->wiphy.max_sched_scan_plans = 1;
	rdev->wiphy.max_sched_scan_plan_interval = U32_MAX;

	return &rdev->wiphy;
}
EXPORT_SYMBOL(wiphy_new_nm);
```

Full code can be obtained from here: http://mirror2.openwrt.org/sources/
And the direct link: http://mirror2.openwrt.org/sources/compat-wireless-2016-01-10.tar.bz2

Anything you need, I'm at your disposal.

lwfinger, 2016-03-09:
That is a strange error. All that section is doing is checking to see that the callbacks required to start and stop a given operation are either both defined or both not defined. In the 3 cases that trigger a warning, both are defined in the driver source, and this warning should not trigger. That is why the x86_64 code works.

Note that this warning is not fatal, and there must be other driver incompatibilities with ARM architecture. Frequently, such problems are caused by issues of variable alignment, where the ARM requirements are stricter than those of the x86_64 architecture. Looking at the code, I see one place that looks suspicious. In the definition of struct wlan_bssid_ex in file include/wlan_bssdef.h, that __packed directive looks suspicious. Try removing that part of that line.

allaudet, 2016-03-10:
Due this weird issues and warning I've decided to change the focus and I get back to kernel 3.18 since it implements de cfg80211 calls.

So instead of compile and use the code provided by "compat-wireless" I've compiled the kernel 3.18 ones.

Now I don't see warnings on dmesg from the module, and seems the mac80211 API is doing his job:
```
root@OpenWrt ~ $ iw wlan0 info
Interface wlan0
        ifindex 3
        wdev 0x1
        addr 24:3c:20:0c:cd:26
        type monitor
        wiphy 0
```

But now I'm facing problems with hostapd.

First I've tried to setup the Dongle using WEXT (if I'm not wrong the v4.1.8 support both wext and nl80211) with this configuration:

```
driver=rtl871xdrv
ieee80211n=1
hw_mode=g
device_name=RTL8188EU
manufacturer=Realtek

auth_algs=1
ignore_broadcast_ssid=0
macaddr_acl=0

channel=11

interface=wlan0
ssid=VUEnnnnn
wpa=2
wpa_passphrase=abwi-rdr
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP

beacon_int=100
wpa_group_rekey=86400
ht_capab=[SHORT-GI-20][SHORT-GI-40]
max_num_sta=8

rts_threshold=2304
fragm_threshold=2304
```

All seems to be configured properly:
```
root@OpenWrt ~/hostapd $ /usr/sbin/hostapd -P /var/run/hostapd-wlan0.pid -B /root/hostapd/hostapd-wext-wlan0.conf -dd
Configuration file: /root/hostapd/hostapd-wext-wlan0.conf
drv->ifindex=3
l2_sock_recv==l2_sock_xmit=0x0x100b8e8
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=2
rtl871x_set_key_ops
rtl871x_set_key_ops
rtl871x_set_key_ops
rtl871x_set_key_ops
Using interface wlan0 with hwaddr 24:3c:20:0c:cd:26 and ssid "VUEnnnnn"
rtl871x_set_wps_assoc_resp_ie
rtl871x_set_wps_beacon_ie
rtl871x_set_wps_probe_resp_ie
rtl871x_set_beacon_ops
rtl871x_set_hidden_ssid_ops
ioctl[RTL_IOCTL_HOSTAPD]: Invalid argument
rtl871x_set_key_ops
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED

root@OpenWrt ~/hostapd # iwconfig
wlan0     IEEE 802.11bgn  ESSID:"VUEnnnnn"
          Mode:Master  Frequency:2.462 GHz  Access Point: 24:3C:20:0C:CD:26
          RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=2/100  Signal level=2/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

But when I try to connect to it, I get:
```
[  157.592497] R8188EU: ap recv deauth reason code(3) sta:e0:b9:a5:2e:e0:95
[  157.600996] R8188EU: set pairwise key to hw: alg:0(WEP40-1 WEP104-5 TKIP-2 AES-4) camid:4
```

A related dmesg chunk:
```
[   32.180703] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   32.189704] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   32.349496] R8188EU: Loaded firmware file rtlwifi/rtl8188eufw.bin
[   32.357153] R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1
[   32.669241] ==> rtl8188e_iol_efuse_patch
[   32.922288] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   33.189678] R8188EU: assoc success
[   33.194615] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   33.202964] R8188EU: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:1
[   33.695817] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   33.705893] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[  157.592497] R8188EU: ap recv deauth reason code(3) sta:e0:b9:a5:2e:e0:95
[  157.600996] R8188EU: set pairwise key to hw: alg:0(WEP40-1 WEP104-5 TKIP-2 AES-4) camid:4
```

And if I uninstall the v4.1.8 and install the v4.1.4 version, that doesn't happens and all works OK (except for an issue which I'm trying to solve using this 4.1.8 version).

lancasrj, 2017-10-03:
allaudet,

I am having this exact same issue

> [ 3147.593467] R8188EU: ap recv deauth reason code(3) sta:c0:ee:fb:5a:60:dc
> [ 3147.600635] R8188EU: set pairwise key to hw: alg:0(WEP40-1 WEP104-5 TKIP-2 AES-4) camid:4 

Did you ever find a solution for v4.1.8?

closed 2023-07-03 by allaudet