lwfinger/rtl8188eu #167: Missing installation step

ivan-radic, 2016-03-23:
Hi there,

thanks for a great repo.
On my Fedora 23 box this is the full installation procedure:

git clone https://github.com/lwfinger/rtl8188eu.git
cd rtl8188eu/
make all
sudo make install

# this step is missing in README.md
# it was not needed in Mint, but it was necessary in Fedora
sudo insmod 8188eu.ko

Best regards,
Ivan

lwfinger, 2016-03-23:
After you do a "make install", you ALWAYS need to load the module. Why not use "sudo modprobe 8188eu"? You could also reboot.

I have trouble believing that it was not needed in Mint.

ivan-radic, 2016-03-24:
I just loaded Fedora Cinnamon spin in live mode, and was able to make wireless connection by doing:

make all
sudo make install
sudo modprobe 8188eu

\# connecting was slow so I restarted network manager with:
systemctl restart NetworkManager

With these steps I was able to get it running without restarting.