lwfinger/rtl8188eu #168: 8188eu: disagrees about version of symbol module_layout

muldy, 2016-03-30:
In ubuntu;
Linux gesrest 4.2.0-34-generic #39-Ubuntu SMP Thu Mar 10 22:11:49 UTC 2016 i686 i686 i686 GNU/Linux

sudo modprobe 8188eu
modprobe: ERROR: could not insert '8188eu': Exec format error

dmesg:
8188eu: disagrees about version of symbol module_layout

lwfinger, 2016-03-30:
You have built the driver against the wrong kernel headers, or the version of gcc you are using is different than was used to build the kernel.

muldy, 2016-04-01:
Thanks!

well the kernel headers match:
make ARCH=i386 CROSS_COMPILE= -C /lib/modules/4.2.0-34-generic/build M=/home/maf/Downloads/rt8192cu  modules
make[1]: Entering directory '/usr/src/linux-headers-4.2.0-34-generic'

must be a gcc version problem...
gcc -v

gcc version 5.2.1 20151010 (Ubuntu 5.2.1-22ubuntu2)

jeremyb31, 2017-02-05:
Hi Larry
I found an issue similar to this on ubuntuforums and the solution seems to be to change the MAKE= in dkms.conf to
MAKE="'make' all KVER=${kernelver}"

as the Make="'make' all" seems to build against the old kernel headers every time rather than the newly installed headers

A similar issue was found in Ubuntu package rtl8812au-dkms, see http://askubuntu.com/a/848037/300665

Have a good day and keep up the good work

lwfinger, 2017-02-06:
The modules are build against the RUNNING kernel. If you want to build against some other source, then you need the KVER option. I have never needed that, thus I do not change the make file. You, however, can do anything you want. That is the beauty of open source/

jeremyb31, 2017-02-06:
Larry,  I am talking about the line in the dkms.conf file not the Makefile.  I would expect the Makefile to work on the running kernel only, but dkms I would expect to compile on the newly installed one

lwfinger, 2017-02-07:
I know nothing of dkms, but taking your word I just pushed that change.

closed 2017-02-07 by lwfinger